arendst/Tasmota #6955: Unwanted dependencies on EnergyReset commands

AIexBV, 2019-11-17:
I am using the Version 6.7.1 on a Gosund SP1 v23 Module. I think it does not make sense to use the whole template here, because the issue is very simple.

In the use case I have, I need the energy-consumption over night from noon to noon (12:00 last day to 12:00 today). So the approach is quite simple. Just readout the energy-consumption of today and yesterday and then reset the counters (_EnergyReset1 0_ & _EnergyReset2 0_).

Unfortunately these resets affect the energy-total counter as well. It will be deducted by the today and yesterday amounts. I currently have no idea how to prevent this and whether it is even intentional or a bug.

Any help will be appreciated.

ascillato2, 2019-11-17:
This is an intended behaviour:

https://github.com/arendst/Tasmota/blob/bb567b523cadd326be308dce2b7011bac1575331/tasmota/xdrv_03_energy.ino#L490-L503

> In the use case I have, I need the energy-consumption over night from noon to noon (12:00 last day to 12:00 today).

So, for your use case that is specific, you can make the calculation in your home automation software. if you don't use any, you can try NodeRed, HomeAssistant or OpenHab.

AIexBV, 2019-11-20:
Actually I do not see an advantage of that behaviour. But anyway, I am able to solve that in the home automation software.

Thank you for clarification.

closed 2019-11-20 by AIexBV