arendst/Tasmota #6956: Is Wifi SDK Flash area erased when forcing short power on/off?

s-hadinger, 2019-11-17:
@arendst I just bricked a Lohas E14 bulb flashing it with the latest 7.0.0.4.

I tuya-converted it, flashed an early 7.0.0.4, did `Reset 5`. Today I flashed a 'Sonoff-minimal 6.7.1 than a 7.0.0.4 from today's commits. I have currently no logs until I solder.

It looks however that Tasmota worked fine (bulb powering up light at latest value) but failed to connect to Wifi. I did several short reboots, and it visibly reset the Module to 1 since the bulb didn't light anymore. The Wifi AP showing is `ESP-xxxxx` instead of Tasmota.

This makes me believe there is corrupt data in the Wifi SDK flash memory.

My question is: when you detect multiple short power off/on, do you also reset the SDK flash area?

arendst, 2019-11-17:
Nope. Any flash resets are currently user induced. Only reset 3 clears sdk flash area.

Shouldn't you now be able to connect to the local AP?

s-hadinger, 2019-11-17:
It boot loops and crashes after `00:00:00 CFG: Use defaults`

``` ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v482516e3
~ld
?bl`r?nlph
00:00:00 CFG: Use defaults

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v482516e3
~ld
?bl`r?nlph
00:00:00 CFG: Use defaults
```

I serial erased flash and flashed it and it came back to life.

s-hadinger, 2019-11-17:
Actually, I suspect a bad flash. The light was at full power when flashing and the Power Supply was maybe too weak to sustain a correct flash.

That's a good lesson, keep the lights OFF when you OTA flash.

closed 2019-11-17 by s-hadinger