arendst/Tasmota #6960: PowerLow, PowerDelta not working with MQTT

toppe, 2019-11-18:
> **GUIDE**
>
> This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
>
> Please DO NOT OPEN AN ISSUE:
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/
>  - If you have a issue when flashing was done via Tuya Convert
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Tasmota/wiki/FAQ) and troubleshooting wiki articles
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
_A clear and concise description of what the bug is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Sonoff Pow R2
- [ ] Tasmota binary firmware version number used: 6ec4373-tasmota
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Mac OS Commandline, OTA
- [x] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
19:47:05 CMD: Backlog Template; Module; GPIO:
19:47:05 MQT: stat/sonoff/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
19:47:05 MQT: stat/sonoff/RESULT = {"Module":{"43":"Sonoff Pow R2"}}
19:47:05 MQT: stat/sonoff/RESULT = {"GPIO":"Not supported"}

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here: 

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
19:49:48 MQT: stat/sonoff/STATUS = {"Status":{"Module":43,"FriendlyName":["Wasmachine"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:49:48 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin","RestartReason":"Power on","Uptime":"0T00:16:37","StartupUTC":"2019-11-18T18:33:11","Sleep":50,"CfgHolder":4617,"BootCount":38,"SaveCount":130,"SaveAddress":"F5000"}}
19:49:48 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"7.0.0.5(6ec4373-tasmota)","BuildDateTime":"2019-11-18T14:00:09","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(bb83b9b)","Hardware":"ESP8266EX"}}
19:49:48 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Joep",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C8000100068000005AFF000000000000","00000000","00000000"]}}
19:49:48 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
19:49:48 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-4519","IPAddress":"192.168.1.26","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"B4:E6:2D:7F:F1:A7","Webserver":2,"WifiConfig":2}}
19:49:48 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7FF1A7","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
19:49:48 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Nov 18 18:49:48 2019","Local":"Mon Nov 18 19:49:48 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:03","Sunset":"17:07"}}
19:49:48 MQT: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerDelta":101,"PowerLow":1,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
19:49:48 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-11-18T19:49:48","ENERGY":{"TotalStartTime":"2019-10-28T16:18:46","Total":16.236,"Yesterday":0.960,"Today":1.290,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":234,"Current":0.000}}}
19:49:48 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-11-18T19:49:48","Uptime":"0T00:16:37","UptimeSec":997,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Joep","BSSId":"78:8A:20:D4:A6:96","Channel":11,"RSSI":42,"LinkCount":1,"Downtime":"0T00:00:06"}}}
19:49:52 MQT: tele/sonoff/STATE = {"Time":"2019-11-18T19:49:52","Uptime":"0T00:16:41","UptimeSec":1001,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Joep","BSSId":"78:8A:20:D4:A6:96","Channel":11,"RSSI":42,"LinkCount":1,"Downtime":"0T00:00:06"}}

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Set PowerLow to 1 or PowerDelta to 101.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
That no MQTT messages are being sent when current power is below threshold or when current power is 0. One "zero usage" message is more than enough.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2019-11-18:
As explained on Discord, auto-telemetry trumps thresholds because telemetry may include information beyond just power metrics. Other sensors do not have threshold settings so these metrics can be interesting even if power is not.

Currently, the only means to eliminate auto-telemetry is to set `TelePeriod 0`. However, this eliminates **_all_** telemetry, not just TelePeriod. **Perhaps a TelePeriod enhancement (e.g., `2`) to eliminate TelePeriod telemetry messages yet retain other telemetry messages.**

ascillato2, 2019-11-22:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-22 by ascillato2

toppe, 2019-11-22:
It’s not answered...?

meingraham, 2019-11-22:
> Currently, the only means to eliminate auto-telemetry is to set `TelePeriod 0`.

This is the answer.

toppe, 2019-11-22:
That is preventing any MQTT message.

How hard can it be? I don’t want MQTT messages if usage is below (or above) the threshold set with ‘PowerLow’

meingraham, 2019-11-22:
> How hard can it be?

Good point. We await your pull request with this enhancement. :wink: