arendst/Tasmota #6961: Losing wifi after upgrade from 7.0.0.3 to 7.0.0.5

s-hadinger, 2019-11-18:
### BUG DESCRIPTION
_A clear and concise description of what the bug is._

@arendst This is the second bulb I fail to upgrade with a similar scenario. Is it possible that DeepSleep gets activated by mistake?

- Tasmota-7.0.0.4 (early version of 7.0.0.4) + Reset 5 + Reset 3
- Tasmota-minimal-6.7.1 (pre-compiled)
- Tasmota-7.0.0.5 (self-compiled)

Wifi comes up for 5 seconds then it is lost until next reboot.

I found the following weird messages in logs:

```
22:19:22 DSL: Reset wrong settings wakeup: -1, slip 10000
22:19:22 RSL: RESULT = {"DeepSleep":{"Time":"1970-01-01T00:59:59","Epoch":-1}}
```
and
```
22:21:35 DSL: Reset wrong settings wakeup: 0, slip 0
22:21:35 RSL: RESULT = {"DeepSleep":{"Time":"1970-01-01T00:59:59","Epoch":-1}}
```

Here is the complete `WebLog 4` until I lose connectivity:

```
00:00:00 CFG: Loaded from flash at F9, Count 599
00:00:00 QPC: Flag 0E
00:00:00 DBG: MY29x1 Found
00:00:00 SRC: Restart
00:00:00 Project tasmota Lampe 1 Version 7.0.0.5(tasmota)-2_6_1
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Livebox-781A in mode 11N as sonoff-5615...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on sonoff-5615 with IP address 192.168.1.66
00:00:04 UPP: Multicast (re)joined
22:31:09 HTP: Hue API (/anonymous/config)
22:31:09 RSL: INFO1 = {"Module":"AiLight","Version":"7.0.0.5(tasmota)","FallbackTopic":"cmnd/DVES_CBD5EF_fb/","GroupTopic":"cmnd/sonoffs/"}
22:31:09 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-5615","IPAddress":"192.168.1.66"}
22:31:09 RSL: INFO3 = {"RestartReason":"External System"}
22:31:09 RSL: RESULT = {"POWER":"ON"}
22:31:09 RSL: POWER = ON
22:31:09 NTP: Drift 0, (UTC) Mon Nov 18 21:31:09 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
22:31:10 QPC: Reset
22:31:11 HTP: Console
22:31:13 RSL: STATE = {"Time":"2019-11-18T22:31:13","Uptime":"0T00:00:11","UptimeSec":11,"Heap":30,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":58,"MqttCount":0,"POWER":"ON","Dimmer":100,"Color":"FF7F0000","HSBColor":"30,100,100","Channel":[100,50,0,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Livebox-781A","BSSId":"30:7C:B2:7D:B6:86","Channel":11,"RSSI":84,"LinkCount":1,"Downtime":"0T00:00:06"}}
22:31:13 DSL: Reset wrong settings wakeup: 0, slip 0
22:31:13 RSL: RESULT = {"DeepSleep":{"Time":"1970-01-01T00:59:59","Epoch":-1}}
22:31:16 HTP: Console
```

### REQUESTED INFORMATION
  Configuration output here:

  ```
22:21:36 RSL: STATUS = {"Status":{"Module":27,"FriendlyName":["Lampe 1"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:21:36 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"External System","Uptime":"0T00:00:12","StartupUTC":"2019-11-18T21:21:24","Sleep":50,"CfgHolder":4617,"BootCount":268,"SaveCount":598,"SaveAddress":"FA000"}}
22:21:36 RSL: STATUS2 = {"StatusFWR":{"Version":"7.0.0.5(tasmota)","BuildDateTime":"2019-11-18T22:08:04","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(bb83b9b)","Hardware":"ESP8266EX"}}
22:21:36 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Livebox-781A",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008001","2805C8000100060000005AFF000000000000","00000000","00000000"]}}
22:21:36 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":478,"Free":524,"Heap":29,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000809","8F0A2383","041483A1","20021705","010000C0","00007880"],"Drivers":"1,2,3,4,5,6,9,10,16,20,21,29","Sensors":"1,2,3,4"}}
22:21:36 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-5615","IPAddress":"192.168.1.66","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"CC:50:E3:CB:D5:EF","Webserver":2,"WifiConfig":4}}
22:21:36 RSL: STATUS7 = {"StatusTIM":{"UTC":"Mon Nov 18 21:21:36 2019","Local":"Mon Nov 18 22:21:36 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:03","Sunset":"17:07"}}
22:21:36 RSL: STATUS10 = {"StatusSNS":{"Time":"2019-11-18T22:21:36"}}
22:21:36 RSL: STATUS11 = {"StatusSTS":{"Time":"2019-11-18T22:21:36","Uptime":"0T00:00:12","UptimeSec":12,"Heap":29,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":53,"MqttCount":0,"POWER":"ON","Dimmer":100,"Color":"FF7F0000","HSBColor":"30,100,100","Channel":[100,50,0,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Livebox-781A","BSSId":"30:7C:B2:7D:B6:86","Channel":11,"RSSI":76,"LinkCount":1,"Downtime":"0T00:00:06"}}}
  ```
  STATUS 0 output here:

  ```

  ```
  Console output here:

  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Unsure how to reproduce, but here is a possible scenario:

- Flash with self-compiled Tasmota-7.0.0.4 (early version of 7.0.0.4)
- Flash with pre-compiled Tasmota-minimal-6.7.1
- Flash with self-compiled Tasmota-7.0.0.5

Wifi comes up for 5 seconds and is definitively lost until next reboot.

s-hadinger, 2019-11-18:
I found the issue:

```
DeepSleepTime
22:38:16 RSL: RESULT = {"DeepSleepTime":-1}
```

Setting `DeepSleepTime 0` solved it (I could send it in the 5 seconds window when I could access the console).

@arendst I don't know why `DeepSleepTime` was -1, but I suggest you set it to `0` if it's out of normal range.

arendst, 2019-11-19:
I've done some testing and still cannot explain why Settings.deepsleep is 0xFFFFFFFF in this case.

``Reset5`` resets flash starting just above the code binary until end of flash to 0xFF. This includes Settings and SDK values AND executes SettingsDefault() which will change all Settings to 0x00 and inits some default values too. Settings.deepsleep should now be 0x0.

``Reset 3`` resets flash starting above the Settings area until end of flash to 0xFF clearing SDK values.

Tasmota-minimal should not change any value in the Settings.

Loading Tasmota 7.0.0.5 should not change any value in Settings too.

I'm a bit eary that apart from Settings.deepsleep being 0xFF there might be more settings set wrong.

In my testing using the debug code allowing me to dump settings area in different stages the settings perform as expected.

Anyway. I updated your PR with what I think does the same as your solution while adding an extra test to not enable deepsleep unwantingly.

Thx again for your self-supporting solution.

arendst, 2019-11-19:
EUREKA!!!!

I know from my EspEasy friend TDer they had issues with certain flash chips. I never noticed it with Tasmota but this time I think we hit the hammer. Chips manufactered by Puya need a special write operation where a flash sector needs to be read first before it can be succesfullt written. I think that's where this goes wrong.

Your flashchip id is "FlashChipId":"146085". The 0x85 is the TUYA manufacturer ID.

I also noted that the core guys did add some code in pre-2.6.0 with regards to this PUYA chip initiated again by TDer. 

For some reason it failed on you but I will dive in deeper and see where it hurts.

Nice issue. I'll reopen for now.

Jason2866, 2019-11-19:
@arendst  needs PUYA support to be enabled?
snipplet from esp.cpp
```
#if PUYA_SUPPORT
static SpiFlashOpResult spi_flash_write_puya(uint32_t offset, uint32_t *data, size_t size) {
    if (data == nullptr) {
      return SPI_FLASH_RESULT_ERR;
    }
    // PUYA flash chips need to read existing data, update in memory and write modified data again.
    static uint32_t *flash_write_puya_buf = nullptr;

    if (flash_write_puya_buf == nullptr) {
        flash_write_puya_buf = (uint32_t*) malloc(PUYA_BUFFER_SIZE);
        // No need to ever free this, since the flash chip will never change at runtime.
        if (flash_write_puya_buf == nullptr) {
            // Memory could not be allocated.
            return SPI_FLASH_RESULT_ERR;
        }
    }

    SpiFlashOpResult rc = SPI_FLASH_RESULT_OK;
    uint32_t* ptr = data;
    size_t bytesLeft = size;
    uint32_t pos = offset;
    while (bytesLeft > 0 && rc == SPI_FLASH_RESULT_OK) {
        size_t bytesNow = bytesLeft;
        if (bytesNow > PUYA_BUFFER_SIZE) {
            bytesNow = PUYA_BUFFER_SIZE;
            bytesLeft -= PUYA_BUFFER_SIZE;
        } else {
            bytesLeft = 0;
        }
        rc = spi_flash_read(pos, flash_write_puya_buf, bytesNow);
        if (rc != SPI_FLASH_RESULT_OK) {
            return rc;
        }
        for (size_t i = 0; i < bytesNow / 4; ++i) {
            flash_write_puya_buf[i] &= *ptr;
            ++ptr;
        }
        rc = spi_flash_write(pos, flash_write_puya_buf, bytesNow);
        pos += bytesNow;
    }
    return rc;
}
#endif
```

arendst, 2019-11-19:
Analysis of the current core versions pre-2.6.0 and up show that the PUYA code is already enabled by default. See file Esp.cpp:
```
#ifndef PUYA_SUPPORT
  #define PUYA_SUPPORT 1
#endif
#ifndef PUYA_BUFFER_SIZE
  // Good alternative for buffer size is: SPI_FLASH_SEC_SIZE (= 4k)
  // Always use a multiple of flash page size (256 bytes)
  #define PUYA_BUFFER_SIZE 256
#endif
```
Using the below platformio compile time directive we have control over PUYA_SUPPORT. Without the parameter the code size is the same as with parameter ``-DPUYA_SUPPORT=1`` so it's safe to conclude the code is implemented:
```
[esp82xx_defaults]
build_flags               = -D NDEBUG
                            -mtarget-align
                            -Wl,-Map,firmware.map
                            -DPUYA_SUPPORT=1
                            -DPUYA_BUFFER_SIZE=4096
```
Disable PUYA_SUPPORT with ``-DPUYA_SUPPORT=0``

Why it then still failed I'm not sure but perhaps the buffer size of just 256 bytes is causing issues. For someone having access to a PUYA flash development system it would be an option to execute the following commands with either the default buffer size of 256 bytes or the full needed flash page size of 4096 bytes. Note also that on systems with TUYA flash this PUYA_BUFFER will be permanently decreasing your RAM size by 4k as it is to be expected that the flash hardware won't chage on the fly ;-)

Test sequence:
- Compile Tasmota with debugging enabled (#define USE_DEBUG_DRIVER)
- Execute command ``CfgDump 0 1000`` and verify correct Settings where most values should be 0x00
- Execute command ``Reset 5`` which will erase flash to 0xFF and restore default settings except your wifi and Mqtt settings. Puya is expected to erase as any other manufacterer. Only write flash is different
- Execute command ``CfgDump 0 1000`` and verify correct Settings where again most values should be 0x00 but some areas may be left to 0xFF

s-hadinger, 2019-11-19:
How do I know if I have a PUYA flash?

I will try the above tonight.

andrethomas, 2019-11-19:
@s-hadinger Your flashchip id is "FlashChipId":"146085". The 0x85

Jason2866, 2019-11-19:
So it is a Puya flash

s-hadinger, 2019-11-19:
I am completely in the dark... I disabled PUYA: `-DPUYA_SUPPORT=0` and checked there was a code size change to confirm.

Before Rest:
```
20:47:40 480:  00 01 03 05  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 490:  00 00 00 00  00 00 1E 00  FF FF FF FF  FF 01 0A 00 |                |
20:47:40 4A0:  00 00 01 00  01 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 4B0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 4C0:  00 00 00 00  00 00 00 00  00 00 FF 00  00 00 70 6F |              po|
20:47:40 4D0:  6F 6C 2E 6E  74 70 2E 6F  72 67 00 00  00 00 00 00 |ol.ntp.org      |
20:47:40 4E0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 6E |               n|
20:47:40 4F0:  6C 2E 70 6F  6F 6C 2E 6E  74 70 2E 6F  72 67 00 00 |l.pool.ntp.org  |
20:47:40 500:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 510:  30 2E 6E 6C  2E 70 6F 6F  6C 2E 6E 74  70 2E 6F 72 |0.nl.pool.ntp.or|
20:47:40 520:  67 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |g               |
20:47:40 530:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 540:  00 00 32 00  00 00 00 00  C0 A8 01 01  FF FF FF 00 |  2             |
20:47:40 550:  C0 A8 01 01  00 00 00 00  25 70 72 65  66 69 78 25 |        %prefix%|
20:47:40 560:  2F 25 74 6F  70 69 63 25  2F 00 00 00  00 00 00 00 |/%topic%/       |
20:47:40 570:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 580:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 590:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 5A0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 5B0:  00 00 00 00  00 00 00 00  00 00 00 00  C0 80 81 55 |               U|
20:47:40 5C0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 5D0:  00 00 00 00  21 16 01 0E  03 48 2E 1A  00 00 00 00 |    !    H.     |
20:47:40 5E0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 5F0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 600:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 610:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 620:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 630:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 640:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 650:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 660:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 32 00 |              2 |
20:47:40 670:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 680:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 690:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 6A0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 6B0:  F8 84 E9 02  AA 02 23 00  00 00 00 00  00 00 00 00 |      #         |
20:47:40 6C0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 6D0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 6E0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 6F0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 700:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 710:  00 00 00 00  00 00 00 00  00 00 FF FF  FF FF FF 11 |                |
20:47:40 720:  47 65 6E 65  72 69 63 00  00 00 00 00  00 00 00 FF |Generic         |
20:47:40 730:  FF FF FF FF  FF FF FF FF  FF FF FF FF  0F 1E EA EA |                |
20:47:40 740:  EA 25 25 25  4F 4F 4F 00  00 00 DD DD  DD 65 C1 15 | %%%OOO      e  |
20:47:40 750:  1F 1F 1F FF  56 61 00 80  00 FA FF FF  1F A3 EC 0E |    Va          |
20:47:40 760:  70 A4 D4 35  35 93 1F 1F  47 C2 66 5A  AF 6F FA FF |p  55   G fZ o  |
20:47:40 770:  FF 99 99 99  00 00 00 00  80 01 20 00  00 00 00 00 |                |
20:47:40 780:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 790:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 7A0:  FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF |                |
20:47:40 7B0:  FF FF FF FF  3B 53 D4 5D  00 00 00 00  FF FF 00 00 |    ;S ]        |
20:47:40 7C0:  00 00 00 00  00 00 00 00  00 00 00 00  29 09 00 00 |            )   |
20:47:40 7D0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 7E0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 7F0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 800:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 810:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 820:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 830:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 840:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 850:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
20:47:40 860:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
[... cut ...]
21:00:11 FD0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:00:11 FE0:  00 00 00 00  00 00 00 00  00 00 00 00  FF FF FF FF |                |
21:00:11 FF0:  FF FF FF FF  FF FF FF FF  BE 49 D4 5D  63 0A 42 6C |         I ]c Bl|
21:00:11 1000:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:00:11 RSL: stat/tasmota/RESULT = {"CfgDump":"Done"}
```

Is it possible that the problem happens only at Power Up and not with Reset which only does a restart?

s-hadinger, 2019-11-19:
I also did a serial flash erase, then flashed with a non-PUYA firmware. I did a full power-up, Wifi configuration and `CfgDump 0 1000`. I still get similar results, no FFs in the deepsleep area...

s-hadinger, 2019-11-19:
Last try, I took another bulb from tuya-conver, flashed directly a test with debug enabled, and here are the values (correct and same as before)...

```
21:32:03 480:  00 01 03 05  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 490:  00 00 00 00  00 00 1E 00  00 00 00 FF  00 01 0A 00 |                |
21:32:03 4A0:  00 00 01 00  01 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 4B0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 4C0:  00 00 00 00  00 00 00 00  00 00 FF 00  00 00 70 6F |              po|
21:32:03 4D0:  6F 6C 2E 6E  74 70 2E 6F  72 67 00 00  00 00 00 00 |ol.ntp.org      |
21:32:03 4E0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 6E |               n|
21:32:03 4F0:  6C 2E 70 6F  6F 6C 2E 6E  74 70 2E 6F  72 67 00 00 |l.pool.ntp.org  |
21:32:03 500:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 510:  30 2E 6E 6C  2E 70 6F 6F  6C 2E 6E 74  70 2E 6F 72 |0.nl.pool.ntp.or|
21:32:03 520:  67 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |g               |
21:32:03 530:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 540:  00 00 32 00  00 00 00 00  C0 A8 01 01  FF FF FF 00 |  2             |
21:32:03 550:  C0 A8 01 01  00 00 00 00  25 70 72 65  66 69 78 25 |        %prefix%|
21:32:03 560:  2F 25 74 6F  70 69 63 25  2F 00 00 00  00 00 00 00 |/%topic%/       |
21:32:03 570:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 580:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 590:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 5A0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 5B0:  00 00 00 00  00 00 00 00  00 00 00 00  C0 80 81 55 |               U|
21:32:03 5C0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 5D0:  00 00 00 00  21 16 01 0E  03 48 2E 1A  00 00 00 00 |    !    H.     |
21:32:03 5E0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 5F0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 600:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 610:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 620:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 630:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 640:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 650:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 660:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 32 00 |              2 |
21:32:03 670:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 680:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 690:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 6A0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 6B0:  F8 84 E9 02  AA 02 23 00  00 00 00 00  00 00 00 00 |      #         |
21:32:03 6C0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 6D0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 6E0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 6F0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 700:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 710:  00 00 00 00  00 00 00 00  00 00 FF FF  FF FF FF 1A |                |
21:32:03 720:  4C 48 2D 35  57 2D 5A 4E  30 31 32 30  34 00 00 00 |LH-5W-ZN01204   |
21:32:03 730:  00 00 00 00  00 00 00 00  8F 00 90 00  00 1E EA EA |                |
21:32:03 740:  EA 25 25 25  4F 4F 4F 00  00 00 DD DD  DD 65 C1 15 | %%%OOO      e  |
21:32:03 750:  1F 1F 1F FF  56 61 00 80  00 FA FF FF  1F A3 EC 0E |    Va          |
21:32:03 760:  70 A4 D4 35  35 93 1F 1F  47 C2 66 5A  AF 6F FA FF |p  55   G fZ o  |
21:32:03 770:  FF 99 99 99  00 00 00 00  80 01 20 00  00 00 00 00 |                |
21:32:03 780:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 790:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 7A0:  FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF |                |
21:32:03 7B0:  FF FF FF FF  8F D1 CD 5D  00 00 00 00  FF FF 00 00 |       ]        |
21:32:03 7C0:  00 00 00 00  00 00 00 00  00 00 00 00  29 09 00 00 |            )   |
21:32:03 7D0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 7E0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 7F0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 800:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 810:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 820:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 830:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 840:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:03 850:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
[... cut ...]
21:32:04 FB0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:04 FC0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:04 FD0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:04 FE0:  00 00 00 00  00 00 00 00  00 00 00 00  FF FF FF FF |                |
21:32:04 FF0:  FF FF FF FF  FF FF FF FF  3C 51 D4 5D  A4 96 28 CA |        <Q ]  ( |
21:32:04 1000:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
21:32:04 RSL: stat/tasmota/RESULT = {"CfgDump":"Done"}
```
```

arendst, 2019-11-19:
I don't dare to ask but if you do the upgrade again from 7.0.0.4 via minimal 6.7.1 to 7.0.0.5 does it work now?

arendst, 2019-11-19:
BTW the two failing ones were they both tuya flash chips?

s-hadinger, 2019-11-19:
Yes, the two failing bulbs where from different brands (Lohas and XMRJ), both PUYA (FlashId ending with 85).

I just tried tuya-convert on a brand new LOHAS, and upgrading from tuya's 7.0.0.3 to 7.0.0.5 with debug enabled: no issue. Then `Reset 5`, still no issue.

I now have a wifi issue, but I'll dig a little and may open a new issue... Maybe linked to 2.6.2.

s-hadinger, 2019-11-19:
Oops, sorry the Lohas chipid is 1440C8 so not PUYA. It may have been a different issue after all, and all the tests above are irrelevant. I need to redo it on the other bulb...

ascillato, 2019-11-19:
> I now have a wifi issue

Try 2.6.1 with SDK: SDK22x_190703. Then do a reset 3 and de-energize it and energize it again.

s-hadinger, 2019-11-19:
I'm now pretty sure I had two distinct issues:
- Lohas bulb (initial one): wifi instability when the light is on, using 2.6.1. Reverting to 2.6.0 seems to avoid the wifi instability though there are occasionally some wifi pauses of 1-2 seconds (retries?).
- XMRJ with a PUYA flash issue.

I need to redo all PUYA tests tomorrow on the XMRJ bulb, and try changing the SDK on the LOHAS.

arendst, 2019-11-20:
Congrats for your tenacity.

I have wifi issues too on low RSSI devices and will also start using SDK22x_190703 on them now.

Th0maz, 2019-11-20:
> * Lohas bulb (initial one): wifi instability when the light is on

Just stumbled upon this. Could this be related to #6608 or #5506 ?
I since have to uncomment `#define PWM_LIGHTSCHEME0_IGNORE_SLEEP` in the builds for my light bulb.

s-hadinger, 2019-11-20:
@Th0maz Thanks, I will try. Although my bulbs don't use software PWM, but the symptoms are indeed the same.

andrethomas, 2019-11-20:
@s-hadinger I believe in @Th0maz 's case its purely the PWM setting sleep to 0 which was causing issues due to the device not having sufficient thermal disipation similar to what we experienced with the sonoff basic R1's. The `#define PWM_LIGHTSCHEME0_IGNORE_SLEEP` just prevents sleep from being set to 0 for light scheme 0.

s-hadinger, 2019-11-20:
@Th0maz @andrethomas Brilliant! Right on the spot!

This was indeed the issue on my Lohas bulb as well. The issue was too fast to be a thermal issue, I guess having `sleep = 0` creates havoc in the SDK.

Now the next question is why does `LightAnimate()` forces `sleep = 0` when the light is on? I don't see any call to `LightAnimate()` except for `FUNC_EVERY_50_MSECOND` which means `sleep = 50` is enough.

I'm willing to change `sleep = 0` with `sleep = 50`, unless there's a good reason for the `0`value.

closed 2019-11-21 by arendst

s-hadinger, 2019-11-21:
I tried a `Reset 5` with the bulb with PUYA and it works fine.

@arendst I think I now understand what happened. Two completely unrelated issues with very similar symptoms - bad luck.

The XMRJ bulb with PUYA flash was tuya-converted a while ago, and the version maybe did not contain the PUYA code at that time. DeepSleep was not activated in my firmware until I flashed a 7.0.0.4 with DeepSleep activated, hence the freeze.

The LOHAS bulb was a `sleep = 0` issue.

Thanks for all your help!

PR #6962 (2019-11-19): Fix check deepsleep for valid values in Settings (#6961)
## Description:

Value of deepsleep happen to be 0xFFFFFFFF (or -1) after some OTA scenarios. This PR resets deepsleep when the settings are out of range.

**Related issue (if applicable):** fixes #6961 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.6
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA]( [... cut]
PR #6980 (2019-11-21): Fix Wifi instability when light is on, due to sleep=0 (#6961, #6608)
## Description:

`sleep = 0` is currently force when light is on, which creates some wifi instability. This PR forces a minimum value `sleep = 50` when lights are on to ensure smooth animations and keep wifi stability.

**Related issue (if applicable):** fixes #6961 #6608 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.6
  - [x] The code change pass travis [... cut]