arendst/Tasmota #6963: Sonoff RF Bridge - RfRaw stops receiving after a while

Waldmensch1, 2019-11-19:
> **GUIDE**

### BUG DESCRIPTION
_A clear and concise description of what the bug is._
- actual Tasmota release "sonoff-DE"
- latest Portisch firmware from /Tools/fw_efm8bb1 Folder

If I enable Rfraw I see packets coming in from several 433MHZ sources around. 

`19:27:40 CMD: RfRaw AAB155
19:27:40 MQT: stat/stadtweg/rfbridge1/RESULT = {"RfRaw":"ON"}
19:27:40 MQT: tele/stadtweg/rfbridge1/RESULT = {"Time":"2019-11-18T19:27:40","RfRaw":{"Data":"AAA055"}}
19:28:10 MQT: tele/stadtweg/rfbridge1/RESULT = {"Time":"2019-11-18T19:28:10","RfRaw":{"Data":"AA B1 06 021C 076C 0BE0 00BE 16DA 0F00 58182B0818184818 55"}}`

This works for a while. Sometimes half an hour sometimes 3 hrs. The receiving seems to be stopped

`21:13:10 MQT: tele/stadtweg/rfbridge1/RESULT = {"Time":"2019-11-18T21:13:10","RfRaw":{"Data":"AA B1 05 0208 0780 0F14 04EC 0726 4818283B18 55"}}
21:13:10 MQT: tele/stadtweg/rfbridge1/RESULT = {"Time":"2019-11-18T21:13:10","RfRaw":{"Data":"AA B1 04 01F4 0794 00C8 05DC 38181A 55"}}
21:13:45 MQT: tele/stadtweg/rfbridge1/RESULT = {"Time":"2019-11-18T21:13:45","RfRaw":{"Data":"AA B1 06 01FE 0730 00A0 04F6 0F1E 0EBA 5818181A183B48 55"}}
21:17:12 MQT: tele/stadtweg/rfbridge1/STATE = {"Time":"2019-11-18T21:17:12","Uptime":"0T05:15:16","UptimeSec":18916,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"iot24","BSSId":"02:9F:C2:DD:2E:79","Channel":1,"RSSI":84,"LinkCount":1,"Downtime":"0T00:00:05"}}
21:22:12 MQT: tele/stadtweg/rfbridge1/STATE = {"Time":"2019-11-18T21:22:12","Uptime":"0T05:20:16","UptimeSec":19216,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"iot24","BSSId":"02:9F:C2:DD:2E:79","Channel":1,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:05"}}
21:27:12 MQT: tele/stadtweg/rfbridge1/STATE = {"Time":"2019-11-18T21:27:12","Uptime":"0T05:25:16","UptimeSec":19516,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"iot24","BSSId":"02:9F:C2:DD:2E:79","Channel":1,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:05"}}
21:32:12 MQT: tele/stadtweg/rfbridge1/STATE = {"Time":"2019-11-18T21:32:12","Uptime":"0T05:30:16","UptimeSec":19816,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":1,"Wifi":{"AP":1,"SSId":"iot24","BSSId":"02:9F:C2:DD:2E:79","Channel":1,"RSSI":86,"LinkCount":1,"Downtime":"0T00:00:05"}}
2`

Checking the RFraw staus shows that RFraw is on

`07:02:26 CMD: RfRaw
07:02:26 MQT: stat/stadtweg/rfbridge1/RESULT = {"RfRaw":"ON"}`

But nothing is received anymore

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Sonoff RF Bridge
- [x ] Tasmota binary firmware version number used: Sonoff Version 6.7.1(sonoff)-STAGE
  - [ ] Pre-compiled
  - [ x] Self-compiled
    - [x ] IDE / Compiler used: VSCode + PlatformIO
- [x ] Flashing tools used: esptool.py to erase the Flash, PlatformIO to flash
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:

07:18:30 CMD: Backlog Template
07:18:31 MQT: stat/stadtweg/rfbridge1/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}

07:19:22 CMD: module
07:19:22 MQT: stat/stadtweg/rfbridge1/RESULT = {"Module":{"25":"Sonoff Bridge"}}

07:19:49 CMD: gpio
07:19:49 MQT: stat/stadtweg/rfbridge1/RESULT = {"GPIO2":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO12":{"0":"None"},"GPIO14":{"0":"None"}}

  ```
  Configuration output here:

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here: 07:20:57 CMD: status 0
07:20:57 MQT: stat/stadtweg/rfbridge1/STATUS = {"Status":{"Module":25,"FriendlyName":["Sonoff"],"Topic":"stadtweg/rfbridge1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
07:20:57 MQT: stat/stadtweg/rfbridge1/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-DE.bin","RestartReason":"Power on","Uptime":"0T15:19:01","StartupUTC":"2019-11-18T15:01:56","Sleep":50,"CfgHolder":4617,"BootCount":7,"SaveCount":14,"SaveAddress":"F6000"}}
07:20:57 MQT: stat/stadtweg/rfbridge1/STATUS2 = {"StatusFWR":{"Version":"6.7.1(sonoff)","BuildDateTime":"2019.11.18 14:38:35","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
07:20:57 MQT: stat/stadtweg/rfbridge1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["iot24","FRITZ!Box Turbo Cable"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000"]}}
07:20:57 MQT: stat/stadtweg/rfbridge1/STATUS4 = {"StatusMEM":{"ProgramSize":538,"Free":464,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000407","8F80E397","043683A1","22B617CD","01001BC0","0000F081"],"Drivers":"1,2,3,4,6,8,9,10,16,18,19,20,21,22,24,26,30","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
07:20:57 MQT: stat/stadtweg/rfbridge1/STATUS5 = {"StatusNET":{"Hostname":"stadtweg/rfbridge1-5133","IPAddress":"192.168.178.177","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.74","Mac":"84:0D:8E:65:94:0D","Webserver":2,"WifiConfig":4}}
07:20:57 MQT: stat/stadtweg/rfbridge1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.46","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_65940D","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
07:20:57 MQT: stat/stadtweg/rfbridge1/STATUS7 = {"StatusTIM":{"UTC":"Tue Nov 19 06:20:57 2019","Local":"Tue Nov 19 07:20:57 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"07:36","Sunset":"16:15"}}
07:20:57 MQT: stat/stadtweg/rfbridge1/STATUS10 = {"StatusSNS":{"Time":"2019-11-19T07:20:57"}}
07:20:57 MQT: stat/stadtweg/rfbridge1/STATUS11 = {"StatusSTS":{"Time":"2019-11-19T07:20:57","Uptime":"0T15:19:01","UptimeSec":55141,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"iot24","BSSId":"02:9F:C2:DD:2E:79","Channel":1,"RSSI":80,"LinkCount":1,"Downtime":"0T00:00:05"}}}

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_
- RfRaw AAB155
- wait a while

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
- if RfRaw is enabled it should receive rawmessages forever

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Waldmensch1, 2019-11-19:
I have to add that disable RFraw and re-enable RFraw start receiving again

07:48:55 CMD: RfRaw 0
07:48:55 MQT: stat/stadtweg/rfbridge1/RESULT = {"RfRaw":"OFF"}
07:49:01 CMD: RfRaw AAB155
07:49:01 MQT: stat/stadtweg/rfbridge1/RESULT = {"RfRaw":"ON"}
07:49:01 MQT: tele/stadtweg/rfbridge1/RESULT = {"Time":"2019-11-19T07:49:01","RfRaw":{"Data":"AAA055"}}
07:49:35 MQT: tele/stadtweg/rfbridge1/RESULT = {"Time":"2019-11-19T07:49:35","RfRaw":{"Data":"AA B1 05 0212 0776 0D3E 03C0 0EF6 48182B 55"}}
07:49:35 MQT: tele/stadtweg/rfbridge1/RESULT = {"Time":"2019-11-19T07:49:35","RfRaw":{"Data":"AA B1 03 0212 076C 0780 281818 55"}}

ascillato2, 2019-11-22:
Hi, I can't reproduce your issue. Sorry.

Please, try:

* update to latest Tasmota from the precompiled bins (http://thehackbox.org/tasmota/tasmota.bin)
* do a `reset 3` to clean the flash
* update portish firmware to latest

and try again.

Waldmensch1, 2019-11-22:
Version":"6.7.1(sonoff)" is the latest release. The Development branch is not compilable as I have seen from commits. Ich checked the tag out and only entered my WiFi and MQTT setup to user_config_override.h. This shouldn't change things in essential code.

The flash was already erased with esptool.py. Otherwise it wasn't possible to write the portisch FW.

Is there a newer portisch than the one with latest file date in /Tools/ folder?

To be honest, I not really need that RfRaw ATM. If you are in close moods, then close. I only found it worth to report because I can reproduce it. Maybe there is a crazy Weatherstation near to the RF Bridge which sends ugly 433 packets around which cannot eaten up by portisch?!

ascillato, 2019-11-22:
> The Development branch is not compilable as I have seen from commits

Sorry, It is working fine. That is why I attached a link to the precompiled bin of latest Tasmota 7.0.0.6

> the flash was already erased with esptool.py. Otherwise it wasn't possible to write the portisch FW.

Those 2 things are not related. Sorry. esptool is for the ESP8266 chip. Portish is for the RF Chip.

> To be honest, I not really need that RfRaw ATM. If you are in close moods, then close.

Sorry that you are misinterpreting what an issue tracker means. You post an issue, so if someone have any idea could help you. You just received my comments trying to help you. Your issue is really rare, not common due to sonoff bridge devices are very reliable. So that leds to a bad flashing, or unclean flash, or bad portish flash. 

Anyway, when you have time to continue testing, please post your results. Thanks.

closed 2019-11-22 by ascillato2

Tommy-LSA, 2019-11-22:
If I would work this way I would get fired. As you maybe aware off, the last essential changes on snfbridge in Git are older than 3 monthes. You can talk this way with a script kiddy from street to push problems away from desk just by paining people with filling kilometers of senseless forms.

Just saying „works4me“ is easy. You haven’t filled the form with your positive test results as I did with my observations. So no clue where you have tested, which HW revision you used for your test. And how long you have tested. As I said, sometimes it occurs after 3hrs. Finally I doubt you own a RF bridge at all.

Hopefully you feel well in your little god-mode 😂😂😂😂😂😂

Just my 2 cent.

meingraham, 2019-11-22:
@Tommy-LSA 

We are taking actions to dock all of @ascillato pay retroactive to when he began volunteering on the project. 

@ascillato, please be advised so you can contact your tax lawyer to file an amended tax return accounting for this decrease of $0 from your income.

Regards.

Tasmota Human Resources

P.S. @Tommy-LSA , This gave you away:
> The flash was already erased with esptool.py. Otherwise it wasn't possible to write the portisch FW.

It's quite possible that you did not follow the procedures properly even after @ascillato provided some suggestions

Tommy-LSA, 2019-11-22:
I erased the flash, flashed latest release, flashed portisch. So another erase would make the flash super-clean? That‘s nonsens, but if you feel better you can still believe in.

meingraham, 2019-11-22:
No... there are two chips that each have their own firmware.

The ESP8266 uses Tasmota and you use esptool to erase and flash THAT chip. It does not touch the RF chip.

THEN you use Tasmota to flash Portisch onto the EFM8BB1.

Your statement indicates that you are conflating these two things.

jziolkowski, 2019-11-22:
You'd be surprised how often users think that when Esptool shows "flash successful" they end up with bad flash, due to for example inefficient power output from the flashing adapter.

meingraham, 2019-11-22:
But what do we know.

Tommy-LSA, 2019-11-22:
If you would read you would know that I did it exactly this way
- erase
- flash Tasmota 
- flash Portisch with flash option in webif 

That‘s what written in Wiki and that‘s what I did. How should I get the rawdata otherwise? Do you really had at least a small view on the first post and you really understood what the request is about? 😳

meingraham, 2019-11-22:
Right from the get go, the issue template asks that this kind of problem be taken to Discord as this situation is not a confirmed bug nor a feature request. If this simple step cannot be followed, then there is skepticism about how thoroughly other procedures have been followed.

> `How should I get the rawdata otherwise? Do you really had at least a small view on the first post and you really understood what the request is about?`

> `So another erase would make the flash super-clean? That‘s nonsens (sic), but if you feel better you can still believe in.`

We do understand. From helping hundreds of other users. Unfortunately even with `esptool.py` stating "flash successful", sometimes it actually isn't (what @jziolkowski was referring to). Thus, the recommendation is to indeed erase, again. Because, something did get written during that successful unsuccessful flash which requires that the flash chip be erased, again. The flash can be "corrupt" due to poor power from the serial adapter, cheap flash memory chips with flaws, etc.

All that we have posted here is mentioned in the wiki. Sometimes it is up to the user to do the work to fully troubleshoot.

Tasmota is provided to users for free. Our time is on a volunteer basis. That means that we can't be fired? No, what it actually means is that we want to help users. But then users like you begin to throw personal insults and attempts to belittle:  
> `Hopefully you feel well in your little god-mode`
> `That‘s nonsens (sic), but if you feel better you can still believe in`

I can practically guarantee you that approach will not advance your cause. But nobody is forcing you to use Tasmota. That's a decision you are free to make.

Tommy-LSA, 2019-11-22:
As I already mentioned, I don’t need the rfraw ATM. My intention was just to help by reporting a bug as it is usually done in Open Source projects. Wasn‘t aware that the only result off filling this sheet ends in „get it away from desk by request details as long as the requester gives up“
So leave it as it is if it is good enough for you. I‘m coding by myself since more than 20 years. Unit tests and support are my daily business. So I don‘t need that somebody talk to me from top to down or provide me URLs to pre-compiled binaries. I have no problem getting rfraw running. It simply stops after a random time.

As said, if a semi-perfect software is the goal, than it‘s as it is.

By the way, again first post not carefully readed: I used the Python tool to erase the flash and flashed with PlatformIO afterwards. What sense bugreports do, if nobody is willing to read or understand the Problem

meingraham, 2019-11-22:
All software is imperfect. In this situation, there are many software components, all of which are imperfect.

The evidence is not conclusive that it's a bug. Only more troubleshooting will determine that one way or the other. A bad flash is out of our control. esptool is provided by the chip manufacturer. All we can do is suggest that the symptoms (runs for a bit and then something causes it to stop) suggest a bad flash of either the ESP or the RF. Support for Tasmota should be directed at the Tasmota server on Discord. We find that platform better for troubleshooting which is what this is at this point. It might also be that the problem lies in the Portisch firmware itself. In that case, we can't troubleshoot that and we would direct you to that project for assistance.

Again, your choice. With over 4 million downloads, it seems a lot of Tasmota users believe it's pretty OK... even if imperfect.

ascillato, 2019-11-22:
@Tommy-LSA Sorry, I don't understand why you have jumped in the thread to attack me. I just tested, couldn't reproduce the bug and ask to follow 3 steps instructions to start the troubleshooting. What is the problem there?

Then the user tell us that ATM is not going to use RFRAW and don't want to do further testings. So, what we should do in that case?

Tommy-LSA, 2019-11-22:
If I would ask for troubleshooting I would do that on Discord. Git is for reporting Bugs and I did that. What you do with Bug reports is up to you. What I have learned - reporting a bug makes no sense because it get ignored and closed. Nobody cares. I would expect that the maintainer of the failing module at least have a look on. As it seems, the maintainer have to be faster than you to see open tickets. 😂😂😂😂

Because I don’t use rfraw at the moment my interest for useless flashing force and back, to get finally the answer that it must be related to earthrays, because it works4you, is less than low

Again and as written in 3rd or 4th comment - do what you want with the Bugreport. I just tried to help.

meingraham, 2019-11-22:
The evidence is not conclusive that it's a Tasmota bug.

Opening a bug report or jumping on one doesn't make it a bug either.

Until proven conclusively or via a dialog on Discord, then we will continue to direct users to Discord. It's how we've chosen to deal with Tasmota issues. Other projects may chose to use GitHub for troubleshooting. Each project has it's own quirks I guess. Ours is Discord.