arendst/Tasmota #6968: VBus Resol Support

javier6d2, 2019-11-19:
**Have you looked for this feature in other issues and in the wiki?**  Yes

**Is your feature request related to a problem? Please describe.**  No

**Describe the solution you'd like**  
It would be nice to have support for Resol solar controllers brand, there are developer codes for Arduino and I think it would be interesting to include it in Tasmota. Currently there are libraries for arduino here: https://www.arduinolibraries.info/libraries/vbus-arduino or good sketch here:
https://github.com/bbqkees/vbus-arduino-domoticz

Note: you need a specific converter circuit to make the VBus data readable for the ESP UART

**Describe alternatives you've considered**  
At the moment I have the sketch reading 4 sensor values form Resol controller. You need to compile one firm to Mega and another for ESP8266. The first read values from Vbus to UART converter and second sent data wireless to host (Domoticz) via HTTP GET requests. This is the scheme:

[image]

I can publish my codes although they are an adaptation of the aforementioned

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2019-11-19:
Added the label to see if there is someone interested on working on this.

ascillato2, 2019-11-22:
Closing this issue as there is noone working on this. Added the label for future reference in case someone wants to work on this. Thanks for sharing your ideas.

closed 2019-11-22 by ascillato2

nagyrobi, 2021-02-25:
Tasmota supports VBus natively using Smart Meter driver:
https://tasmota.github.io/docs/Smart-Meter-Interface/#resol-deltasol-bs-plus-vbus

javier6d2, 2021-02-26:
@nagyrobi when I enable dump with: sensor53 d1 my results shows headers with more than 10 hex characters long. What I can be doing wrong?

[image]

Thanks a lot for this new feature.

bsp9493, 2022-07-11:
Did you find an answer to this dump.

I am in the same boat.

mid it a bad conversion circuit, or something else.

javier6d2, 2022-07-11:
> Did you find an answer to this dump.
> 
> I am in the same boat.
> 
> Is it a bad conversion circuit, or something else.
Hi,
If you see no values in WEBGUI you probably forgot:
#define SML_REPLACE_VARS 
before compile firmware. Please, read the tasmota docs for further info

bsp9493, 2022-07-11:
it's there, along with other's that are recommended:(

#ifndef USE_SCRIPT
#define USE_SCRIPT
#endif
#ifndef USE_SML_M
#define USE_SML_M
#endif
#ifdef USE_RULES
#undef USE_RULES
#endif
#ifndef SML_REPLACE_VARS
#undef SML_REPLACE_VARS
#endif

pieterdt, 2022-12-10:
@bsp9493 , @javier6d2 , which converter circuit do you use?