arendst/Tasmota #6969: Enhance or Eliminate SetOption51

meingraham, 2019-11-19:
**Have you looked for this feature in other issues and in the wiki?**  
Yes

**Is your feature request related to a problem? Please describe.**  
N/A

**Describe the solution you'd like**  
As of [this commit](https://github.com/arendst/Tasmota/commit/7b10bec8fe63bcfa464bd221c5eeb4fae0d03a98), Tasmota is able to query for the type of ESP chip in the device via `GetDeviceHardware()`. Currently, `SetOption51` allows a user to "expose" `GPIO9` and `GPIO10` as available pins. This is provided since the ESP8255 frees these pins up. However, the option does not prevent a user for enabling `SetOption51` if they have an ESP8266. Now that Tasmota is able to know the type of chip, it ought to prevent a user from enabling this feature if the device uses an ESP8266.

**Describe alternatives you've considered**  

- Leave as is and let users do something they shouldn't.
- Get rid of SetOption51 altogether and just "manage" `GPIO9` and `GPIO10` automatically.

arendst, 2019-11-20:
Mike, you're too quick. I noticed some devices do not report the correct chip. Needs more time before abandoning SO51.

meingraham, 2019-11-20:
Noted. Archive the request for now then.

Thanks

Mike

P.S.
> Mike, you're too quick.

Hmmm, my wife said something similar the other night ;-)

ascillato2, 2019-11-21:
> Archive the request for now then.

Ok. Let's reopen this issue if we can add this automatic selection.

closed 2019-11-21 by ascillato2

arendst, 2019-11-21:
It now automaticcaly selects the correct functionality based on hardware. User can still overrule wit SO51 1