arendst/Tasmota #6970: Rename Sensors / give them friendly names

ryaske, 2019-11-19:
Hello, I have three DS18B20 connected and working perfectly.

However, the names are DS18B20 -1, DS18B20-2, DS18B20-3

How can these be renamed to friendly names? I am assuming they cant be because I was not able to find any info about this what-so-ever.

Might the only way this is possible is compiling my own build with the names embedded?

I think the feature request in that case is obvious.
Thank you
Ryan

ascillato2, 2019-11-19:
> How can these be renamed to friendly names?

They can't sorry. Not Supported.

> Might the only way this is possible is compiling my own build with the names embedded?

yes, exactly.

> I think the feature request in that case is obvious.

Added the label to see if there is someone interested on working on this.

ascillato, 2019-11-19:
If you don't need this RENAMED SENSOR in Home Assistant Discovery, as a workaround you can use **rules** feature of Tasmota and save the value in a VAR and then Send by MQTT with the name you want.

If you want this sensor renamed in Home Assistant, you can edit its name by Home Assistant UI.

ryaske, 2019-11-20:
> If you don't need this RENAMED SENSOR in Home Assistant Discovery, as a workaround you can use **rules** feature of Tasmota and save the value in a VAR and then Send it by MQTT with the name you want.
> 
> If you want this sensor renamed in Home Assistant, you can edit its name by Home Assistant UI.

Thanks so much, ascillato. 

I do want this to reflect properly directly on the Tasmota UI, which that solution doesn't cover, correct? 
(It would certainly be nice to have it sent as the default MQTT name as well to make things more cohesive as I often use MQTT Explorer for troubleshooting)

I found two references in the rule cookbook for modifying MQTT, and studied the Rules wiki page---  but they seem to only ever modify the variable, not the MQTT sensor name. I cant figure out how to create a rule that does that. Do you have any suggestions? 

as I try and figure this out,,  is it possible to change the names any other way? Can you confirm it would be possibly be editing the firmware? (Any tips there or in getting this rule to work would be appreciated.

Finally, will the device alaways send 6 messages, 3 with the old names and 3 with my new names? Can I have it only do the latter? Less MQTT traffic the better.

ascillato, 2019-11-21:
Example of a rule that sends the temperature with a friendly name:

On the Tasmota console type:
```
Rule1 1 
Rule1 on tele-DS18B20-1#Temperature do publish stat/kidsroom/temperature %value% endon on tele-DS18B20-2#Temperature do publish stat/kidsroom/temperaturebed1 %value% endon
```

ascillato, 2019-11-21:
> I do want this to reflect properly directly on the Tasmota UI

Tasmota UI is for configuration. For daily use, Tasmota is intended to be used by mqtt, so you should use the UI of your home automation software.

ascillato2, 2019-11-21:
Closing this issue as there is a workaround and no one is working on this.

Thanks for sharing your ideas.

ryaske, 2019-11-21:
Thank you very much Adrian, for providing guidance towaqrds that rule! I wlil make that happen today.

I agree that I will use HA primarily, but for troubleshooting-- and when my HA is down for whatever reason, I do utilize the Tasmota UI and so I want it to reflect the same. Otherwise it is very unclear.  

I will spend some time trying to bake my change into a custom firmware edit --- and if I find it reasonably possible I will post a short how-to here to help anyone else that may want to do this.  Because, I agree that the feature request is not worth baking into a main release.

Thanks again!!

mlavelle42, 2020-02-26:
Any progress on this?

ascillato, 2020-02-26:
Sorry. There isn't, because this issue is closed and a workaround exist for this.

kctrey, 2020-06-17:
So this issue got closed after *2 days*?

Is regards to ascillato's comment, it seems crazy to say that the UI is for configuration only. There are plenty of us that use these device completely stand-alone as timers, thermostats, and security systems. This is one of the most useful requests I have heard, maybe even better than "I wish i could define a custom UI color scheme", which, by the way, seems like a waste of memory for a UI that is made for configuration only.

mlavelle42, 2020-06-17:
I, too, was surprised that DS18B20 sensors could not be named within Tasmota. This, in my opinion, is a significant omission. I understand your logic, that naming can and should be managed within Home Assistant (or whatever platform). But don't you detect, within your own comments, that you are "right" on this matter, while others "don't understand what Tasmota is all about"?

My take is different. Tasmota is the single best software app to come along (for this type of work) ever. Period. The number of configuration arguments are over-the-top. Incredible. However, the ability to name a sensor within a Tasmota screen is, from my experience, a fundamental property that should have been included from the beginning. The Rule workaround is clumsy if only because it requires text editing rather than form editing. 

Please reconsider a sensor naming feature within Tasmota's UI. It would definitely simplify setup and support.

dieckfr, 2020-07-10:
> Thank you very much Adrian, for providing guidance towaqrds that rule! I wlil make that happen today.
>
> I agree that I will use HA primarily, but for troubleshooting-- and when my HA is down for [...]

Did you find a solution? I think that is was a very nice feature to do that, yust for a stand alone solution... Nice greetings!

ilgrank, 2020-07-14:
+1 for this, but I really doubt @ascillato  will be looking here again, since this issue is closed.

ascillato, 2020-07-14:
You are free to provide a pull request with the change you need.
If the code addition uses few bytes (flash and RAM) you have a good chance that Theo will merge it.

ilgrank, 2020-07-14:
@ascillato :aren't pull request meant to contribute code? Alas, I've no code to contribute, just asking for an enhancement.

ascillato, 2020-07-14:
Hi, 

Yes, a Pull Request is code changes. Tasmota is open for pull requests, but someone needs to work on that. 

In this issue it has been requested an option to rename sensors but nobody have spent their free time addressing this. So the label **requested feature (hold over)** has been added.
Also, a workaround has been proposed with the use of rules for sending by mqtt, the sensor name you want. 

If you want, you are free to fork Tasmota and modify it to suit your needs and then provide a PR for adding those changes to the main repository.

There are many opened feature requests, but we need more people to work on them in order to transform wishes into code. 

As you can see from old issues, people just spend their time only on the new feature they are interested. I can reopen this issue if that is what you want me to do.

ilgrank, 2020-07-15:
I'd gladly contribute, but I'm not a programmer. So I can't provide any useful code, that's why I'd call it a feature request, rather than a pull request. 
There is a workaround indeed, but a proper solution would help. Maybe re-opening the issue is the right thing to do, thanks

ilgrank, 2020-07-15:
if @dieckfr comes out with any code, that would be great for a pull request. btw 👍

ascillato2, 2020-07-15:
Reopening as requested.

ilgrank, 2020-07-15:
..duplicated of what other issue?

dieckfr, 2020-07-15:
Hello again! Sorry for the late response... So normaly I'am Developer for Codesys Software, I Engineering machines, and as you can read, my German is better then my Englisch :)

I had a look inside the Code, and I must say, wow, what a great Job in Style and Design and flexibility, that is very impressive... And thats the Problem, not very Easy for a  DAU Programmer like ME... I searched much time to understand how is the HTML Code produced, picked up, by feeling thousend pointers, and found at least, the correct ( I hope, not testet yet) line for building the Names of the DS18XXXX. That is what I need. I hope i can test it tonight (hardcoded) and can Show you the result, so that everbody can make a user Build.  I dont think that I can implement a Command for User Sensor Names in the next Weeks, or the next Years :) ist not my Code, and flexibility of the Code makes it not easyer... 

But what is about a Request? Iam newbie in Github, and Iam not understand the Complete Prozesses here at the Moment... How can I make a Request for this Feature, to Mr. Arends or a good programmer :). Can sombody explain me the procedure? I want to make it, and learn the schematics of Github :)

OffTopic:
By the way i started whit Tasmota while buying a House and want to make it a little Smart... at the moment i start to read out my old ferraris Power Meter. It Counts good in Tasmota, but is it possible to Show a Countervalue no thats not right ... somthing like this Countervalue -> Variable (its Countervalue/75) -> WebUi as kWh. that is what i want... ther are only Smart Meter options in Tasmota, but i found nothing to use one whith only a counter Value.   

HAve a great Time! And nice greetings from Mettingen Germany

kugelkopf123, 2020-07-16:
> OffTopic:
> By the way i started whit Tasmota while buying a House and want to make it a little Smart... at the moment i start to read out my old ferraris Power Meter. It Counts good in Tasmota, but [...]

You could use the smart meter interface as explained in the [docs](https://tasmota.github.io/docs/Smart-Meter-Interface/). 
Just look for counters. There are some options to manipulate the counter value.
Most of the things are for smart Meters. You will find some examples that show the use of counters. Mostly they are used for gas and water. But it is no problem to use this option for electricity as well.

ilgrank, 2020-07-16:
@dieckfr : as for code contribution, you have to go to repository for Tasmota and fill a Pull request (mush as you did to open this issue) and subit the proposed code

dieckfr, 2020-07-19:
Hello again :),

I have found a solution, and for is it working... A little bit quick and
dirty, you have to do a hardcoding for the names, and at the Moment i have
done it only for the DS18 Sensors, still working for DHT support, and i
think finally i will make a config file, or when i have more Time to learn
about the code language and the Projekt itself, maybe a Command
solution...  I will put it to Github... when i know how to do that...

I post here the prinzipal changes, it is not very much, but i spend a
couple of time to finger out how everything works together...

*1. For the DS18XX Sensor change in xsns_05_ds18x20 at codenumber 507:*

#ifdef USE_WEBSERVER
      } else {
        if (0 == i) {
          WSContentSend_PD(HTTP_SNS_TEMP_USER, PSTR("USERNAME1"),
temperature, TempUnit());   // HTTP_SNS_TEMP_USER is without show
"Temperatur in WebUi show the second Change
        }
        else if (1 == i) {
            WSContentSend_PD(HTTP_SNS_TEMP, PSTR("USERNAME1"), temperature,
TempUnit());
            }
        else if (2 == i) {
            WSContentSend_PD(HTTP_SNS_TEMP, PSTR("USERNAME1"), temperature,
TempUnit());
            }
        else if (3 == i) {
            WSContentSend_PD(HTTP_SNS_TEMP, PSTR("USERNAME1"), temperature,
TempUnit());
            }
        else {
            WSContentSend_PD(HTTP_SNS_TEMP, ds18x20_types, temperature,
TempUnit());
            }

#endif

2. Change in i18n.h at codenumber 732

// xdrv_02_webserver.ino
#ifdef USE_WEBSERVER
// {s} = <tr><th>, {m} = </th><td>, {e} = </td></tr>

const char HTTP_SNS_TEMP[]          PROGMEM = "{s}%s "  "</th><th>"
D_TEMPERATURE         "{m}%s " D_UNIT_DEGREE            "%c{e}"; // Whit
automatic Temperature shown, ("</th><th>" ) is only for "formatierung" :)
const char HTTP_SNS_TEMP_USER[]     PROGMEM = "{s}%s "
   "{m}%s " D_UNIT_DEGREE            "%c{e}";   // Shows only the Username

The Result is on the Picture.

I will do that for DHT and a few other Sensors i use, and you can get the
stuff on Github: https://github.com/dieckfr/Custom-Tasmota-TEST-Github

Maybe take a while, as this time i dont know how to take the stuff to
github the best way.

Every Comment is welcome :)

mlavelle42, 2020-07-20:
Great start. Thank you.

ilgrank, 2020-07-20:
@dieckfr: thanks for the contribution. The correct place to publish your code tho is in a Pull request, not in an issue

dieckfr, 2020-07-20:
I have to study Github principals :) By the way, i have done a little bit work in formatting, and the DHT Sensor works also.

ilgrank, 2020-07-20:
That's great news @dieckfr !
All you have to do is go here:
[https://github.com/arendst/Tasmota/pulls](https://github.com/arendst/Tasmota/pulls)
and propose your code (if it is well documented / commented, the better)
If it causes no problems, and it is not too heavy it will probably be merged in the main Tasmota code :)

dieckfr, 2020-07-20:
Oh that is no stuff for a pull request, because its only for user compiled
bin files...
Grüsse aus dem Münsterland / Kind Regards

Franz-Josef Dieckmann
Brookstrasse 72
49497 Mettingen
Tel: 016096478553

MaxVonEvil, 2023-05-12:
Hey all, might be kicking a metabolically challenged horse here, yet just curious if mentioned ds18b20 alias ever made it into the main repo? I too have multiple  tempsensors bolted on to some devices, i.e. do my best to follow a logical naming convention for all entities, thus super helpful if this ever got implemented. Thank you //MvE

sfromis, 2023-05-13:
What you have now is the command `DS18Alias` allowing you to select a number, or you can do a build replacing the default non-fixed number by an id derived from the address. Due to the limited capacity of the ESP8266, Tasmota does not provide for defining string names for all sorts of sensors.

MaxVonEvil, 2023-05-15:
Is the DS18Alias perhaps in another build? 
I'm running Tas 12.5.0 and it does not recognize said command?

barbudor, 2023-05-15:
The code is there but not compiled in any standard build
You need to self compile with
```
#define DS18x20_USE_ID_ALIAS
```

PR #16649 (2022-09-29): Add support alias for ds18x20 sensors. 
## Description:

**Related issue (if applicable):** fixes #6970
ESP8266 and ESP32 support now.
Added command
`DS18Alias XXXXXXXXXXXXXXXX,N`
where XXXXXXXXXXXXXXXX **full** SN of sensor and N - number 1..255
If use command without arguments in result printing current **full** SN and name in JSON
After set aliases you can use this in scripts or rules.
Initial settings can be make from autoexec.bat or MQTT

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/f [... cut]
PR #19026 (2023-08-12): [DS18x20] Enhance use of aliases
## Description:

`#define DS18x20_USE_ID_AS_NAME`: Always show part of the address, even for one sensor 
`#define DS18x20_USE_ID_ALIAS`: The command `DS18Alias` can now be use with alphanumeric aliases, which replace the sensor name

**Related issue (if applicable):** fixes #6970 

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8 [... cut]