arendst/Tasmota #6971: Trouble with default IP parameters after flash sonoff basic R2

majherek, 2019-11-19:
Hi, 

I have trouble with default IP parameters after flash sonoff basic R2. 
I compile tasmota with:
```
#define WIFI_IP_ADDRESS        "192.166.20.3"
#define WIFI_GATEWAY           "192.168.20.254"
#define WIFI_SUBNETMASK        "255.255.255.0"
#define WIFI_DNS               "192.168.199.251"
```

After flashing it with Tasmota-PyFlasher:
```
Command: esptool.py --port COM3 --baud 115200 --after no_reset write_flash --flash_mode dout 0x00000 C:\Users\majherek\Documents\PlatformIO\Tasmota\.pioenvs\tasmota\firmware.bin --erase-all

esptool.py v2.8
Serial port COM3
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi, Embedded Flash
Crystal is 26MHz
MAC: dc:4f:22:e4:b6:a9
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Erasing flash (this may take a while)...
Chip erase completed successfully in 2.8s
Compressed 509344 bytes to 354856...
Wrote 509344 bytes (354856 compressed) at 0x00000000 in 31.5 seconds (effective 129.3 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.

Firmware successfully flashed. Unplug/replug or reset device
to switch back to normal boot mode.
```
Sonoff Basic use different ip addresses. It's look like it take them from DHCP server:
```
--- Miniterm on COM3  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
00:00:05 WIF: Connected
00:00:05 CMD: IPADDRESS1
00:00:05 RSL: stat/sonoff3/RESULT = {"IPAddress1":"192.166.20.3 (192.168.20.10)"}
00:00:06 MQT: Attempting connection...
00:00:16 MQT: TLS connection error: -1001
00:00:16 MQT: Connect failed to mqtt.majchrowski.waw.pl:8883, rc -2. Retry in 10 sec
00:00:16 CMD: IPADDRESS4
00:00:16 RSL: stat/sonoff3/RESULT = {"IPAddress4":"8.8.8.8"}
```
The address 192.168.20.10 become from my Cisco ASA 5505, which is dhcp server for this network (I have one ip address for lease). 

I change the configuration in asa to:
```
dhcpd address 192.168.20.40-192.168.20.40 atomix
dhcpd enable atomix
```
And now the IP address is 192.168.20.40. 
```
00:14:32 CMD: reset 2
00:14:32 RSL: stat/sonoff3/RESULT = {"Reset":"Erase, Reset and Restarting"}
00:14:33 CFG: Use defaults
00:14:35 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v482516e3
~ld
�␛␜r␘␂r�n␎#`␃␒�␇rn�␃�9�␂ph
00:00:00 CFG: Loaded from flash at FB, Count 1
00:00:00 Project tasmota Tasmota Version 7.0.0.5(tasmota)-2_6_1
00:00:00 WIF: Connecting to AP1 atomix in mode 11N as sonoff3...
00:00:04 CMD: IPADDRESS1
00:00:04 RSL: stat/sonoff3/RESULT = {"IPAddress1":"192.166.20.3 ((IP unset))"}
00:00:05 WIF: Connected
00:00:06 MQT: Attempting connection...
00:00:16 MQT: TLS connection error: -1001
00:00:16 MQT: Connect failed to mqtt.majchrowski.waw.pl:8883, rc -2. Retry in 10 sec
00:00:19 CMD: IPADDRESS1
00:00:19 RSL: stat/sonoff3/RESULT = {"IPAddress1":"192.166.20.3 (192.168.20.40)"}
00:00:27 MQT: Attempting connection...
```
A have 4 VLAN in home.
4 network. I route traffic between them.
IoT devices are in special VLAN atomix and network 192.168.20.0/24.
DNS server is different server. When I change IPADDRESS4 to 192.168.199.251 the DNS server respons for queries.
I can change IPADDERSS1 and 4, do restart 1 and everythink is working.
```
00:11:31 CMD: IPADDRESS4 192.168.199.251
00:11:31 RSL: stat/sonoff3/RESULT = {"IPAddress4":"192.168.199.251"}
00:11:31 MQT: Attempting connection...
00:11:41 MQT: TLS connection error: -1001
00:11:41 MQT: Connect failed to mqtt.majchrowski.waw.pl:8883, rc -2. Retry in 10 sec
00:11:41 CMD: IPADDRESS1 192.168.20.3
00:11:41 RSL: stat/sonoff3/RESULT = {"IPAddress1":"192.168.20.3 (192.168.20.40)"}
00:11:45 CMD: RESTART 1
00:11:45 RSL: stat/sonoff3/RESULT = {"Restart":"Restarting"}
00:11:47 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v482516e3
~ld
�␛␜r␘␂r�n␎#�ph
00:00:00 CFG: Loaded from flash at F6, Count 6
00:00:00 Project tasmota Tasmota Version 7.0.0.5(tasmota)-2_6_1
00:00:00 WIF: Connecting to AP1 atomix in mode 11N as sonoff3...
00:00:05 WIF: Connected
22:36:56 MQT: Attempting connection...
22:36:56 MQT: TLS connected in 633 ms, max ThunkStack used 3204
22:36:56 MQT: MFLN not supported by TLS server
22:36:56 MQT: Connected
22:36:56 MQT: tele/sonoff3/LWT = Online (retained)
22:36:56 MQT: cmnd/sonoff3/POWER =
22:36:57 MQT: tele/sonoff3/INFO1 = {"Module":"Sonoff Basic","Version":"7.0.0.5(tasmota)","FallbackTopic":"cmnd/DVES_E4B6A9_fb/","GroupTopic":"cmnd/sonoffs/"}
22:36:57 MQT: tele/sonoff3/INFO3 = {"RestartReason":"Software/System restart"}
22:36:57 MQT: stat/sonoff3/RESULT = {"POWER":"OFF"}
22:36:57 MQT: stat/sonoff3/POWER = OFF
22:36:57 MQT: stat/sonoff3/RESULT = {"POWER":"OFF"}
22:36:57 MQT: stat/sonoff3/POWER = OFF
```

Status 0:
```
22:06:08 CMD: STATUS 0
22:06:08 MQT: stat/sonoff3/STATUS = {"Status":{"Module":1,"FriendlyName":["Tasmota"],"Topic":"sonoff3","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:06:08 MQT: stat/sonoff3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff-ota.majchrowski.waw.pl:8888/api/arduino/tasmota.bin","RestartReason":"Power on","Uptime":"0T00:00:32","StartupUTC":"2019-11-19T21:05:36","Sleep":50,"CfgHolder":4817,"BootCount":3,"SaveCount":9,"SaveAddress":"FB000"}}
22:06:08 MQT: stat/sonoff3/STATUS2 = {"StatusFWR":{"Version":"7.0.0.5(tasmota)","BuildDateTime":"2019-11-18T22:05:43","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(bb83b9b)","Hardware":"ESP8285"}}
22:06:08 MQT: stat/sonoff3/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":0,"MqttLog":0,"SysLog":2,"LogHost":"192.168.199.251","LogPort":514,"SSId":["atomix",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
22:06:08 MQT: stat/sonoff3/STATUS4 = {"StatusMEM":{"ProgramSize":497,"Free":504,"Heap":28,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8F8821C6","040003A0","020207C7","000000C0","00000000"],"Drivers":"1,2,3,4,8,9,10","Sensors":"1,3,5,6,7,8,9,14,22"}}
22:06:09 MQT: stat/sonoff3/STATUS5 = {"StatusNET":{"Hostname":"sonoff3","IPAddress":"192.168.20.3","Gateway":"192.168.20.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.199.251","Mac":"DC:4F:22:E4:B6:A9","Webserver":0,"WifiConfig":4}}
22:06:09 MQT: stat/sonoff3/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.majchrowski.waw.pl","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E4B6A9","MqttUser":"sonoff","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
22:06:09 MQT: stat/sonoff3/STATUS7 = {"StatusTIM":{"UTC":"Tue Nov 19 21:06:09 2019","Local":"Tue Nov 19 22:06:09 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:03","Sunset":"15:38"}}
22:06:09 MQT: stat/sonoff3/STATUS10 = {"StatusSNS":{"Time":"2019-11-19T22:06:09"}}
22:06:09 MQT: stat/sonoff3/STATUS11 = {"StatusSTS":{"Time":"2019-11-19T22:06:09","Uptime":"0T00:00:33","UptimeSec":33,"Vcc":2.917,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"atomix","BSSId":"06:8D:DB:DB:97:47","Channel":11,"RSSI":64,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```

ascillato2, 2019-11-19:
You need to erase all flash before flashing. For support, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-19 by ascillato2

majherek, 2019-11-19:
I erase it with esptool.py at the beginning. 
```
sudo ./esptool.py  --port /dev/ttyUSB1  erase_flash
```

I alread post this to TAsmota Support Chat, Yesterday at 10:21 PM.

meingraham, 2019-11-19:
```
#define WIFI_IP_ADDRESS        "192.166.20.3"
#define WIFI_GATEWAY           "192.168.20.254"
#define WIFI_SUBNETMASK        "255.255.255.0"
#define WIFI_DNS               "192.168.199.251"
```

The device is on `192.166` and your gateway is on `192.168`.

majherek, 2019-11-19:
I'm blind. 

Thanks. 
Sorry for disturbing.

edwolfe807, 2019-12-09:
@majherek did that and it didn't work.

@meingraham not according to my_user.config.h

edwolfe807, 2019-12-09:
@ascillato2 Tried to post there but could not.  As I said to meingraham above, my_user_config.h file has 0.0.0.0 indicating use DHCP.  In addition, the ip showing by the ipaddress command shows 192.168.10.106 which is no where to be found in my configuration files. 

I didn't see it anywhere, but is there a serial command to determine the full DHCP information (like gateway, dns server, etc.) of the DNS server that responded?

Tommy-LSA, 2019-12-10:
Your Status 5 shows 192.168.10.1 as Gateway. So I would expect that IP address get set in this range?!

IPAddress":"192.168.10.106","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer":"172.32.0.250"