arendst/Tasmota #6972: Wrong default IP address in static configuration

majherek, 2019-11-19:
# BUG DESCRIPTION
Wrong default IP address in static configuration after flashing sonoff basic R2 which flash was erased earlier.

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic R2
- [x] Tasmota binary firmware version number used: 7.0.0.5
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: Visual Studio Code with PIO 
- [x] Flashing tools used: VSC with PIO/Tasmota-pyFlasher
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
   ```
   00:11:22 CMD: Backlog Template; Module; GPIO
   00:11:23 RSL: stat/sonoff3/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
   00:11:23 RSL: stat/sonoff3/RESULT = {"Module":{"1":"Sonoff Basic"}}
   00:11:23 RSL: stat/sonoff3/RESULT = {"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO14":{"0":"None"}}
  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

  ```
- [x] Provide the output of this command: ``Status 0``:
  ```
   00:07:53 CMD: STATUS 0
   00:07:53 RSL: stat/sonoff3/STATUS = {"Status":{"Module":1,"FriendlyName":["Tasmota"],"Topic":"sonoff3","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
   00:07:53 RSL: stat/sonoff3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff-ota.majchrowski.waw.pl:8888/api/arduino/tasmota.bin","RestartReason":"Power on","Uptime":"0T00:04:10","StartupUTC":"","Sleep":50,"CfgHolder":4817,"BootCount":1,"SaveCount":3,"SaveAddress":"F9000"}}
   00:07:53 RSL: stat/sonoff3/STATUS2 = {"StatusFWR":{"Version":"7.0.0.5(tasmota)","BuildDateTime":"2019-11-19T22:25:45","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(bb83b9b)","Hardware":"ESP8285"}}
   00:07:53 RSL: stat/sonoff3/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":0,"MqttLog":0,"SysLog":2,"LogHost":"192.168.199.251","LogPort":514,"SSId":["atomix",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
   00:07:53 RSL: stat/sonoff3/STATUS4 = {"StatusMEM":{"ProgramSize":497,"Free":504,"Heap":28,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8F8821C6","040003A0","020207C7","000000C0","00000000"],"Drivers":"1,2,3,4,8,9,10","Sensors":"1,3,5,6,7,8,9,14,22"}}
   00:07:53 RSL: stat/sonoff3/STATUS5 = {"StatusNET":{"Hostname":"sonoff3","IPAddress":"192.168.20.40","Gateway":"192.168.20.254","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"DC:4F:22:E4:B6:A9","Webserver":0,"WifiConfig":4}}
   00:07:53 RSL: stat/sonoff3/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.majchrowski.waw.pl","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E4B6A9","MqttUser":"sonoff","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
   00:07:53 RSL: stat/sonoff3/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:07:53 1970","Local":"Thu Jan 01 00:07:53 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"06:45","Sunset":"14:32"}}
   00:07:53 RSL: stat/sonoff3/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:07:53"}}
   00:07:53 RSL: stat/sonoff3/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:07:53","Uptime":"0T00:04:10","UptimeSec":250,"Vcc":3.385,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"atomix","BSSId":"06:8D:DB:DB:97:47","Channel":11,"RSSI":56,"LinkCount":1,"Downtime":"0T00:00:06"}}}
  ```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
   00:12:34 CMD: SerialLog 4
   00:12:34 RSL: stat/sonoff3/RESULT = {"SerialLog":{"4":{"Active":"4"}}}
   00:12:34 MQT: Attempting connection...
   00:12:44 MQT: TLS connection error: -1001
   00:12:44 MQT: Connect failed to mqtt.majchrowski.waw.pl:8883, rc -2. Retry in 10 sec
   00:12:44 CMD: reset 2
   00:12:44 SRC: Serial
   00:12:44 CMD: Group 0, Index 1, Command "RESET", Data "2"
   00:12:44 RSL: stat/sonoff3/RESULT = {"Reset":"Erase, Reset and Restarting"}
   00:12:44 CFG: Saved to flash at F8, Count 4, Bytes 4096
   00:12:44 APP: Erase 131 sectors
   00:12:46 CFG: Use defaults
   00:12:46 CFG: Saved to flash at FB, Count 1, Bytes 4096
   00:12:47 APP: Restarting

    ets Jan  8 2013,rst cause:1, boot mode:(3,7)

   load 0x4010f000, len 1384, room 16
   tail 8
   chksum 0x2d
   csum 0x2d
   v482516e3
   ~ld
   �␛␜r␘␂r�n␎#`␃␒�␇rn�␃�9�␂ph
   00:00:00 CFG: Loaded from flash at FB, Count 1
   00:00:00 Project tasmota Tasmota Version 7.0.0.5(tasmota)-2_6_1
   00:00:00 WIF: Connecting to AP1 atomix in mode 11N as sonoff3...
[... 134 lines cut ...]

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v482516e3
~ld
�␛␜r␘␂r�n␎#�ph
00:00:00 CFG: Loaded from flash at F6, Count 6
00:00:00 Project tasmota Tasmota Version 7.0.0.5(tasmota)-2_6_1
00:00:00 WIF: Connecting to AP1 atomix in mode 11N as sonoff3...
00:00:05 WIF: Connected
22:36:56 MQT: Attempting connection...
22:36:56 MQT: TLS connected in 633 ms, max ThunkStack used 3204
22:36:56 MQT: MFLN not supported by TLS server
22:36:56 MQT: Connected
22:36:56 MQT: tele/sonoff3/LWT = Online (retained)
22:36:56 MQT: cmnd/sonoff3/POWER =
22:36:57 MQT: tele/sonoff3/INFO1 = {"Module":"Sonoff Basic","Version":"7.0.0.5(tasmota)","FallbackTopic":"cmnd/DVES_E4B6A9_fb/","GroupTopic":"cmnd/sonoffs/"}
22:36:57 MQT: tele/sonoff3/INFO3 = {"RestartReason":"Software/System restart"}
22:36:57 MQT: stat/sonoff3/RESULT = {"POWER":"OFF"}
22:36:57 MQT: stat/sonoff3/POWER = OFF
22:36:57 MQT: stat/sonoff3/RESULT = {"POWER":"OFF"}
22:36:57 MQT: stat/sonoff3/POWER = OFF
```

majherek, 2019-11-19:
Sorry, someone show me my mistake.

majherek, 2019-11-19:
The IPADDRESS1 is from different network segment (192.166 in configuration). It shou;d be 192.168). Sorry for disturbing.

ascillato2, 2019-11-19:
Please, DO NOT POST DUPLICATED ISSUES. (https://github.com/arendst/Tasmota/issues/6971)

closed 2019-11-19 by ascillato2