arendst/Tasmota #6979: Relay assignment for ShutterRelay command

buenyamin-olgun, 2019-11-20:
**Have you looked for this feature in other issues and in the wiki?**  
Yes, the wiki explains my problem really clearly: 
"This relay's mate, the next higher numbered relay, closes the shutter." (Command: ShutterRelay<x>)

**Is your feature request related to a problem? Please describe.**  
I have flashed my device via Tuya-Convert & it worked really well up to this point.
The device I am talking about is this one:
https://www.amazon.de/gp/product/B07H9P57SZ

The device has 3 buttons: First one triggers "Up", second one triggers "Stop" and third one triggers "Down" of the shutters. Since I'm going to trigger this via MQTT and integrate it in my homebridge I need to get the current status of the blinds for which you added Shutter Support (thanks for your amazing work by the way!).

I've enabled the shutter feature which triggers Relay 1 to open the shutters (perfect!) and Relay 2 to to close the shutters (which stops the shutters in reality).

**Describe the solution you'd like**  
What I would need would be an assignment of ShutterRelay to a user-selected list of Relays, sth like ShutterRelay<x> 1,3

**Describe alternatives you've considered**  
I also tried to change the source code but I don't understand where Relay+1 is set for the second relay.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

buenyamin-olgun, 2019-11-20:
Advice on Discord:
Today, that device could probably make use of the built-in Blinds & Shutters driver as it stands today... but it wasn't in Tasmota when that template was contributed. The configuration would be quite different I suspect.

closed 2019-11-20 by buenyamin-olgun

calcipher49, 2019-11-21:
Hi, no new feature needed, just assign properly and add a rule.

You must assign physical buttons as follow:
Up button to Tasmota button1
Down button to Tasmota button2
In this way Shutter fonctionnality work properly.

Then assign physical Stop button to Tasmota button3 and define a Rule:
On Button3#State=On Do ShutterStop1 EndOn
So press this button will stop shutter

fennec622, 2019-12-22:
> Hi, no new feature needed, just assign properly and add a rule.
>
> You must assign physical buttons as follow:
> Up button to Tasmota button1
> Down button to Tasmota button2
> In this way Shutter [...]
Hi

I have same problem
How can i do for assign physical buttons to Tasmota button

my template is

{"NAME":"Tuya Shutter","GPIO":[157,0,54,10,22,19,0,0,17,21,53,23,52],"FLAG":0,"BASE":18}

cgatnet, 2020-08-03:
Same issue here. All works fine, but when using the slider, button2 is used for "close" direction. But button2 is "stop" button.
How to solve this? @calcipher49: how to reassign button2/3? 
Using same template as fennec622.

fennec622, 2020-08-03:
> Same issue here. All works fine, but when using the slider, button2 is used for "close" direction. But button2 is "stop" button.
> How to solve this? @calcipher49: how to reassign button2/3?
> Using same template as fennec622.

try this template

{"NAME":"Etersky","GPIO":[157,0,53,19,23,18,0,0,17,21,54,22,52],"FLAG":0,"BASE":18}