arendst/Tasmota #6981: change status ds18b20

michau86, 2019-11-21:
How to change the state of the relay on / off when the temperature (ds18b20 sensor) is above 35 degrees? what command

Tommy-LSA, 2019-11-21:
-> rules

Check Wiki, there is an example

michau86, 2019-11-21:
I put "rule1 on DS18B20_1#temperature>35 do power1 1 endon" but nothing changed

Tommy-LSA, 2019-11-21:
Type

Rule1

And post output here

michau86, 2019-11-21:
04:56:02 CMD: rule1
04:56:02 RSL: stat/sonoff/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":466,"Rules":"on DS18B20_1#temperature>30 do power7 1 endon"}

michau86, 2019-11-21:
04:58:07 CMD: rule1
04:58:07 RSL: stat/sonoff/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":466,"Rules":"on DS18B20_1#temperature>35 do power1 1 endon"}

Tommy-LSA, 2019-11-21:
Your Sensor shows values in the log? Please post the logoutput where the sensor goes through the 35

michau86, 2019-11-21:
what command can I see the logs for?

Tommy-LSA, 2019-11-21:
If your sensor properly works you‘ll see cyclic log entries for the sensor. Each time you see a log entry with „Sensor“ the rule get fired.

michau86, 2019-11-21:
05:30:17 RSL: tele/sonoff/STATE = {"Time":"2019-11-21T05:30:17","Uptime":"0T00:29:59","UptimeSec":1799,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"OFF","Wifi":{"AP":1,"SSId":"airlive_1","BSSId":"84:16:F9:9B:E9:CC","Channel":1,"RSSI":58,"LinkCount":1,"Downtime":"0T00:00:05"}}
05:30:17 RSL: tele/sonoff/SENSOR = {"Time":"2019-11-21T05:30:17","DS18B20":{"Temperature":34.5},"TempUnit":"C"}

michau86, 2019-11-21:
i  changed temperature to 30 but i don't see switching when temp is above

Tommy-LSA, 2019-11-21:
Change your rule from

on DS18B20_1#temperature>35 do power1 1 endon

into

on DS18B20#temperature>35 do power1 1 endon

You only have one sensor and you have to use the same sensor name as
shown in the log

Tommy-LSA, 2019-11-21:
I use a device with 2 sensors. Then tasmota shows the sensors
enumerated, but not if you have only one sensor.

05:44:03 MQT: tele/stadtweg/eg/heizung/SENSOR =
{"Time":"2019-11-21T05:44:03","DS18B20-1":{"Id":"04167492A1FF","Temperature":53.9},"DS18B20-2":{"Id":"0516735E7CFF","Temperature":35.6},"TempUnit":"C"}

closed 2019-11-21 by ascillato2