arendst/Tasmota #6982: Fix Arduino IDE support

arendst, 2019-11-21:
Currently Arduino IDE fails to compile. 

Fix Arduino IDE basic compile before release of 7.1

arendst, 2019-11-21:
Progress. It compiles but...

It all comes down to placing header files in the correct location.

arendst, 2019-11-21:
It looks like the IDE prototyping isn't working as expected.

On Tasmota 7.0.0.1 it does it correctly and starts adding prototypes after the last global definition:
```
#ifdef SUPPORT_IF_STATEMENT
  #include <LinkedList.h>
  LinkedList<String> backlog;               // Command backlog implemented with LinkedList
  #define BACKLOG_EMPTY (backlog.size() == 0)
#else
  uint8_t backlog_index = 0;                // Command backlog index
  uint8_t backlog_pointer = 0;              // Command backlog pointer
  String backlog[MAX_BACKLOG];              // Command backlog buffer
  #define BACKLOG_EMPTY (backlog_pointer == backlog_index)
#endif

/********************************************************************************************/

#line 186 "C:\\Users\\Theo\\Documents\\ArduinoESP8266\\Tasmota\\Tasmota-7.0.0.1\\tasmota\\tasmota.ino"
char * Format(char* output, const char* input, int size);
#line 224 "C:\\Users\\Theo\\Documents\\ArduinoESP8266\\Tasmota\\Tasmota-7.0.0.1\\tasmota\\tasmota.ino"
char * GetOtaUrl(char *otaurl, size_t otaurl_size);
#line 238 "C:\\Users\\Theo\\Documents\\ArduinoESP8266\\Tasmota\\Tasmota-7.0.0.1\\tasmota\\tasmota.ino"
char * GetTopic_P(char *stopic, uint32_t prefix, char *topic, const char* subtopic);
#line 293 "C:\\Users\\Theo\\Documents\\ArduinoESP8266\\Tasmota\\Tasmota-7.0.0.1\\tasmota\\tasmota.ino"
char * GetGroupTopic_P(char *stopic, const char* subtopic);
```
On later versions of Tasmota it places the prototypes too early in the code missing type definitions from both Tasmota and core:
```
// Location specific includes
#include <core_version.h>                   // Arduino_Esp8266 version information (ARDUINO_ESP8266_RELEASE and ARDUINO_ESP8266_RELEASE_2_3_0)
#include "tasmota_version.h"                // Tasmota version information
#include "tasmota.h"                        // Enumeration used in my_user_config.h
#include "my_user_config.h"                 // Fixed user configurable options
#line 185 "C:\\Users\\Theo\\Documents\\ArduinoESP8266\\Tasmota\\Tasmota-7.0.0.5\\tasmota\\tasmota.ino"
char * Format(char* output, const char* input, int size);
#line 223 "C:\\Users\\Theo\\Documents\\ArduinoESP8266\\Tasmota\\Tasmota-7.0.0.5\\tasmota\\tasmota.ino"
char * GetOtaUrl(char *otaurl, size_t otaurl_size);
#line 237 "C:\\Users\\Theo\\Documents\\ArduinoESP8266\\Tasmota\\Tasmota-7.0.0.5\\tasmota\\tasmota.ino"
char * GetTopic_P(char *stopic, uint32_t prefix, char *topic, const char* subtopic);
#line 292 "C:\\Users\\Theo\\Documents\\ArduinoESP8266\\Tasmota\\Tasmota-7.0.0.5\\tasmota\\tasmota.ino"
char * GetGroupTopic_P(char *stopic, const char* subtopic);
```
As a workaround I moved Tasmota type definitions in Tasmota.h including file settings.h and the core ESP8266WiFi.h.

It compiles that way but it is ugly and there must be a better way than restructuring the code into header and cpp files by me. 

I see no difference in both versions of Tasmota leading to the Arduino IDE placing the prototypes to early in the code...

arendst, 2019-11-21:
For now the FIRMWARE_BASIC and FIRMWARE_MINIMAL are compiling just fine.

Others fail during the "Generating function prototypes" phase.
As this is an Arduino IDE in combination with Tasmota issue I'll stop investigating and hand over to the IDE die-hards.

Jason2866, 2019-11-21:
For all users who dont want or cant install VSC.  Try this configured Portable VSC/Platformio version: https://github.com/Jason2866/Portable_VSC_PlatformIO
Just unpack to a (fast) extern drive and compile Tasmota without installing anything on computer

arendst, 2019-11-23:
Disabling #define USE_IR_REMOTE in my_user_config.h will result on succesful compilation.

As the IRremoteESP8266 library has been updated and tweaked recently I suspect the issue there.

I'll investigate.

arendst, 2019-11-24:
Can anyone explain to me why Arduino IDE prototyping fails on this line (33) in xdrv_05_irremote.ino

```
void (* const IrRemoteCommand[])(void) PROGMEM = { &CmndIrSend };
```
and works perfectly when changed to these lines
```
void (* const IrRemoteCommand[])(void) PROGMEM = {
  &CmndIrSend };
```
Which is the solution for compiling latest Tasmota's in Arduino IDE....

arendst, 2019-11-24:
Updated all possible Tasmota function arrays and verified correct Arduino IDE compilations

closed 2019-11-24 by arendst

arendst, 2019-11-24:
Background information here https://github.com/arduino/Arduino/issues/8309

Fixed in Arduino IDE 1.9.0