arendst/Tasmota #6989: DS2413 1-wire output?

btsimonh, 2019-11-21:
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

**Have you looked for this feature in other issues and in the wiki?**  
yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
I'd like to use a DS2413 1-wire GPIO because it offers up to 28v open collector outputs.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Support DS2413 for output (and input?)

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**  
Just a question, has anyone done this?  a Google search 'tasmota DS2413' comes up empty....
Pointers to a good addon starting point could result in a PR (did some work on tasmota a few years ago; but I'm not a little rusty!).

btsimonh, 2019-11-21:
or a generic 1-wire I/O implementation?  I'll be driving from node-red.....

ascillato2, 2019-11-21:
> I'll have a play once I've sorted getting latest tasmota on my device...

Cool. When you are ready, you can provide a PR. If you need help, you can find us in the Tasmota Support chat. Thanks.

closed 2019-11-21 by ascillato2

btsimonh, 2019-11-23:
Just got the latest running (lots of pain with dead and half dead ESP12s :( ).
I could not imagine how tasmota would move on in 2 years - it's just getting sweeter and sweeter!
Nice work, will now see If I can contribute.
Even got it to run with 3M of available flash space on Wemod1.

btsimonh, 2019-11-23:
ok, I did some work on this, but in actuality, I've gone for a simpler approach of using a couple of 5v relays I had.
However, Tasmota's relay pins are always in 'output' mode, and the 5v driven relay boards interpret the 'driven' 3.3v as 'low', and so the relay is always on.  I've added 16 more relays types, Relay<n>oc, Relay<n>oci. 
Basically, these set the pin to INPUT if the written value is != 0, and set the pin to OUTPUT if the written value is 0.
However, keeping in mind the memory constraints, these apparently simple mods add up to a fair few bytes, just in terms of new strings. :(.
There may be some saving to be made in code; but to save on the strings would be more difficult without reworking the drivers list.
What we could consider would be a generic configuration for relays; then the names could be removed, and replaced by a byte of flags in the config representing the GPIO mode for relay drive?
Stopping for today, as I now have an operable window opening system.... (yes, I should investigate the new shutters stuff!)

Thoughts on output pin modes?