arendst/Tasmota #6990: SetOption68 1 Channel control not working when using only one channel

Povilas1, 2019-11-21:
PWM Dimmer does not work when setting Option68 to 1 and only one PWM output enabled. I can only on / off, but no channel control when enabled only one PWM output with no relays.

21:37:14 CMD: Channel1 50
21:37:14 RSL: stat/sonoff/RESULT = {"POWER1":"ON","Channel1":100,"Color":"FF"}

s-hadinger, 2019-11-21:
I don't dare asking why you want SetOption68 1 with only one channel. 

I will take a look.

ascillato2, 2019-11-21:
SetOption68 is for Multi-channel PWM.

Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-21 by ascillato2

Povilas1, 2019-11-21:
Currently my ESP is wired to only one light. Today I discovered in Option68 and wanted to try it out before making any changes to the hardware side :D

ascillato2, 2019-11-21:
Which Tasmota version are you using?

Povilas1, 2019-11-22:
@ascillato2 Sonoff-Tasmota 6.7.1

ascillato2, 2019-11-22:
ah, ok, would you be able to test on latest version from development branch: http://thehackbox.org/tasmota/tasmota.bin ? Thanks

s-hadinger, 2019-11-24:
Fixes in #7020

PR #7020 (2019-11-24): Multiple fixes in light
## Description:

This PR fixes the following:

- `SetOption68 1` with only 1 channel now working
- `SetOption37 128` with 4 or 5 PWM, status now reports 2 Power and 2 Dimmers instead of only 1.
- `SetOption37 128`: now `Dimmer`, `HSBColor`, `CT` only change the corresponding `Power`
- `Channel4/5` now disables RGB if `SetOption37 < 128`

**Related issue (if applicable):** fixes #6990 #6799 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code [... cut]