arendst/Tasmota #6992: How to disconnect damaged button?

kayser21, 2019-11-21:
I have a smart relay with Tasmota.
Once I dropped it and the button is damaged.

Few minutes after I turn it on power goes off automatically.

Maybe this damaged button is the reason.
Is there any way to make this button not-active?
[image]
Przyci1 means Button1

  Configuration output here:
{"NAME":"SmartDGM Plug","GPIO":[0,0,0,17,134,132,0,0,131,56,21,0,0],"FLAG":0,"BASE":18}

  STATUS 0 output here:
https://pastebin.com/kahEkbVA

  Console output here:
log is the same before and after "POWER OFF"

ascillato, 2019-11-21:
Did you tried to change button1 to none?

ascillato2, 2019-11-21:
If not, please try it. If that don't work, please complete the troubleshooting template in order to have more debug information or address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-21 by ascillato2

kayser21, 2019-11-21:
> Did you tried to change button1 to none?

Yes, it was my first try ;) I will try with the Chat. Thank you