arendst/Tasmota #6997: Damned Rules not working!!

frablaser, 2019-11-22:
> **GUIDE**
>
> This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
>
> Please DO NOT OPEN AN ISSUE:
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/
>  - If you have a issue when flashing was done via Tuya Convert
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Tasmota/wiki/FAQ) and troubleshooting wiki articles
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
_A clear and concise description of what the bug is._
Using IF ELSE works in console but it don't work properly in the rules. 
I have a Wemos D1 Mini used to light a walk-in light. The light open before sunrise, after a IR motion detection only if the door is closed (to avoid to wake up the person staying in the bed)... But after Sunrise, it light at the door opening.

IO: 
1- touch switch (button)
2- magnet switch (door open or closed)
3- IR Motion Sensor 

To initialize switches... 
`Rule1 on power2#boot do power2 0 endon on power3#boot do power3 1 endon`

```
Rule2
on power2#state=1 do backlog var1 1; RuleTimer1 1 endon
on power2#state=0 do backlog var1 0 endon
on power3#state=1 do backlog var2 1;RuleTimer1 1 endon
on power3#state=0 do backlog var2 0; IF((Time - Sunset)<0 OR Time>Sunrise) power1 1 ELSE power1 0 ENDIF endon
on Rules#Timer=1 do backlog var3 = var1 * var2; IF(var3==1 AND ((Time - Sunset)>0 OR Time<Sunrise)) power1 1; RuleTimer2 30 ELSE power1 0 endon
on Rules#Timer=2 do backlog power1 0 endon
```
If the Var1 and Var2 are initialized at 1 and I enter in the console: 
var3 = var1 * var2; IF(var3==1 AND ((Time - Sunset)>0 OR Time<Sunrise)) power1 1; RuleTimer2 30 ELSE power1 0
It works like a charm! But inner the rule, it fail all the way...

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., ): _____Wemos D1 mini
- [ x] Tasmota binary firmware version number used: _____7.0.0.5
       Version Core/SDK | 2_6_1/2.2.2-dev(bb83b9b)
  - [ ] Pre-compiled
  - [ x] Self-compiled
    - [x ] IDE / Compiler used: _____VS Code Plateform IO
- [ x] Flashing tools used: _____
- [x ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
1- touch switch (button)
2- magnet switch (door open or closed)
3- IR Motion Sensor 

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
Rule2
on power2#state=1 do backlog var1 1; RuleTimer1 1 endon
on power2#state=0 do backlog var1 0 endon
on power3#state=1 do backlog var2 1;RuleTimer1 1 endon
on power3#state=0 do backlog var2 0; IF((Time - Sunset)<0 OR Time>Sunrise) power1 1 ELSE power1 0 ENDIF endon
on Rules#Timer=1 do backlog var3 = var1 * var2; IF(var3==1 AND ((Time - Sunset)>0 OR Time<Sunrise)) power1 1; RuleTimer2 30 ELSE power1 0 endon
on Rules#Timer=2 do backlog power1 0 endon

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
11:22:11 RSL: stat/tasmota/RESULT = {"Var2":"0"}
11:22:11 RSL: stat/tasmota/RESULT = {"If":"Done"}
11:22:15 RSL: stat/tasmota/RESULT = {"POWER3":"ON"}
11:22:15 RSL: stat/tasmota/POWER3 = ON
11:22:15 RUL: POWER3#STATE=1 performs "backlog var2 1;RuleTimer1 1"
11:22:16 RSL: stat/tasmota/RESULT = {"Var2":"1"}
11:22:16 RSL: stat/tasmota/RESULT = {"T1":1,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
11:22:17 RUL: RULES#TIMER=1 performs "backlog var3 = var1 * var2; IF(var3==1 AND ((Time - Sunset)>0 OR Time<Sunrise)) power1 1; RuleTimer2 30 ELSE power1 0"
11:22:17 RSL: stat/tasmota/RESULT = {"Var3":"0.000"}
11:22:18 RSL: stat/tasmota/RESULT = {"If":"Done"}
11:22:18 RSL: stat/tasmota/RESULT = {"T1":0,"T2":30,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
11:22:24 RSL: stat/tasmota/RESULT = {"POWER2":"ON"}
11:22:24 RSL: stat/tasmota/POWER2 = ON
11:22:24 RUL: POWER2#STATE=1 performs "backlog var1 1; RuleTimer1 1"
11:22:24 RSL: stat/tasmota/RESULT = {"Var1":"1"}
11:22:24 RSL: stat/tasmota/RESULT = {"T1":0,"T2":23,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
11:22:26 RUL: RULES#TIMER=1 performs "backlog var3 = var1 * var2; IF(var3==1 AND ((Time - Sunset)>0 OR Time<Sunrise)) power1 1; RuleTimer2 30 ELSE power1 0"
11:22:26 RSL: stat/tasmota/RESULT = {"Var3":"1.000"}
11:22:27 RSL: stat/tasmota/RESULT = {"If":"Done"}
11:22:27 RSL: stat/tasmota/RESULT = {"T1":0,"T2":30,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}

  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Follow the text over

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Follow the text over

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2019-11-22:
Lots of information but critical information requested by the issue template is missing. You have mentioned only input IO but yet your rules are using output commands (power) and triggers.

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

frablaser, 2019-11-22:
> template is missing

I did it on the two screenshots but, I don't know why git don't show them... 🙄

meingraham, 2019-11-22:
Status 0
Template
Module
GPIO
The actual output from the rules commands

That's all missing

In any case - the issue is with your configuration and your incorrect rules logic/syntax.

Please use Discord for these kinds of questions.

closed 2019-11-22 by ascillato2

frablaser, 2019-11-22:
Shit Adrian you're fast to close everything!!! Wow! Next time I will close it before have an answer to be sure you will be satisfied! 😂😂😂

frablaser, 2019-11-22:
Mr. Ingraham, 
The template is here:
[Template GPIO](https://user-images.githubusercontent.com/29067539/69446768-21c81a80-0d23-11ea-89dc-04ee2f00bbe1.png)
Module is Wemos D1 Mini Generic 18 (in the text description but could be any ESP 8266 the )

Regards

frablaser, 2019-11-22:
Ok... then I give up... After two days of loosing my time on it. I let it go. Rules are not ready to work in the way I need it...  I will use the HA automation, just to bad, Tasmota isn't enough mature yet. 

I tried to help for debugging the development , but there is no way to talk... 
Then thanks again!

meingraham, 2019-11-22:
> Tasmota isn't enough mature yet.

:joy: :joy: :joy: :joy: :joy: :joy: :joy: :joy: :joy: :joy: :joy: :joy: :joy: :joy: :joy: :joy: :joy: :joy: :joy: :joy: :joy:

frablaser, 2019-11-22:
For the rules... 😂

meingraham, 2019-11-22:
@frablaser 

At the minimum, you're missing an endif. I'll let you start there.

frablaser, 2019-11-22:
Yes... at the end of the on Rules#Timer=1 but even that, I tested it directly in the console (with the ENDIF at the end as well), and it worked like a charm, but inner the rule, it doesn't trigger the power1 1 and start the second Timer... even if the IF conditions is false... 🙄

frablaser, 2019-11-22:
It seems that the rule timers do a thread where the IF doesn't fit...

meingraham, 2019-11-22:
> If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue.

Are you?

I'm fairly certain that `IF` inside a `Backlog` is not supported.

If I were you, I would send a strongly worded letter and ask for a **full refund**! Tasmota is unacceptable.

frablaser, 2019-11-22:
Thanks again for your answer Mr Ingraham,

My version date less than a week I think...

Version du programme | 7.0.0.5(tasmota)
Date & Heure de build | 2019-11-21T09:55:59
Version Core/SDK | 2_6_1/2.2.2-dev(bb83b9b)

meingraham, 2019-11-22:
This is the reason Status 0 is requested.

Alter you rule to separate `Backlog` and `IF` - i.e., repeat the trigger.

frablaser, 2019-11-22:
I removed all the Backlog instructions...

```
Rule2
on power2#state=1 do var1 1;RuleTimer1 1 endon
on power2#state=0 do var1 0 endon
on power3#state=1 do var2 1;RuleTimer1 1 endon
on power3#state=0 do var2 0; IF((Time - Sunset)<0 OR Time>Sunrise) power1 1 ELSE power1 0 ENDIF endon
on Rules#Timer=1 do var3 = var1 * var2; IF(var3==1 AND ((Time - Sunset)>0 OR Time<Sunrise)) power1 1; RuleTimer2 30 ENDIF; IF(var3!=1) power1 0 ENDIF endon
on Rules#Timer=2 do power1 0 endon
```
But it result of... now my var1 include the IF... that's a separation fault. It don't handle the semicolon
RSL: stat/tasmota/RESULT = {"Var2":"0; IF((Time - Sunset)<0 OR Time>"} 

> 15:01:45 RUL: POWER3#STATE=0 performs "var2 0; IF((Time - Sunset)<0 OR Time>Sunrise) power1 1 ELSE power1 0 ENDIF"
> 15:01:45 RSL: stat/tasmota/RESULT = {"Var2":"0; IF((Time - Sunset)<0 OR Time>"}
> 15:01:46 RSL: tele/tasmota/STATE = {"Time":"2019-11-22T15:01:46","Uptime":"0T21:40:06","UptimeSec":78006,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"ON","POWER3":"OFF","Wifi":{"AP":1,"SSId":"WaxHome_2G","BSSId":"50:D4:F7:37:AB:CB","Channel":2,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:06"}}
> 15:01:46 RSL: tele/tasmota/SENSOR = {"Time":"2019-11-22T15:01:46","Switch2":"ON","Switch3":"ON"}
> 15:02:21 CMD: var2
> 15:02:21 RSL: stat/tasmota/RESULT = {"Var2":"0; IF((Time - Sunset)<0 OR Time>"}

frablaser, 2019-11-22:
If I put that IF in the console... It works! Yhea!

15:13:56 CMD: IF((Time - Sunset)<0 OR (Time>Sunrise AND Time<Sunset)) power1 1 ELSE power1 0 ENDIF
15:13:56 RSL: stat/tasmota/RESULT = {"If":"Done"}
15:13:56 RSL: stat/tasmota/RESULT = {"POWER1":"ON"}
15:13:56 RSL: stat/tasmota/POWER1 = ON

meingraham, 2019-11-22:
>  i.e., repeat the trigger.

You did not.

```
on power3#state=0 do backlog var2 0; IF((Time - Sunset)<0 OR Time>Sunrise) power1 1 ELSE power1 0 ENDIF endon
```

becomes
```
on power3#state=0 do var2 0 endon
on power3#state=0 do IF((Time - Sunset)<0 OR Time>Sunrise) power1 1 ELSE power1 0 ENDIF endon
```

and
```
on Rules#Timer=1 do backlog var3 = var1 * var2; IF(var3==1 AND ((Time - Sunset)>0 OR Time<Sunrise)) power1 1; RuleTimer2 30 ELSE power1 0 endon
```

becomes
```
on Rules#Timer=1 do var3 = var1 * var2 endon
on Rules#Timer=1 do IF(var3==1 AND ((Time - Sunset)>0 OR Time<Sunrise)) power1 1; RuleTimer2 30 ELSE power1 0 endon
```

frablaser, 2019-11-22:
Thanks Mr Ingraham for your answer,

I did'nt know that it was possible to track the same trigger many times... 
But I'm bumping into another problem... 🙄 Now I reached the limit of a rule string length I think... ?!
If I try to insert it in a whole line as :
on power2#state=1 do var1 1
on power2#state=1 do RuleTimer1 1 endon
on power2#state=0 do var1 0 endon
on power3#state=1 do var2 1 
on power3#state=1 do RuleTimer1 1 endon
on power3#state=0 do var2 0
on power3#state=0 do IF((Time - Sunset)<0 OR (Time>Sunrise AND Time<Sunset)) power1 1 ELSE power1 0 ENDIF endon
on Rules#Timer=1 do var3 = var1 * var2
on Rules#Timer=1 do IF(var3==1 AND ((Time - Sunset)>0 OR Time<Sunrise)) power1 1; RuleTimer2 30 ENDIF; IF(var3!=1) power1 0 ENDIF endon
on Rules#Timer=2 do power1 0 endon

It don't insert anything... Then I tried to using the append rule instruction as (+) like:
Rule2 on power2#boot do power2 0 endon on power3#boot do power3 1 endon
Rule2 + on power2#state=1 do var1 1
Rule2 + on power2#state=1 do RuleTimer1 1 endon
Rule2 + on power2#state=0 do var1 0 endon
Rule2 + on power3#state=1 do var2 1 
Rule2 + on power3#state=1 do RuleTimer1 1 endon
Rule2 + on power3#state=0 do var2 0
Rule2 + on power3#state=0 do IF((Time - Sunset)<0 OR (Time>Sunrise AND Time<Sunset)) power1 1 ELSE power1 0 ENDIF endon
Rule2 + on Rules#Timer=1 do var3 = var1 * var2
Rule2 + on Rules#Timer=1 do IF(var3==1 AND ((Time - Sunset)>0 OR Time<Sunrise)) power1 1; RuleTimer2 30 ENDIF; IF(var3!=1) power1 0 ENDIF endon
Rule2 + on Rules#Timer=2 do power1 0 endon

It works until the third last line... 

My question would be, can I insert Rule Timers into an other rule(x) entry? And could it be triggered by an other rule?

frablaser, 2019-11-22:
I corrected my endon missing at the end in my previous splitting...
Ok I tried to split it into two rule (2-3) to fit on the maximum string length of rules as: 
```

Rule2 on power2#state=1 do var1 1 endon on power2#state=1 do RuleTimer1 1 endon on power2#state=0 do var1 0 endon on power3#state=1 do var2 1 endon on power3#state=1 do RuleTimer1 1 endon on power3#state=0 do var2 0 endon on power3#state=0 do IF((Time - Sunset)<0 OR (Time>Sunrise AND Time<Sunset)) power1 1 ELSE power1 0 ENDIF endon on Rules#Timer=1 do var3 = var1 * var2 endon 
Rule3 on Rules#Timer=1 do IF(var3==1 AND ((Time - Sunset)>0 OR Time<Sunrise)) power1 1; RuleTimer2 30 ENDIF; IF(var3!=1) power1 0 ENDIF endon on Rules#Timer=2 do power1 0 endon
```

But I got the answer... Rules#Timer called from rule2 are not triggered in rule3... then I face again in a deep black hole! 

Can I ask for a refund??

meingraham, 2019-11-22:
This replicates your logic. It shouldn't matter which rule set a particular rule is in (e.g., RuleTimer2)

1. Clears everything.
2. The rules.
3. Enable rules sets.

```
Backlog Rule1 ""; Rule2 ""; Rule3 ""; Rule1 0; Rule2 0; Rule3 0

Rule1
  on power2#boot do power2 0 endon
  on power3#boot do power3 1 endon
  on power2#state do var1 %value% endon
  on power2#state=1 do RuleTimer1 1 endon
  on power3#state do var2 %value% endon
  on power3#state=1 do RuleTimer1 1 endon
  on power3#state=0 do IF((Time-Sunset)<0 OR (Time>Sunrise AND Time<Sunset)) power1 1 ELSE power1 0 ENDIF endon

Rule2
  on Rules#Timer=1 do var3=var1*var2
  on Rules#Timer=1 do IF(var3==1 AND ((Time-Sunset)>0 OR Time<Sunrise)) power1 1; RuleTimer2 30 ENDIF; IF(var3!=1) power1 0 ENDIF endon
  on Rules#Timer=2 do power1 0 endon

Backlog Rule1 1; Rule2 1
```

frablaser, 2019-11-22:
UPDATE!!!
I put my Rule3 at ON and now it works!!!!

Thanks Mr Ingraham for your help. It's very appreciated.

And the good thing, no refund is needed! 😂😂😂

arendst, 2019-11-22:
A donation would do

frablaser, 2019-11-25:
It is what I did! 😉

arendst, 2019-11-25:
Relating tags to names keeps being a challange but I think I digged it. 👍