arendst/Tasmota #7003: Add Iskra MT 174 to SmartMeter Descriptor Wiki

kugelkopf123, 2019-11-23:
Add Example for Iskra MT 174 Smart Meter to the Wiki Smart Meter Descriptor.
Its the first example with a two way communication. I would have add it by myself but it isnt possible anymore to edit it. 

```
; Iskra MT 174
>D
>B
=>sensor53 r
>M 1
+1,3,o,0,300,STROM,1,100,2F3F210D0A
1,1-0:1.8.1*255(@1,Total Verbrauch,KWh,Total_in,3
1,1-0:2.8.1*255(@1,Total Verkauf,KWh,Total_in,3
1,1-0:0.0.0*255(@#),Zähler Nr,,Meter_number,0
#
```

blakadder, 2019-11-23:
Will do! It will be visible in the new documentation once its unveiled

closed 2019-11-23 by kugelkopf123

kugelkopf123, 2019-11-24:
Found a bug. It was 2 times the same MQTT Message "Total_in". It has to be "Total_in" and Total_out"
```
>D
>B
=>sensor53 r
>M 1
+1,3,o,0,300,STROM,1,100,2F3F210D0A
1,1-0:1.8.1*255(@1,Total Verbrauch,KWh,Total_in,3
1,1-0:2.8.1*255(@1,Total Verkauf,KWh,Total_out,3
1,1-0:0.0.0*255(@#),Zähler Nr,,Meter_number,0
#
```

meingraham, 2019-11-24:
OK, I updated the (new) wiki

MaxDaNr1, 2023-03-29:
hey, in order to use more Data from Iskra MT147 I added some more lines to the Script:

>D
>B
->sensor53 r
>M 1
+1,3,o,0,300,STROM,1,100,2F3F210D0A
1,1-0:1.7.0*255(@1,Verbrauch,KW,consume,3
1,1-0:1.8.0*255(@1,Total Consumed,KWh,Total_in,3
1,1-0:1.8.1*255(@1,Total Consumed Day,KWh,Total_in_day,3
1,1-0:1.8.2*255(@1,Total Consumed Night,KWh,Total_in_night,3
1,1-0:2.7.0*255(@1,Einspeisung,KW,feed,3
1,1-0:2.8.0*255(@1,Total Delivered,KWh,Total_out,3
1,1-0:0.0.0*255(@#),Meter Number,,Meter_number,0
#

I especially wanted to know what my current power consumption is. Is it possible to increase the data rate?

christophhhh, 2023-05-19:
I have the same meter (MT174)and would like to speed up the communication.
However, I can't get it to work properly. Was anyoine successfull with this meter?

Here is my script:

``
>D
; Script needs v12+
res=0
scnt=0
>B
=>sensor53 r
>F
;F section is executed every 100ms
scnt+=1
switch scnt
case 1
;set sml driver to 300 baud and send /?! as HEX to trigger the Meter
res=sml(1 0 300)
res=sml(1 1 "2F3F210D0A")
;500ms later \> Send ACK and ask for switching to 9600 baud
case 5
res=sml(1 1 "06300500D0A")
res=sml(1 0 9600)
case 100
; 10s later \> restart sequence
scnt=0
ends  
>M 1
+1,3,o,0,9600,Strom,1
1,1-0:1.7.0*255(@1,Bezug aktuell,KW,P_in,3
1,1-0:2.7.0*255(@1,Einspeisung aktuell,KW,P_out,3
1,1-0:1.8.0*255(@1,Bezug total,KWh,Total_in,3
1,1-0:1.8.1*255(@1,Bezug total Tag,KWh,Total_in_day,3
1,1-0:1.8.2*255(@1,Bezug total Nacht,KWh,Total_in_night,3
1,1-0:2.8.0*255(@1,Einspeisung total,KWh,Total_out,3
1,=m 1-2 @1,Leistung,KW,P_abs,3
1,1-0:0.0.0*255(@#),Meter Number,,Meter_number,0
#``

MaxDaNr1, 2023-05-19:
I Have the same problem. Here is what I already tried: https://github.com/arendst/Tasmota/discussions/18332#discussioncomment-5602783
I still have no solution. Still just fresh data every ~5min.