arendst/Tasmota #7004: Add support for a different Parity to support more Modbus Smartmeter

kugelkopf123, 2019-11-23:
It would be nice to use the SBC ALE3 Modbus SmartMeter. But its using another parity than the current SML driver accept. As far as i can see, this is the Problem.
I've tried 8E1 on my Raspberrypi with this SBC Smart Meter and it worked. 
It would be nice to choose another parity (more than just 7E1 and 8N1) within this Driver to support more Meter. Just like you can switch the Baudrate.

Link Modbus SmartMeter from SBC: 
[https://www.sbc-support.com/uploads/tx_srcproducts/26-527_ENG_DS_EnergyMeter-ALE3-with-Modbus_01.pdf](SBC ALE3 Modbus)

Library I tried on my Raspberry Pi:
[https://github.com/volkszaehler/mbmd/blob/master/meters/rs485/sbc.go](https://github.com/volkszaehler/mbmd/blob/master/meters/rs485/sbc.go)

Thanks!

meingraham, 2019-11-23:
@gemu2015 ⬆️

s-hadinger, 2019-11-24:
TasmotaSerial currently does not support parity at all. It only supports 8N1 or 8N2.

7E1 is emulated by the SML driver by calculating the 8th bit, but TasmotaSerial just passes the 8 bits. This would require TasmotaSerial to send an additional parity bit, but we are already very short in IRAM.

@arendst and @gemu2015 any idea?

arendst, 2019-11-24:
It's also a matter of timing. Adding a test for parity screws current timing unless for every parity a seperate handler is used. This increases code size considerably.

Alternative is using hardware serial only. That could work with less code.

In any case it's back to sqair one and test and measure.

arendst, 2019-11-24:
We might reconsider TasmotaSerial based on https://github.com/plerup/espsoftwareserial again

s-hadinger, 2019-11-24:
I was thinking of only sending parity but not testing parity on receive. Selecting 8N2 will silently ignore the received parity.

Send might be easier to change without IRAM impact. What is the baud rate for Modbus?

kugelkopf123, 2019-11-24:
As far as I know, there is no special baudrate for Modbus. It could be from 4800 to 19200. Typically its 9600baud.

gemu2015, 2019-11-25:
i implemented 8E1 for my sml driver (on my fork) for hardware serial.
i am waiting for confirmation by @kugelkopf123 to to a pr.

s-hadinger, 2019-11-25:
9600 baud is slow enough and shouldn't require that the code sending serial is in IRAM. Hardware serial is still the preferred option.

closed 2019-11-25 by arendst

andig, 2020-02-23:
Hi, I'm author of mbmd linked above. I've just tried connecting an ALE3 meter to mbmd and failed- device does not answer. Did you experience any problem talking to ALE3?

nikito7, 2021-03-29:
My meter use 8o2

gemu2015, 2021-03-30:
added full serial config option to modbus

usage:

after type m
+1,3,m,0,9600,*,1,2,01040000,01040002,01040004

you may specify
N =no parity
E =even parity
O =odd parity

and number of stop-bits

mN1
mN2
mE1
mE2
mO1
mO2

e.g
+1,3,mO2,0,9600,*,1,2,01040000,01040002,01040004

nikito7, 2021-03-30:
Needs Hardware Serial?

gemu2015, 2021-03-30:
Yes only hardware serial

nikito7, 2021-03-30:
Working. ESP32.

Thank you

RaphaelSquair, 2026-10-20:
@gemu2015 

 hi I know this is a rerally old post , but you mentioned that it would be possible to use 8N2  on modbus with Software serrial. How would i Configure this in the script? i tried using m2 as other posts suggested but its not working and loking at line 4880 on the file xsns_53  
   if (mp->meter_ss->begin(mp->params)) { 

This look like the call that start the software serial for esp8266 but it don't appear to be calling the smode that is configured with the m2 command in the script ( smode = SERIAL_8N2; )  shouldn't this parameter be passed to the ss->begin?

gemu2015, 2026-10-20:
Hi @RaphaelSquair — you read it right, that is the gap. smode (set by m2 / mN2) is computed in the SML driver, but on the ESP8266 the software serial is started without it, so it always runs 8N1; only the hardware UART gets the config via Serial.begin(...).

Since you compile yourself, please try this one-line change in xsns_53_sml.ino, in the #ifdef ESP8266 block just after smode is worked out (around line 4880 in your copy):

#ifdef ESP8266
-       if (mp->meter_ss->begin(mp->params)) {
+       if (mp->meter_ss->begin(mp->params, smode)) {
          mp->meter_ss->flush();
        }

TasmotaSerial::begin(speed, config) takes the stop bits, so m2 / mN2 should then give you 8N2 on software serial. Note that TasmotaSerial has no real parity: E and O are only emulated with an extra stop bit, so for 8N2 Modbus this is exactly what you need, for 8E1/8O1 meters it may behave differently from before.

RaphaelSquair, 2026-10-20:
Hi @gemu2015  This worked perfectly, i can see in t he osciloscope the correct behavior with the 8N2 vs the 8N1 default behavior.  could this solution you gave   if (mp->meter_ss->begin(mp->params, smode)) { be applied in a future PR? I think other users could have similar issue, manly when comunicating with old non standard modbus devices.

PR #7029 (2019-11-25): sml support for 8e1
## Description:

support for hardware serial 8E1

**Related issue (if applicable):** fixes #7004

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.6.1
  - [ ] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).
PR #11514 (2021-03-30): Fixes
## Description:

**Related issue (if applicable):** fixes #11486
**Related issue (if applicable):** fixes #7004

fixes MLX90614 erratic readings
SML better modbus serial config
fix scripter executing restart vector on boot

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with core ESP32 V.1.0.6
  - [x [... cut]