arendst/Tasmota #7005: platformio.ini override

bvduijn, 2019-11-23:
**Have you looked for this feature in other issues and in the wiki?**  
yup

**Is your feature request related to a problem? Please describe.**  
No

**Describe the solution you'd like**  
Add an option to override the default platformio.ini to retain user specific settings

**Describe alternatives you've considered**  
Modify the file everytime i do a pull from the master repo

**Additional context**  
If you could add a line in the platformio.ini like:

```
[platformio]
extra_configs = platformio_override.ini

```
This file could function the same way as user_config_override.h. Put in .gitignore and this way userspecific settings could remain in that file. Settings like:
- default_envs
- build_flags (-DUSE_CONFIG_OVERRIDE
- upload_port
- ... any other settings 

Kind regards,
Bert

Jason2866, 2019-11-23:
@bvduijn  I like the idea. Please make a PR for

ascillato2, 2019-11-24:
Closing as PR #7014 was proposed

closed 2019-11-24 by ascillato2