arendst/Tasmota #7006: stefanbode - Stepper Motor Shutter Control

meingraham, 2019-11-23:
@stefanbode 

Is it possible to set up the shutter driver to work with the [Stepper Motor driver](https://github.com/arendst/Tasmota/wiki/A4988-Stepper-Motor-Controller)? The driver uses either a  A4988 or a DRV8825. My thought is that whereas your driver expects two relay components, the stepper uses rotation direction and on/off which could correspond to the two relays. The stepping increment could be "static" and set via [MotorMIS](Commands#motormis). Due to the way stepper motors shudder, power enabled would need to be managed as well. Perhaps it would be the second relay instead of motor on/off and instead the setup would drive motor on/off always on (e.g., a third relay always powered on)?

I'm piggybacking on @TrDA-hab's  (TrDA on Discord) personal project he's built to control his roller shutter. I just wanted to put the two of you in touch since you're not on Discord. I think this is an opportunity to provide an working solution using both your driver and the stepper motor driver.

Cheers!

Mike

titomontes, 2019-11-26:
I stumbled upon the conversation, very interesting stuff. Hope @stefanbode sees this.

titomontes, 2019-11-26:
[https://media.discordapp.net/attachments/526796380316893185/647515427429154858/GT2-1_v26.jpg?width=953&height=481](url)

stefanbode, 2019-11-26:
> I stumbled upon the conversation, very interesting stuff. Hope @stefanbode sees this.

Come on this is really over-engineered. You really want to operate a Schutter with a Stepper Motor?

meingraham, 2019-11-26:
I've seen many DIY shutters on the Internet that use steppers. Do, from what I have seen, yes. And from these two additional data points, ...

stefanbode, 2019-11-26:
Ok maybe I can give it a try. But how does the current driver behave if i change the position during movement. I assume this is all in sync whatever you do. Are there limits or can I use unsigned long variables for the position. This might the easiest way to implement.  Internally I already have something that I can use for the position command.

meingraham, 2019-11-26:
@TrDA-hab ⬆️

stefanbode, 2019-11-26:
Is there an option to get the current position of the stepper. For example how to stop it. Power off is maybe not a good idea. I assume it is not self locking. I saw the wiki with some commands but no info what and how stuff is reported back. Look I need to assemble one and give it a try to see what happens

meingraham, 2019-11-26:
Adding @Leuselator who wrote the stepper driver.

joba-1, 2019-11-27:
my2ct: Steppers themselves usually don't report anything.

You have (at least) one endstop that reports a known position and from there on it is step counting by the „user“ and trusting no steps are lost. 

Usually this works very well, see 3D printers as proof...

TrDA-hab, 2019-11-27:
@stefanbode

But how does the current driver behave if i change the position during movement?
 1. To change the movement, it is necessary to stop the stepper motor (STOP command), and then give the stepper motor a command to move in the opposite direction.
 2. To stop the stepper motor, it is necessary to issue the STOP command, but this command is currently not implemented in the firmware.
 3. Currently, you can only wait for the current command to execute, only then give another command.

Is there an option to get the current position of the stepper?
 1. There is no feedback in standard drivers / motors, it is not possible to get the position of the motor.
 2. We can only send a command to the driver with the indicated number of steps + direction of movement + rotation speed, and we hope that there will be no skipping steps at the stepper motor.

stefanbode, 2019-11-30:
Hi gents, I have now a working Nema17 with a a4988 stepper. In general this will be more complicated than I expected because you cannot stop the motor. Once you send a command to the driver the ESP ist busy. It looks like the command is blocking. The only way to get it something similar to the current behaviour is to split the command in 100ms chunks....

stefanbode, 2019-12-01:
I enhanced the motordriver in different ways. Also worked with the most course stepping of 1 to get the maximum time delay between two commands. Additionally made the numbers of remaining steps public, so you can control. Anyhow the required behavior makes it really tough to use it in Tasmota. I will now check, if I can use the PWM with a frequency "hook" to get more control during operation of the stepper driver.

stefanbode, 2019-12-02:
Good news, friends. I have an updated shutter driver that works with the stepper motor. Anyhow this is really tricky stuff right now and I need some investigation if I broke down something else. The good news is we do not need the 10k code from the Stepper driver.

RiRilot, 2019-12-02:
Happy to test. I’m currently using a NodeMCU to drive a stepper to operate my roller blinds. Using rules at the moment to deal with open and close. One thing that would be really useful would be if acceleration could be implemented so as to avoid the hard start when the stepper starts moving. Also - this is mainly for quality of life - would it be possible to get a settings page for MotorRPM and MotorSPR? The current build doesn’t seem to remember those settings after reboots (although that could be an issue with my build). I’ve set them pre-compile now to get around it.

stefanbode, 2019-12-02:
Oh, You will love it :-) I had exactly the same in mind. Ramp is definitely a good idea, to avoid overload the motor. Regarding the RPM this is not as easy. I have a fixed 1000 steps/sec which is 5RPS when having stepping to 1. The ramp is 0.5sec; which is fast enough but also gives some room for accelerating the heavy load. I still work on the timing because as always it is NOT straight forward just to use just the stepper driver. This is unusable because it blocks TASMOTA for the complete operation.

meingraham, 2019-12-02:
Stefan - Thank you! As always, a better outcome than originally envisioned and much more useful for Tasmota users!

RiRilot, 2019-12-02:
Stefan - That sounds great. Looking forward to testing. The wife will not be happy that I'm going to disassemble the kitchen blinds again......

Leuselator, 2019-12-02:
Great job guys. I was wondering, why my code results in so much additional driversize as well. I was happy enough to get it work somehow and yes - the way I drive the motor is blocking Tasmota and not elegant at all. Unfortunately I've no time @ the moment to put effort into this - but I cross fingers for you!
Greetings

TrDA-hab, 2019-12-02:
@stefanbode
I used to use these libraries DRV8825.h
Maybe you can use this library?
https://github.com/laurb9/StepperDriver

for information only.

stefanbode, 2019-12-02:
I have added the stepper driver support for shutters/blinds. Please start testing with the following configuration:
```
Schuttermode 2 Failsafe Configuration. No INTERLOCK
D1: Relay1i: A4988 PIN: EN
D2: Relay2 : A4988 PIN: DIR
D3: PWM1: A4988 PIN: STP
D4: COUNTER1: Connect to D3
```

The Stepper will operate with 1000 steps/sec. Ramp up 0..1000 in 10 steps in 0.5 sek. Ramp down with 500 steps/sec.

`Shutteropenduration `and `shuttercloseduration `must be equal.

Ramp will take 0.25sec extra per move. You CAN define `shuttermotordelay 0.25`, but not required.

You can start/stop/change direction at any time. Stepper will go with full speed near endpoint and then slow down to touch correct position. If shuttermotordelay is too high the stepper will overrun the endpoint. bad idea! I did several movements forward and backward. There was no single step missing. Position was always targeted 100%.

RiRilot, 2019-12-02:
Thanks Stefan. I'll get testing tomorrow evening. Thanks for all your efforts.

stefanbode, 2019-12-02:
At the end it was really fun to overcome the issue that after some time you did not exactly get back where you should be. Using the COUNTER and measure what you do with a ramp starting from 0Hz did the trick at the end. The PWM is much smarter because integrated in the core. All the drivers block the code. Very inconvenient, if you expect a runtime of 20sec or more.

TrDA-hab, 2019-12-03:
@stefanbode

On the issue of receiving feedback from stepper motors, there are three good projects with great potential. Receiving the information:
- direction of movement.
- speed of movement.
- restore missed steps.
- and other.
You might find this interesting.

1. Ustepper:
https://ustepper.com/store/ustepper-boards/27-46-ustepper-s.html#/28-connector-dc_jack/31-stepper_motor-no_stepper_motor
https://github.com/uStepper/uStepperS
https://nl.aliexpress.com/item/4000418431198.html

2. Mechaduino:
https://tropical-labs.com/shop/mechaduino-0-2-pcb/
https://github.com/jcchurch13
https://nl.aliexpress.com/item/32952527069.html

3. MKS-SERVO:
https://github.com/makerbase-mks/MKS-SERVO42A
https://nl.aliexpress.com/item/32917408111.html

for information only.

meingraham, 2019-12-03:
@titomontes @TrDA-hab @RiRilot @Leuselator 

Are you able to test this new feature and report back your success or any issues encountered?

RiRilot, 2019-12-03:
Yes, I will this evening UK time. I have compiled the latest dev branch and stuck it on a NodeMCU. I'll hook it up and report back.

TrDA-hab, 2019-12-03:
Yes, I find this solution very interesting and will do many tests. I will report the result back. I will do it now.

RiRilot, 2019-12-03:
I’m struggling to get the stepper to move more than a few steps in either direction. I think this is because I’m not very familiar with the shutter configuration. I can’t seem to get them in to shutter mode 2.

RiRilot, 2019-12-03:
What I'm seeing is the stepper moving a few steps then stopping. The counter is continuing to count steps which says to me that the EN pin isn't staying high.

TrDA-hab, 2019-12-03:
It doesn’t work for me either. The error is the same.
[image]
[image]
[image]

stefanbode, 2019-12-03:
You have compiled you own version and enabled the shutter in the user config? Only then you have access to the shutter commands. The shutter mode 2 is default and wil be selected depending on interlock and pulsetimers. If you do nothing everything is fine. Please see the wiki. You have to execute setoption80 1 and reboot to enable the shutter. There is an actual wiki also for steppers already online

stefanbode, 2019-12-03:
https://tasmota.github.io/docs/#/Blinds-and-Shutters

TrDA-hab, 2019-12-03:
Yes, I forgot to execute setoption80 1. The error is the same.

`23:38:52 CMD: SetOption80
23:38:52 MQT: stat/tasmota/RESULT = {"SetOption80":"ON"}
23:39:08 CMD: ShutterOpenDuration
23:39:08 MQT: stat/tasmota/RESULT = {"ShutterOpenDuration1":"500.0"}
23:39:20 CMD: ShutterCloseDuration
23:39:20 MQT: stat/tasmota/RESULT = {"ShutterCloseDuration1":"500.0"}
23:39:46 CMD: ShutterRelay1 1
23:39:46 SHT: Relay 1 is 1
23:39:46 SHT: Accuracy digits: 1
23:39:46 SHT: Shutter 0 (Relay:1): Init. Pos: 0 [0 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 1000000, Opentime 500.0 [s], Closetime 500.0 [s], CoedffCalc: c0: 300, c1 500, c2: 700, c3: 900, c4: 1000, binmask 3, is inverted 0, shuttermode 1,motordelay 0
23:39:46 MQT: stat/tasmota/RESULT = {"ShutterRelay1":1}`

stefanbode, 2019-12-03:
Shutter mode is 1. There is an interlock defined. Please execute interlock off and restart

stefanbode, 2019-12-03:
Your open duration is really 500 sek? I have to check if my variables are all big enough.

stefanbode, 2019-12-03:
The rest looks ok. Please do not forget the counter 1 as described in the wiki

RiRilot, 2019-12-03:
00:00:32 CMD: shutterrelay1 1
00:00:32 SHT: Relay 1 is 1
00:00:32 SHT: Accuracy digits: 1
00:00:32 SHT: Shutter 0 (Relay:1): Init. Pos: 18000 [90 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 20000, Opentime 10.0 [s], Closetime 10.0 [s], CoedffCalc: c0: 300, c1 500, c2: 700, c3: 900, c4: 1000, binmask 3, is inverted 0, shuttermode 2,motordelay 0
00:00:32 RSL: stat/tasmota/RESULT = {"ShutterRelay1":1}

I have GPIOs defines as this:
[image]

On togging a relay motor moves a few steps then stops. Counter continues to count pulses. Motor stays stationary:

[image]

stefanbode, 2019-12-03:
Let’s assume the wire ring to the a4988 is correct. Can you increase log level to 4 and send the log. Looks still ok from here. The current on the driver should be 0.3A. There is a small screw on the driver that you can change with a plastic !!! Screwdriver. 1a, 1b and 2a, 2b are correct? There should be a non infinite resistance on the 1 and the 2’s.

TrDA-hab, 2019-12-03:
`23:59:24 CMD: shuttermotordelay 0.25
23:59:24 SHT: Accuracy digits: 1
23:59:24 SHT: Shutter 0 (Relay:1): Init. Pos: 0 [0 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 100000, Opentime 50.0 [s], Closetime 50.0 [s], CoedffCalc: c0: 300, c1 500, c2: 700, c3: 900, c4: 1000, binmask 3, is inverted 0, shuttermode 1,motordelay 5
23:59:24 MQT: stat/tasmota/RESULT = {"ShutterMotorDelay1":"0.25"}
23:59:34 CMD: ShutterRelay
23:59:34 MQT: stat/tasmota/RESULT = {"ShutterRelay1":1}
00:00:06 CMD: ShutterRelay1 1
00:00:06 SHT: Relay 1 is 1
00:00:06 SHT: Accuracy digits: 1
00:00:06 SHT: Shutter 0 (Relay:1): Init. Pos: 0 [0 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 100000, Opentime 50.0 [s], Closetime 50.0 [s], CoedffCalc: c0: 300, c1 500, c2: 700, c3: 900, c4: 1000, binmask 3, is inverted 0, shuttermode 1,motordelay 5
00:00:06 MQT: stat/tasmota/RESULT = {"ShutterRelay1":1}
00:01:43 CMD: Shutteropenduration 500
00:01:43 SHT: Accuracy digits: 1
00:01:43 SHT: Shutter 0 (Relay:1): Init. Pos: 0 [0 %], Open Vel.: 100 Close Vel.: 1000 , Max Way: 1000000, Opentime 500.0 [s], Closetime 50.0 [s], CoedffCalc: c0: 300, c1 500, c2: 700, c3: 900, c4: 1000, binmask 3, is inverted 0, shuttermode 1,motordelay 5
00:01:43 MQT: stat/tasmota/RESULT = {"ShutterOpenDuration1":"500.0"}
00:01:59 CMD: ShutterCloseDuration 500
00:01:59 SHT: Accuracy digits: 1
00:01:59 SHT: Shutter 0 (Relay:1): Init. Pos: 0 [0 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 1000000, Opentime 500.0 [s], Closetime 500.0 [s], CoedffCalc: c0: 300, c1 500, c2: 700, c3: 900, c4: 1000, binmask 3, is inverted 0, shuttermode 1,motordelay 5
00:01:59 MQT: stat/tasmota/RESULT = {"ShutterCloseDuration1":"500.0"}`

stefanbode, 2019-12-03:
Se d the log when sending shutteropen or Shutterclose with the 10sec. To make it simple first

stefanbode, 2019-12-03:
Shutter mode is again 1. This is wrong again

TrDA-hab, 2019-12-03:
`00:06:39 CMD: shutteropen 10
00:06:39 SRC: WebConsole from 192.168.1.52
00:06:39 CMD: Group 0, Index 1, Command "SHUTTEROPEN", Data "10"
00:06:39 SHT: Position in: payload 10 (2), payload 100, index 1, source 7
00:06:39 SHT: lastsource 7:, realpos 0, target 1000000, payload 100
00:06:39 SRC: Shutter
00:06:39 SHT: Switched relay: 0 by Shutter
00:06:39 MQT: stat/tasmota/RESULT = {"POWER1":"OFF"}
00:06:39 MQT: stat/tasmota/POWER1 = OFF
00:06:39 SHT: Wait for Motorstop 500
00:06:39 SRC: Shutter
00:06:39 SHT: Switched relay: 0 by Shutter
00:06:39 MQT: stat/tasmota/RESULT = {"POWER2":"OFF"}
00:06:39 MQT: stat/tasmota/POWER2 = OFF
00:06:39 SRC: Shutter
00:06:39 SHT: Switched relay: 1 by Shutter
00:06:39 MQT: stat/tasmota/RESULT = {"POWER1":"ON"}
00:06:39 MQT: stat/tasmota/POWER1 = ON
00:06:39 MQT: stat/tasmota/RESULT = {"ShutterOpen1":100}
00:06:39 SHT: Shutter 0: Real Pos: 800, Target 1000000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 0.6  [s]
00:06:39 SRC: PulseTimer
00:06:39 SHT: Switched relay: 1 by PulseTime
00:06:39 MQT: stat/tasmota/RESULT = {"POWER1":"OFF"}
00:06:39 MQT: stat/tasmota/POWER1 = OFF
00:06:40 WIF: Checking connection...
00:06:40 WIF: Connected
00:06:40 SHT: Shutter 0: Real Pos: 2800, Target 1000000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 1.6  [s]
00:06:41 SHT: Shutter 0: Real Pos: 4900, Target 1000000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 2.7  [s]
00:06:42 SHT: Shutter 0: Real Pos: 6900, Target 1000000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 3.7  [s]
00:06:43 SHT: Shutter 0: Real Pos: 8900, Target 1000000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 4.7  [s]
00:06:44 SHT: Shutter 0: Real Pos: 10900, Target 1000000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 5.7  [s]
00:06:45 SHT: Shutter 0: Real Pos: 12900, Target 1000000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 6.7  [s]
00:06:46 SHT: Shutter 0: Real Pos: 14900, Target 1000000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 7.7  [s]
00:06:47 SHT: Shutter 0: Real Pos: 16800, Target 1000000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 8.6  [s]
00:06:48 SHT: Shutter 0: Real Pos: 18800, Target 1000000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 9.6  [s]
00:06:49 SHT: Shutter 0: Real Pos: 20800, Target 1000000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 10.6  [s]
00:06:50 SHT: Shutter 0: Real Pos: 22900, Target 1000000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 11.7  [s]
00:06:51 SHT: Shutter 0: Real Pos: 24900, Target 1000000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 12.7  [s]
00:06:52 SHT: Shutter 0: Real Pos: 26800, Target 1000000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 13.6  [s]
00:06:53 SHT: Shutter 0: Real Pos: 28900, Target 1000000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 14.7  [s]
00:06:54 SHT: Shutter 0: Real Pos: 30900, Target 1000000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 15.7  [s]
00:06:55 SHT: Shutter 0: Real Pos: 32800, Target 1000000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 16.6  [s]
00:06:56 SHT: Shutter 0: Real Pos: 34800, Target 1000000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 17.6  [s]
00:06:57 SHT: Shutter 0: Real Pos: 36900, Target 1000000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 18.7  [s]
00:06:58 SHT: Shutter 0: Real Pos: 38800, Target 1000000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 19.6  [s]
00:06:59 SHT: Shutter 0: Real Pos: 40800, Target 1000000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 20.6  [s]
00:07:00 WIF: Checking connection...
00:07:00 WIF: Connected
00:07:00 SHT: Shutter 0: Real Pos: 42800, Target 1000000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 21.6  [s]
00:07:01 SHT: Shutter 0: Real Pos: 44800, Target 1000000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 22.6  [s]
00:07:02 SHT: Shutter 0: Real Pos: 46800, Target 1000000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 23.6  [s]
00:07:03 SHT: Shutter 0: Real Pos: 48800, Target 1000000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 24.6  [s]
00:07:04 SHT: Shutter 0: Real Pos: 50800, Target 1000000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 25.6  [s]`
......................

RiRilot, 2019-12-03:
Driver is fine. I can controll it with the normal A4988 code.

22:07:21 CMD: shutterclose
22:07:21 SHT: Start shutter in direction -1
22:07:21 RSL: stat/tasmota/RESULT = {"POWER2":"ON"}
22:07:21 RSL: stat/tasmota/POWER2 = ON
22:07:21 RSL: stat/tasmota/RESULT = {"ShutterClose1":0}
22:07:21 RSL: stat/tasmota/RESULT = {"POWER2":"OFF"}
22:07:21 RSL: stat/tasmota/POWER2 = OFF
22:07:21 SHT: Shutter 0: Real Pos: 19000, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 0.5  [s]
22:07:22 SHT: Shutter 0: Real Pos: 17000, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 1.5  [s]
22:07:23 SHT: Shutter 0: Real Pos: 15000, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 2.5  [s]
22:07:24 SHT: Shutter 0: Real Pos: 13000, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 3.5  [s]
22:07:25 SHT: Shutter 0: Real Pos: 11000, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 4.5  [s]
22:07:26 SHT: Shutter 0: Real Pos: 9000, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 5.5  [s]
22:07:27 SHT: Shutter 0: Real Pos: 7000, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 6.5  [s]
22:07:28 SHT: Shutter 0: Real Pos: 5000, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 7.5  [s]
22:07:29 SHT: Shutter 0: Real Pos: 3000, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 8.5  [s]
22:07:30 SHT: Shutter 0: Real Pos: 1000, Target 0, source: PulseTimer, start-pos: 100 %, direction: -1, rtcshutter: 9.5  [s]
22:07:31 RSL: stat/tasmota/RESULT = {"POWER2":"ON"}
22:07:31 RSL: stat/tasmota/POWER2 = ON
22:07:31 RSL: stat/tasmota/SHUTTER1 = 0
22:07:31 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":0}}
22:07:31 RSL: stat/tasmota/RESULT = {"POWER2":"OFF"}
22:07:31 RSL: stat/tasmota/POWER2 = OFF
22:07:40 CMD: shutteropen
22:07:40 SHT: Start shutter in direction 1
22:07:40 RSL: stat/tasmota/RESULT = {"POWER1":"ON"}
22:07:40 RSL: stat/tasmota/POWER1 = ON
22:07:40 RSL: stat/tasmota/RESULT = {"ShutterOpen1":100}
22:07:40 SHT: Shutter 0: Real Pos: 400, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 0.2  [s]
22:07:40 RSL: stat/tasmota/RESULT = {"POWER1":"OFF"}
22:07:40 RSL: stat/tasmota/POWER1 = OFF
22:07:41 SHT: Shutter 0: Real Pos: 2400, Target 20000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 1.2  [s]
22:07:42 SHT: Shutter 0: Real Pos: 4400, Target 20000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 2.2  [s]
22:07:43 SHT: Shutter 0: Real Pos: 6400, Target 20000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 3.2  [s]
22:07:44 SHT: Shutter 0: Real Pos: 8400, Target 20000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 4.2  [s]
22:07:45 SHT: Shutter 0: Real Pos: 10400, Target 20000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 5.2  [s]
22:07:46 SHT: Shutter 0: Real Pos: 12400, Target 20000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 6.2  [s]
22:07:47 SHT: Shutter 0: Real Pos: 14400, Target 20000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 7.2  [s]
22:07:48 SHT: Shutter 0: Real Pos: 16400, Target 20000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 8.2  [s]
22:07:49 SHT: Shutter 0: Real Pos: 18300, Target 20000, source: PulseTimer, start-pos: 0 %, direction: 1, rtcshutter: 9.1  [s]
22:07:50 RSL: stat/tasmota/RESULT = {"POWER1":"ON"}
22:07:50 RSL: stat/tasmota/POWER1 = ON
22:07:50 RSL: stat/tasmota/SHUTTER1 = 100
22:07:50 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":100,"direction":0}}
22:07:50 RSL: stat/tasmota/RESULT = {"POWER1":"OFF"}
22:07:50 RSL: stat/tasmota/POWER1 = OFF

stefanbode, 2019-12-03:
Shuttermode 2 does mean Relay1 enable or disable the motor. Relay2 is for the direction. The Pwm Pulses the stepper.

TrDA-hab, 2019-12-03:
How can I install Shuttermode 2 ???

stefanbode, 2019-12-03:
Pulsetime 0 on both relay and no interlock in the shutterrelays.

stefanbode, 2019-12-03:
The modes get selected by the definition of interlock and pulsetime. Please take a look again in the wiki for the circuit safe configuration

TrDA-hab, 2019-12-03:
Backlog PulseTime1 0; PulseTime2 0
Backlog Interlock 1,2; Interlock OFF

correctly ???
correctly ???

stefanbode, 2019-12-03:
Interlock 1,2 is also not needed. Just in case put interlock 3,4. So sure relays are not in

RiRilot, 2019-12-03:
I have set those and now it's working. That isn't shuttermode 2 according to the docs. That's what threw me.

Docs say:

Stepper motors (ShutterMode: 2) - First relay: OFF/DOWN PULSE, Second relay: OFF/UP PULSE
Backlog PulseTime1 2; PulseTime2 2
Backlog Interlock 1,2; Interlock ON (Interlocked relay pair)

I did:

backlog pulsetime1 0; pulsetime 2 0
backlog interlock 1,2; Interlock OFF

stefanbode, 2019-12-03:
Uups. Really?

RiRilot, 2019-12-03:
Yup

stefanbode, 2019-12-03:
Oh yes long long day for me. It is mode 1. My fault. Any progress?

RiRilot, 2019-12-03:
Yeah, it's all working now! I've managed to set a good open and close time. No lost steps.

RiRilot, 2019-12-03:
We could really do with a setting to increase the motor RPM in this mode as I use 5:1 panetary geared steppers and the current speed is slooooooooooooow.

stefanbode, 2019-12-03:
You have ms1,ms2,ms3 to ground to get full stars and not only 1/16?

TrDA-hab, 2019-12-03:
00:16:15 CMD: Backlog PulseTime1 0; PulseTime2 0
00:16:15 SRC: WebConsole from 192.168.1.52
00:16:15 CMD: Group 0, Index 1, Command "BACKLOG", Data "PulseTime1 0; PulseTime2 0"
00:16:16 SRC: Backlog
00:16:16 CMD: Group 0, Index 1, Command "PULSETIME", Data "0"
00:16:16 MQT: stat/tasmota/RESULT = {"PulseTime1":{"Set":0,"Remaining":0}}
00:16:16 SRC: Backlog
00:16:16 CMD: Group 0, Index 2, Command "PULSETIME", Data "0"
00:16:16 MQT: stat/tasmota/RESULT = {"PulseTime2":{"Set":0,"Remaining":0}}
00:16:16 CFG: Saved to flash at F7, Count 125, Bytes 4096
00:16:20 WIF: Checking connection...
00:16:20 WIF: Connected
00:16:25 CMD: Backlog Interlock 3,4; Interlock OFF
00:16:25 SRC: WebConsole from 192.168.1.52
00:16:25 CMD: Group 0, Index 1, Command "BACKLOG", Data "Interlock 3,4; Interlock OFF"
00:16:25 SRC: Backlog
00:16:25 CMD: Group 0, Index 1, Command "INTERLOCK", Data "3,4"
00:16:25 MQT: stat/tasmota/RESULT = {"Interlock":"OFF","Groups":"1,2"}
00:16:26 SRC: Backlog
00:16:26 CMD: Group 0, Index 1, Command "INTERLOCK", Data "OFF"
00:16:26 MQT: stat/tasmota/RESULT = {"Interlock":"OFF","Groups":"1,2"}
00:16:26 CFG: Saved to flash at F6, Count 126, Bytes 4096
00:16:33 HTP: Main Menu
00:16:38 HTP: Console
00:16:40 WIF: Checking connection...
... and the stepper works without stopping!?!

RiRilot, 2019-12-03:
Yeah, I have my 4988 on a breakout with dip switches I can configure. Got them all set to 1/1 stepping.

stefanbode, 2019-12-03:
It makes 5rps. The current is 1000Hz. After 1500 it starts slipping. All ms must be LOW

stefanbode, 2019-12-03:
This are 300 rpm and this is very fast normally. One round are 200 pulses.

RiRilot, 2019-12-03:
At 1/5 gear ratio it need 1000 pulses for 1 revolution at 1/1 microstepping.

[40 more replies]

PR #7152 (2019-12-09): Update xdrv_27_shutter.ino
fix #7006 support for multiple shutters with steppers
+ add ShutterOpen/Close/Stop can get shutter index as parameter. e.g. shutteropen 3 to open shutter 3

## Description:

**Related issue (if applicable):** fixes #<Tasmota issue number goes here>

## Checklist:
  - [x ] The pull request is done against the latest dev branch
  - [x ] Only relevant files were touched
  - [ x] Only one feature/fix was added per PR.
  - [x ] The code change is tested and works on core 2.6.1
  - [ ] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x [... cut]
commit 7cc7c2cc (2019-12-09): Update xdrv_27_shutter.ino

fix #7006 support for multiple shutters with steppers
+ add ShutterOpen/Close/Stop can get shutter index as parameter. e.g. shutteropen 3 to open shutter 3
closed 2019-12-09 by arendst