arendst/Tasmota #7011: MQTT command line parsing does not observer quotes

GaryMilne, 2019-11-23:
> **GUIDE**
>
> This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
>
> Please DO NOT OPEN AN ISSUE:
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/
>  - If you have a issue when flashing was done via Tuya Convert
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Tasmota/wiki/FAQ) and troubleshooting wiki articles
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
_A clear and concise description of what the bug is._
Sending this command
publish cmnd/Tasmota4/backlog "Rule2 on power1#boot DO color 9 ENDON"
Generates this output. Command line is parsed at first space and quotations are ignored.
[image]

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): sonoff.bin_____
- [ ] Tasmota binary firmware version number used: 6.7.1_____
  - [Yes ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X ] Flashing tools used: Serial_____
- [X] Provide the output of command: ``Backlog Template; Module; GPIO``:
  22:38:38 CMD: Backlog Template; Module; GPIO
22:38:38 MQT: stat/Tasmota4/RESULT = {"NAME":"GlobeRGBWW","GPIO":[0,0,0,0,37,40,0,0,38,41,39,0,0],"FLAG":0,"BASE":18}
22:38:38 MQT: stat/Tasmota4/RESULT = {"Module":{"0":"GlobeRGBWW"}}
22:38:39 MQT: stat/Tasmota4/RESULT = {"GPIO":"Not supported"}
  Configuration output here:
[image]

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
22:39:31 CMD: status 0
22:39:31 MQT: stat/Tasmota4/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota4"],"Topic":"Tasmota4","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:39:31 MQT: stat/Tasmota4/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:09:27","StartupUTC":"2019-11-23T21:30:04","Sleep":50,"CfgHolder":4617,"BootCount":72,"SaveCount":2284,"SaveAddress":"FA000"}}
22:39:31 MQT: stat/Tasmota4/STATUS2 = {"StatusFWR":{"Version":"6.7.1(sonoff)","BuildDateTime":"2019-10-26T13:19:13","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
22:39:31 MQT: stat/Tasmota4/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["5441",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000"]}}
22:39:31 MQT: stat/Tasmota4/STATUS4 = {"StatusMEM":{"ProgramSize":564,"Free":440,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","0000F081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,30","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
22:39:31 MQT: stat/Tasmota4/STATUS5 = {"StatusNET":{"Hostname":"Tasmota4","IPAddress":"192.168.0.124","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"84:F3:EB:36:FD:79","Webserver":2,"WifiConfig":4}}
22:39:31 MQT: stat/Tasmota4/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.40","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_36FD79","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
22:39:31 MQT: stat/Tasmota4/STATUS7 = {"StatusTIM":{"UTC":"Sat Nov 23 21:39:31 2019","Local":"Sat Nov 23 22:39:31 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:11","Sunset":"17:02"}}
22:39:31 MQT: stat/Tasmota4/STATUS10 = {"StatusSNS":{"Time":"2019-11-23T22:39:31"}}
22:39:31 MQT: stat/Tasmota4/STATUS11 = {"StatusSTS":{"Time":"2019-11-23T22:39:31","Uptime":"0T00:09:27","UptimeSec":567,"Heap":26,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":781,"MqttCount":1,"POWER":"ON","Dimmer":94,"Color":"A020F00000","HSBColor":"277,87,94","Channel":[63,13,94,0,0],"CT":153,"Scheme":0,"Fade":"OFF","Speed":3,"LedTable":"ON","Wifi":{"AP":1,"SSId":"5441","BSSId":"A0:04:60:95:0E:62","Channel":3,"RSSI":66,"LinkCount":1,"Downtime":"0T00:00:07

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_
publish cmnd/Tasmota4/backlog "Rule2 on power1#boot DO color 9 ENDON"

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The entire string between the quotes would be transmitted.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I think it is self explanatory.  
P.S. I can issue the equivalent command from the mosquitto command line and it works there.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-11-24:
Hi,

Your command has a syntax error.

It should be:

```
publish cmnd/Tasmota4/Rule2 on power1#boot DO color 9 ENDON
```

ascillato2, 2019-11-24:
If you need further assistant, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. 

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

GaryMilne, 2019-11-24:
Unfortunately no-one responded when I posted the message on there originally.
You say it is a syntax error but that is not true. That command is perfectly legitimate but admittedly unusual to issue a backlog command with a single command, but I was simplifying.

However, if you post the command as you suggest:
publish cmnd/Tasmota4/Rule2 on power1#boot DO color 9 ENDON

the result is: rule2 "on

which is clearly not its intention. It parses the line only as far as the first white space and considers that to be the payload and drops the rest of the command.

ascillato, 2019-11-24:
Hi, the syntax error I was referring were the quotes. Quotes are not needed nor supported.

Anyway, using any of the following valid expressions, the outcome is not the expected:
```
publish cmnd/Tasmota4/Rule2 on power1#boot DO color 9 ENDON

publish cmnd/Tasmota4/backlog Rule2 on power1#boot DO color 9 ENDON
```

@arendst 

Is there any way of sending rules from mqtt?

GaryMilne, 2019-11-24:
and to reiterate my prior point you can send these commands via mosquitto in the format shown below which does observe the use of quoted strings.
 mosquitto_pub  -t cmnd/KitchenCounter/Backlog -m "Rule2 on power1#boot do DIMMER 100 ENDON; Rule2 1"

and in case you are wondering why the heck you would want to do this here is my use case.  I'm writing a high quality RGBW Tasmota driver for Smartthings and I would like to be able to perform the above action with a single key press for a lighting group. It works directly to the IP but not via MQTT. I could certainly send the same command via the mosquitto server but that requires that I add a mosquitto config, IP, Port and credentials which I can avoid if I can effectively use the currently configured bulb as a proxy for my commands.

My use case is that I have a number of bulbs that are on regular switches. My girlfriend hates when I leave the bulbs in any state other than "normal white". This ensures that after I have changed the lighting, as soon as they are turned off they will always come back on in specific state, what I am calling a default color.  She also knows that she can always get it back to "default" by turning the bulbs off an on, which makes her happy.

I hope that all makes sense.

blakadder, 2019-11-24:
Depending on the number and type of rules, isn't it better to set up the rules then use mqtt to activate or deactivate them?

GaryMilne, 2019-11-24:
> Depending on the number and type of rules, isn't it better to set up the rules then use mqtt to activate or deactivate them?

I would say that you are right most of the time but bear in mind that in this use case you are using a color picker to select a color, observing the bulb and when you are happy you click the button to apply this as a powerup color via the rule. So it's an iterative and interactive process designed for spontaneous use by anyone in the household.

oywino, 2019-12-13:
Hi all, 
I have several Sonoff-basic devices Tasmotized with v6.6.0 integrated with Home Assistant 0.103.0 running in a container and connected through Mosquitto MQTT 1.6.8
Everything works just fine. Even If I restart Home Assistant now and then, everyting is still ok.
But when I delete the container in order to update to a new release of Home Assistant (downloaded from GitHub), all my Sonoffs fail to automatically reconnect and show up in Home Assistant. They appear as unavailable. But I can publish commands and control each one successfully from Home Assistant Developer Page --> MQTT, and they do respond correctly! But if I open the Entity Setting, they are listed as unavailable, and my configured Lovelace cards just state "Entity not available" on a yellow card. If I unplug the power from each Sonoff, and immediately power it back on, it magically shows up as available.  As you can imagine, this is not very convenient. 
Any help to solve this mystery, is highly appreciated.

ascillato, 2019-12-13:
Hi, your problem is not related to this issue. Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2019-12-17:
Closing this issue as the proposed usage of sending rules by publish command is not supported, and would need a rework of the publish command that is not going to be done soon. Sorry.
Thanks for sharing your report.

closed 2019-12-17 by ascillato2