arendst/Tasmota #7016: Any way to sync sonoff rf with a new rf wall switch other then pressing the button for 3 sec?

catalin31, 2019-11-24:
Hi guys.

I am searching the internet since yesterday trying to find if there is a way to sync sonoff rf with a new rf wall switch other then pressing the button on the device for 3 sec. looking at the commands or options I do not see any that will put the sonoff in pairing mode, but maybe I mised something? or can this be added as a feature request? It will be super handy than having to messing again with the bulb. 

Thanks,
Catalin

meingraham, 2019-11-24:
Our ability to provide assistance is greatly hampered since you did not take a few minutes to complete the  [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) with the requested information. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

For example, there is a `Sonoff RF` and a `Sonoff RF Bridge'.

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2019-11-24:
Sorry, that pairing is done by the rf chip and can't be done by software. You need to follow that manufacturer procedure.

closed 2019-11-24 by ascillato2