arendst/Tasmota #7017: ShutterRelay1 cant be in 0

jlpetracca, 2019-11-24:
Hi! 
- using tasmota t1 us ch3
- i am trying to force ShutterRelay1 0 just to use ShutterRelay2 1 and ShutterRelay3 1

But it always turn ShutterRelay1 1 again.

i want to use only Relay 2 and Relay 3 for shutters and relay 1 for lights, thats why

Thanks

> **GUIDE**
>
> This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
>
> Please DO NOT OPEN AN ISSUE:
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/
>  - If you have a issue when flashing was done via Tuya Convert
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Tasmota/wiki/FAQ) and troubleshooting wiki articles
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
_A clear and concise description of what the bug is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-11-24:
Please complete the template and also post the output of interlock command

jlpetracca, 2019-11-24:
Interlock off

{"NAME":"Generic","GPIO":[17,255,255,255,23,22,18,19,21,56,0,0,0],"FLAG":0,"BASE":30}

ascillato, 2019-11-24:
Hi, sorry. The template is still incomplete. Why we ask to complete the template? It is to not lose time you and me with needed questions. Example. Which Tasmota version are you using? You didn't specify. So, we can't narrow to a reproducible scenario and see if it is a bug that needs to be addressed (and where it is) or if it is just a misconfiguration.
So, please, take some few minutes on sharing the requested information in order to have more debug data to properly help you. Thanks for understanding.

jlpetracca, 2019-11-24:
Program Version | 6.7.1(sonoff)
-- | --
2019-11-24T15:18:01
2_6_1/2.2.2-dev(5ab15d1)
0T00:01:07
12 at 0xF8000
4
Power on
Sonoff
 
Blanca (80%)
sonoff-5175
192.168.1.143
192.168.1.1
255.255.255.0
192.168.1.1
DC:4F:22:DB:F4:37
 
 
1883
mosqui
DVES_DBF437
sonoff
sonoffs
sonoff/cmnd/
cmnd/DVES_DBF437_fb/
 
None
Disabled
 
14414903
0x144051
1024kB
1024kB
496kB
504kB
28kB

jlpetracca, 2019-11-24:
I compiled myself because shutter is not in any version.

jlpetracca, 2019-11-24:
9:47:32 RSL: homeassistant/light/DBF437_LI_7/config = 
19:47:32 RSL: homeassistant/switch/DBF437_RL_7/config = 
19:47:32 RSL: homeassistant/light/DBF437_LI_8/config = 
19:47:32 RSL: homeassistant/switch/DBF437_RL_8/config = 
19:47:32 RSL: homeassistant/binary_sensor/DBF437_BTN_1/config = {"name":"Sonoff Button1","stat_t":"~stat/RESULT","value_template":"{{value_json.BUTTON1}}","pl_on":"TOGGLE","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"DBF437_BTN_1","device":{"identifiers":["DBF437"],"connections":[["mac","DC:4F:22:DB:F4:37"]]},"~":"sonoff/","off_delay":1}
19:47:32 RSL: homeassistant/binary_sensor/DBF437_BTN_2/config = 
19:47:32 RSL: homeassistant/binary_sensor/DBF437_BTN_3/config = 
19:47:32 RSL: homeassistant/binary_sensor/DBF437_BTN_4/config = 
19:47:32 RSL: homeassistant/binary_sensor/DBF437_SW_1/config = 
19:47:32 RSL: homeassistant/binary_sensor/DBF437_SW_2/config = 
19:47:32 RSL: homeassistant/binary_sensor/DBF437_SW_3/config = 
19:47:32 RSL: homeassistant/binary_sensor/DBF437_SW_4/config = 
19:47:32 RSL: homeassistant/binary_sensor/DBF437_SW_5/config = 
19:47:32 RSL: homeassistant/binary_sensor/DBF437_SW_6/config = 
19:47:32 RSL: homeassistant/binary_sensor/DBF437_SW_7/config = 
19:47:32 RSL: homeassistant/binary_sensor/DBF437_SW_8/config = 
19:47:32 RSL: homeassistant/sensor/DBF437_status/config = {"name":"Sonoff status","stat_t":"~HASS_STATE","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attributes_topic":"~HASS_STATE","unit_of_meas":" ","val_tpl":"{{value_json['RSSI']}}","ic":"mdi:information-outline","uniq_id":"DBF437_status","device":{"identifiers":["DBF437"],"connections":[["mac","DC:4F:22:DB:F4:37"]],"name":"Sonoff","model":"Sonoff Basic","sw_version":"6.7.1(sonoff)","manufacturer":"Tasmota"},"~":"sonoff/tele/"}
19:47:39 RSL: sonoff/tele/STATE = {"Time":"2019-11-24T19:47:39","Uptime":"0T00:00:16","UptimeSec":16,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"Blanca","BSSId":"00:1A:70:5D:D4:A6","Channel":10,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:05"}}
19:48:07 RSL: sonoff/stat/RESULT = {"BUTTON1":"TOGGLE"}
19:48:07 RSL: sonoff/tele/STATE = {"Time":"2019-11-24T19:48:07","Uptime":"0T00:00:44","UptimeSec":44,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Blanca","BSSId":"00:1A:70:5D:D4:A6","Channel":10,"RSSI":80,"LinkCount":1,"Downtime":"0T00:00:05"}}
19:48:07 RSL: sonoff/stat/RESULT = {"POWER":"OFF"}
19:48:07 RSL: sonoff/stat/POWER = OFF
19:48:58 RSL: sonoff/stat/RESULT = {"BUTTON1":"TOGGLE"}
19:48:58 RSL: sonoff/tele/STATE = {"Time":"2019-11-24T19:48:58","Uptime":"0T00:01:35","UptimeSec":95,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"Blanca","BSSId":"00:1A:70:5D:D4:A6","Channel":10,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:05"}}
19:48:58 RSL: sonoff/stat/RESULT = {"POWER":"ON"}
19:48:58 RSL: sonoff/stat/POWER = ON
19:49:09 RSL: sonoff/stat/RESULT = {"BUTTON1":"TOGGLE"}
19:49:09 RSL: sonoff/tele/STATE = {"Time":"2019-11-24T19:49:09","Uptime":"0T00:01:46","UptimeSec":106,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Blanca","BSSId":"00:1A:70:5D:D4:A6","Channel":10,"RSSI":84,"LinkCount":1,"Downtime":"0T00:00:05"}}
19:49:09 RSL: sonoff/stat/RESULT = {"POWER":"OFF"}
19:49:09 RSL: sonoff/stat/POWER = OFF
19:50:19 RSL: sonoff/stat/RESULT = {"BUTTON1":"TOGGLE"}
19:50:19 RSL: sonoff/tele/STATE = {"Time":"2019-11-24T19:50:19","Uptime":"0T00:02:56","UptimeSec":176,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"Blanca","BSSId":"00:1A:70:5D:D4:A6","Channel":10,"RSSI":78,"LinkCount":1,"Downtime":"0T00:00:05"}}
19:50:19 RSL: sonoff/stat/RESULT = {"POWER":"ON"}
19:50:19 RSL: sonoff/stat/POWER = ON
19:51:12 CMD: ShutterRelay1
19:51:12 RSL: sonoff/stat/RESULT = {"Command":"Unknown"}
19:51:17 CMD: SetOption80
19:51:17 RSL: sonoff/stat/RESULT = {"SetOption80":"OFF"}
19:51:20 CMD: SetOption80 1
19:51:20 RSL: sonoff/stat/RESULT = {"SetOption80":"ON"}
19:51:23 CMD: ShutterRelay1
19:51:23 RSL: sonoff/stat/RESULT = {"ShutterRelay1":0}
19:51:30 CMD: ShutterRelay1
19:51:30 RSL: sonoff/stat/RESULT = {"ShutterRelay1":0}
19:51:41 CMD: ShutterRelay1 1
19:51:41 SHT: Relay 1 is 1
19:51:41 SHT: Accuracy digits: 0
19:51:41 SHT: RESET/INIT CALIBRATION MATRIX DIV 0
19:51:41 SHT: Shutter 0 (Relay:1): Init. Pos: 0 [0 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 20000, Opentime 10.0 [s], Closetime 10.0 [s], CoedffCalc: c0: 300, c1 500, c2: 700, c3: 900, c4: 1000, binmask 3, is inverted 0, shuttermode 1
19:51:41 RSL: sonoff/stat/RESULT = {"ShutterRelay1":1}
19:51:49 CMD: ShutterRelay1 0
19:51:49 SHT: Relay 1 is 0
19:51:49 SHT: Accuracy digits: 1
19:51:49 SHT: Shutter 0 (Relay:1): Init. Pos: 0 [0 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 20000, Opentime 10.0 [s], Closetime 10.0 [s], CoedffCalc: c0: 300, c1 500, c2: 700, c3: 900, c4: 1000, binmask 3, is inverted 0, shuttermode 1
19:51:49 RSL: sonoff/stat/RESULT = {"ShutterRelay1":1}
19:51:53 CMD: ShutterRelay1
19:51:53 RSL: sonoff/stat/RESULT = {"ShutterRelay1":1}

ascillato, 2019-11-24:
Ok, you don't want to complete the template. At least, have you tried with latest Tasmota (7.0.0.6)?

ascillato2, 2019-11-24:
Not related to your issue, but I see your sdk version is not the default. That could make you have some wifi issues. If you see wifi or mqtt reconnections, update to latest Tasmota with default sdk and then perform a reset 3 to recalibrate wifi and then power cycle your device (a reboot is not enough, it must be a power cycle).

meingraham, 2019-11-24:
The driver requires the first relay to be odd numbered. Put the shutter on 1&2 and your light on 3.

closed 2019-11-24 by meingraham

ascillato, 2019-11-24:
If the reason that the light was in relay1 was because of the buttons, you can use relay3 for the light as @meingraham explained, and then change the buttons behaviour using rules. If you need assistant with that, please, just ask here or in the chat.