arendst/Tasmota #7019: Suport for Iotty Switch

AFUman, 2019-11-24:
Is there any support for the Iotty Smart Switches or is there already a plan?

Chris

meingraham, 2019-11-24:
New devices can be supported through the Tasmota Template feature. Please refer to the wiki.

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-24 by meingraham

ascillato2, 2019-11-24:
From a google search, there are no internal photos of that device. If that device has an ESP8266 chip, it might be supported by Tasmota. If does not have an ESP8266, it won't be supported.

AFUman, 2019-12-25:
I have make Picture from the Iotty Switch this has a ESP-12S and a 
CY8C4245PVI-482
1725 0 03
CYP 625977

[image]
[image]
[image]