arendst/Tasmota #7021: SML Driver does not display any values in the WebUI

kugelkopf123, 2019-11-24:
### BUG DESCRIPTION
The SML driver does no longer displays values in the main menu.

### REQUESTED INFORMATION
Something has changed in the Dev branch so that the values aren't in the WebUI anymore.

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Wemos D1 Mini v3
- [ ] Tasmota binary firmware version number used: 
  - [ ] Pre-compiled
  - [ x] Self-compiled
    - [ x] IDE / Compiler used: MS VS 1.40.1 PIO 4.1.0
- [ x] Flashing tools used: USB
- [x ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
00:16:46 RSL: stat/sonoff/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
00:16:46 RSL: stat/sonoff/RESULT = {"Module":{"0":"Generic"}}
00:16:46 RSL: stat/sonoff/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

  ```

- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
00:17:35 CMD: status 0
00:17:35 RSL: stat/sonoff/STATUS = {"Status":{"Module":0,"FriendlyName":["ESP-SWG-TEST"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:17:35 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T00:03:21","StartupUTC":"2019-11-24T23:14:14","Sleep":50,"CfgHolder":4617,"BootCount":3,"SaveCount":10,"SaveAddress":"F9000"}}
00:17:35 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"7.0.0.6(tasmota)","BuildDateTime":"2019.11.25 00:13:24","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX"}}
00:17:35 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WiFi_home",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
00:17:35 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":589,"Free":412,"Heap":24,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000407","87DAE397","043E83A1","22B617CD","01001BC0","00007889"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34,53"}}
00:17:35 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-6579","IPAddress":"192.168.1.70","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"84:F3:EB:E3:99:B3","Webserver":2,"WifiConfig":4}}
00:17:35 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E399B3","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
00:17:35 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sun Nov 24 23:17:35 2019","Local":"Mon Nov 25 00:17:35 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:13","Sunset":"17:00"}}
00:17:35 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-11-25T00:17:35","Strom":{"w1":0.0000,"Power_curr":0,"Total_in":0.0000},"Wasser":{"Count":0.000},"Gas":{"Count":0.000},"Regen":{"Count":0}}}
00:17:35 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-11-25T00:17:35","Uptime":"0T00:03:21","UptimeSec":201,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"Wifi":{"AP":1,"SSId":"WiFi_home","BSSId":"10:9A:DD:84:A0:07","Channel":6,"RSSI":74,"LinkCount":1,"Downtime":"0T00:00:06"}}}

  ```
- [x ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
00:19:21 RSL: tele/sonoff/SENSOR = {"Time":"2019-11-25T00:19:21","Strom":{"w1":0.0000,"Power_curr":0,"Total_in":0.0000},"Wasser":{"Count":0.000},"Gas":{"Count":0.000},"Regen":{"Count":0}}

  ```

### TO REPRODUCE
Compile the latest Dev version with: 
```
#define USE_SML_M
#ifndef USE_SCRIPT
#define USE_SCRIPT  
#endif
#ifdef USE_RULES
#undef USE_RULES
#endif  
#define USE_SCRIPT_WEB_DISPLAY
#define USE_SCRIPT_JSON_EXPORT
#define USE_EXPRESSION                       
#define SUPPORT_IF_STATEMENT      
```
in user_config_override.h
Add Script to webUI:

```
>D
>B
=>sensor53 r
>M 4
+1,3,s,16,9600,Strom
+2,12,c,0,-80,Wasser
+3,14,c,1,50,Gas
+4,13,c,0,80,Regen
1,77070100010801ff@1000,W1,kWh,w1,4
1,77070100010700ff@1,Aktueller Verbrauch,W,Power_curr,0
1,77070100010800ff@1000,Verbrauch,KWh,Total_in,4
2,1-0:1.8.0*255(@1000,Zählerstand,cbm,Count,3
3,1-0:1.8.0*255(@100,Zählerstand,cbm,Count,3
4,1-0:1.8.0*255(@1,Impulse,,Count,0
4,=h==
#
```

### EXPECTED BEHAVIOUR
As in Version 6.7.1, you can see the values on the Mainpage from the WebUI (see Screenshots from 6.7.1 White Background)

### SCREENSHOTS
[image]
[image]

### ADDITIONAL CONTEXT
---

gemu2015, 2019-11-25:
USE_SCRIPT_WEB_DISPLAY is broken.  don't define this at the moment. i currently have no time to evaluate it further, sorry. may be next week or so.

gemu2015, 2019-11-25:
ok, is an mqtt_data buffer overflow.
@arendst 
HTTP_SCRIPT_ROOT gets to large with new function lc()
larger buffer or split ?

arendst, 2019-11-25:
I'll update it now. Pls hold on.

arendst, 2019-11-25:
Pls retry.

gemu2015, 2019-11-25:
works, fixed!

kugelkopf123, 2019-11-26:
Yes it works! Thanks!

closed 2019-11-26 by kugelkopf123