arendst/Tasmota #7023: Hidden SSID with Sonoff TX T2UK2C

Ruddyrum, 2019-11-25:
### ISSUE DESCRIPTION - TROUBLESHOOTING
I have flashed a Sonoff TX T2UK2C with various firmwares, but it is unable to connect to my hidden SSID. I knew many of you will say "unhide your SSID" but I have 7 other devices running Tasmota and are connecting to my network without any issue.

I have flashed both a pre-compiled build of the latest version of Tasmota and used termite to manually send Wifi credentials, and also self compiled and flashed an image with all my credentials pre-loaded. Both result in:
`Connect failed as AP cannot be reached`

I tried SetOption56/57 hoping it would connect as the device is near my AP... no joy either.

STATUS2 = {"StatusFWR":{"Version":"7.0.0.6(tasmota)","BuildDateTime":"2019-11-24T16:40:21","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285"}}

### REQUESTED INFORMATION

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in issues (https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in the wiki (https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [X ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X ] Device used (Sonoff TX T2UK2C)
- [X ] Tasmota binary firmware version number used: 7.0.0.6
  - [ X] Pre-compiled
  - [X ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X ] Flashing tools used: Tasmota-PyFlasher / Atom / Flash ESP
- [X ] Provide the output of this command: ``Backlog Template; Module; GPIO``:
  ```
Configuration output here:
Backlog Template; Module; GPIO
10:31:14 CMD: Backlog Template; Module; GPIO
10:31:14 RSL: RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
10:31:14 RSL: RESULT = {"Module":{"1":"Sonoff Basic"}}
10:31:14 RSL: RESULT = {"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO14":{"0":"None"}}

Jason2866, 2019-11-25:
The wifi behaviour can NOT be changed in Tasmota. How this is managed is done in the used Espressif SDK core. Latest Tasmota use Arduino Core 2.6.1 with SDK 2.2.2-dev(38a443e)
Since the SDK is closed source there can nothing be done in Tasmota code.

Ruddyrum, 2019-11-25:
Odd that it isn't a problem on my other devices, only this one... but thanks for the info!

meingraham, 2019-11-25:
What is the newest version on a device that works? Please post `status 2` for that device.

Ruddyrum, 2019-11-25:
Last sonoff Mini:
STATUS2 = {"StatusFWR":{"Version":"6.7.1(basic)","BuildDateTime":"2019-10-26T13:20:09","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}

Last Sonoff Basic
STATUS2 = {"StatusFWR":{"Version":"6.7.1(sonoff)","BuildDateTime":"2019-10-26T13:19:13","Boot":7,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}

Both Work fone with my hidden SSID

arendst, 2019-11-25:
There were some changes in wifi connection code between 6.7.1 and the latests version.

If you compile yourself do the following.

- Using VCS search for "WifiSetMode"
- You'll find 5 references
- Go to the 4 function call references (like "WifiSetMode(WIFI_STA)") and disable function WifiSetMode and re-enable the line just above it (like "WiFi.mode(WIFI_STA)");
- Re-compile and test

Ruddyrum, 2019-11-25:
Thanks, was hoping you were on to something there, but no joy!
`  WiFi.mode(WIFI_STA);      // Disable AP mode
//  WifiSetMode(WIFI_STA);`

re-compiled and flashed:
`16:44:25 WIF: Connect failed as AP cannot be reached`

As soon as SSID Broadcast is enabled:
`Connected`

arendst, 2019-11-25:
What is the status of SetOption56 and SetOption57

Ruddyrum, 2019-11-25:
They were both set to 0 (so off) as I thought networks have to be visible to use this feature. I had also tried them during a previous troubleshooting step with no luck.

However, I have just re-enabled SetOption56 & 57 with WiFi.mode(WIFI_STA) and it still won't connect to my hidden SSID

arendst, 2019-11-25:
I'm out of ideas for now.

JirkaS80, 2019-11-27:
Hello,
I had the same problem (sonoff basic, ESP01, ESP01s).
It helped me setup in Arduino IDE tab
Tools - Esspresif FW: "nonos-sdk 2.2.1 (legacy)"
I am sorry for my English.

nagyrobi, 2019-11-27:
Hiding the SSID is a bad practice anyway. Zero security and autoscan/autochannel selection on neighboring access points can go wrong because of it -> extra source of interference -> less reliable network.

meingraham, 2019-11-27:
In this case, the reason for hiding the AP is not security related.

olewales, 2019-11-27:
This might be related to issue I've encountered recently but did not get to reporting it yet. Is your network by any chance set to 802.11n-only mode?

Ruddyrum, 2019-11-30:
> This might be related to issue I've encountered recently but did not get to reporting it yet. Is your network by any chance set to 802.11n-only mode?

No, the option where that would be is set to 'auto' (other options would be N and Legacy)

effelle, 2019-12-20:
Some time ago I got a Sonoff Basic R2 refusing to connect on N, even if not hidden, but when I changed my AP setting to G/N it started to work as expected.
Since was the only device I had that was acting strange and I had plenty to deploy at the time, I've simply replaced it. I thought was more an hardware issue but now I have some doubts...

jannispinter, 2020-01-17:
I have the same problem since Tasmota 7 on my Sonoff S20 devices. I have all my Tasmota devices on a separate VLAN and provide a separate WiFi network with a hidden SSID for them (it's hidden for convenience). It worked well with Tasmota 6 and previous versions.

Is there any way to report this to the upstream SDK vendor?

ascillato, 2020-01-17:
> Is there any way to report this to the upstream SDK vendor?

yes, in the arduino esp8266 core (https://github.com/esp8266/Arduino)

ascillato2, 2020-01-22:
Closing this issue as it arduino core related. If someone finds a solution from Tasmota code, we can reopen this. Thanks everyone for sharing your ideas.

closed 2020-01-22 by ascillato2