arendst/Tasmota #7024: Bricked OBI Wifi Socket after update to 6.7.1

rucksman, 2019-11-25:
I tried to update two OBI Wifi Sockets from 6.6.0 to 6.7.1 (OTA update with sonoff-DE.bin). The process finished smoothly for the first socket like all the updates I did earlier.

But the update for the second device went completetly wrong. After starting des update process nothing happend anymore. Even after half an hour I was not able to reach the socket via Wifi. No LED was on, no reaction after using the button, regardless which clicking pattern I used. Then I unplugged the socket for 2 days, plugged it in again, but no Change in behaviour. No LED, no reaction after using the button, not reachable via Wifi.

Can I revive the socket somehow? If yes, how? Thans for helping!

Jason2866, 2019-11-25:
Erase the flash via serial and flash it again.

ascillato2, 2019-11-25:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you.

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-25 by ascillato2