arendst/Tasmota #7027: Help with publish domoticz/in

turin400, 2019-11-25:
Good afternoon, I don't know if it can be done or I'm doing something wrong, I'm trying to update a domotiz user variable from a sonoff RFfbridge.

I write:

publish domoticz/in {"command": "setuservariable", "idx": 7, "value": "15" }

and I receive:

17:33:47 CMD: publish domoticz/in {"command": "setuservariable", "idx": 7, "value": "15" }
17:33:47 MQT: domoticz/in = {"command":

and I receive in domoticz:

 Error: MQTT: Invalid data received!

I'm testing all the commands on this page: https://www.domoticz.com/wiki/MQTT

and only this one works:

publish domoticz/in {"idx":24,"svalue":"AAA6060338A0A4F34055"}

I am using an unmodified binary.
Program Version | 7.0.0.6(63b0010-tasmota)

Greetings and thanks for everything.

meingraham, 2019-11-25:
Our ability to provide assistance is greatly hampered since you did not take a few minutes to complete the  [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) with the requested information. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

turin400, 2019-11-25:
I leave the template filled:

### ISSUE DESCRIPTION - TROUBLESHOOTING

_Explained in the message above

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in issues (https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Sonoff Bridge Module
- [ ] Tasmota binary firmware version number used: Program Version | 7.0.0.6(63b0010-tasmota)
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ x] Flashing tools used: web firmware upgrade
- [ x] Provide the output of this command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
18:34:16 CMD: Backlog Template; Module; GPIO
18:34:16 MQT: stat/RFBridge433/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":1,"BASE":18}
18:34:17 MQT: stat/RFBridge433/RESULT = {"Module":{"25":"Sonoff Bridge"}}
18:34:17 MQT: stat/RFBridge433/RESULT = {"GPIO2":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO12":{"0":"None"},"GPIO14":{"0":"None"}}
18:34:23 MQT: tele/RFBridge433/STATE = {"Time":"2019-11-25T18:34:23","Uptime":"2T04:00:16","UptimeSec":187216,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"Wifi":{"AP":1,"SSId":"xxxxxxxx","BSSId":"xxxxxxxxxxxx","Channel":13,"RSSI":100,"LinkCount":2,"Downtime":"0T00:00:38"}}

  ```
- [x ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
18:36:04 CMD: Backlog Rule1; Rule2; Rule3
18:36:04 MQT: stat/RFBridge433/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":338,"Rules":"on System#Boot do rule2 0 endon on Mqtt#Connected do backlog rfraw 166; delay 100; rule2 1 endon on Mqtt#Disconnected do rule2 0 endon on Wifi#Disconnected  do rule2 0 endon"}
18:36:05 MQT: stat/RFBridge433/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Free":439,"Rules":"on RfRaw#Data do publish domoticz/in {"idx":24,"svalue":"%value%"} endon"}
18:36:05 MQT: stat/RFBridge433/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

  ```
- [ x] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:

18:37:18 CMD: STATUS 0
18:37:18 MQT: stat/RFBridge433/STATUS = {"Status":{"Module":25,"FriendlyName":["RFBridge433"],"Topic":"RFBridge433","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":6,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:37:18 MQT: stat/RFBridge433/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin","RestartReason":"Software/System restart","Uptime":"2T04:03:11","StartupUTC":"2019-11-23T13:34:07","Sleep":50,"CfgHolder":4617,"BootCount":83,"SaveCount":421,"SaveAddress":"F8000"}}
18:37:18 MQT: stat/RFBridge433/STATUS2 = {"StatusFWR":{"Version":"7.0.0.6(63b0010-tasmota)","BuildDateTime":"2019-11-23T15:00:09","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285"}}
18:37:18 MQT: stat/RFBridge433/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ATICO","IOT"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000008","00000000"]}}
18:37:18 MQT: stat/RFBridge433/STATUS4 = {"StatusMEM":{"ProgramSize":561,"Free":440,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
18:37:18 MQT: stat/RFBridge433/STATUS5 = {"StatusNET":{"Hostname":"RFBridge433","IPAddress":"xxxxxxxx","Gateway":"xxxxxxx","Subnetmask":"255.255.255.0","DNSServer":"xxxxxxx","Mac":"xxxxxxx","Webserver":2,"WifiConfig":4}}
18:37:18 MQT: stat/RFBridge433/STATUS6 = {"StatusMQT":{"MqttHost":"xxxxxxx","MqttPort":1883,"MqttClientMask":"RFBridge433","MqttClient":"RFBridge433","MqttUser":"RFBridge433","MqttCount":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
18:37:18 MQT: stat/RFBridge433/STATUS7 = {"StatusTIM":{"UTC":"Mon Nov 25 17:37:18 2019","Local":"Mon Nov 25 18:37:18 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"08:13","Sunset":"17:00"}}
18:37:18 MQT: stat/RFBridge433/STATUS10 = {"StatusSNS":{"Time":"2019-11-25T18:37:18"}}
18:37:18 MQT: stat/RFBridge433/STATUS11 = {"StatusSTS":{"Time":"2019-11-25T18:37:18","Uptime":"2T04:03:11","UptimeSec":187391,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"Wifi":{"AP":1,"SSId":"xxxxxxx","BSSId":"xxxxxxx","Channel":13,"RSSI":100,"LinkCount":2,"Downtime":"0T00:00:38"}}}

  ```
[... cut ...] ur issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
18:39:05 CMD: publish domoticz/in {"command": "setuservariable", "idx": 7, "value": "15" }
18:39:05 SRC: WebConsole from 192.168.200.2
18:39:05 CMD: Group 0, Index 1, Command "PUBLISH", Data "domoticz/in {"command": "setuservariable", "idx": 7, "value": "15" }"
18:39:05 MQT: domoticz/in = {"command":

  ```

**(Please, remember to close the issue when the problem has been addressed)**

turin400, 2019-11-29:
Already solved, the problem was the spaces in the sentence.
Thank you.

This one does work:
publish domoticz/in {"command":"setuservariable","idx":7,"value":"prueba"}

closed 2019-11-29 by turin400