arendst/Tasmota #7035: Shutters rare problem

jlpetracca, 2019-11-26:
Hello everyone 

I try this with two versions:
- Program Version | 6.7.1(sonoff)
- 7.0.x

**i was trying to do this**
`Normal Operation (ShutterMode: 0) - First relay: OFF/DOWN, Second relay: OFF/UP
Backlog PulseTime1 0; PulseTime2 0
Backlog Interlock 1,2; Interlock ON (Interlocked relay pair)`
https://github.com/arendst/Tasmota/wiki/Blinds-and-Shutters

I dont know If i am doing something wrong, or understood wrong...

- FIRST CASE :  I had to activate shutterrelay1 1, because it wasnt on default

`PulseTime1 0
PulseTime2 0
Backlog Interlock 1,2
Interlock ON`

Shutter was only working over relay 1, and relay 2 wasnt working.
[
[image]
](url)

- SECOND CASE :  

`Shutterelay1 1
Shutterrelay2 1
PulseTime1 0
PulseTime2 0
Backlog Interlock 1,2
Interlock ON`

There were two shutters working together in the same.

- **THIS WAS THE SOLUTION**

`Shutterelay1 1
Shutterrelay2 1
Shutterrelay2 0
PulseTime1 0
PulseTime2 0
Backlog Interlock 1,2
Interlock ON`

> **GUIDE**
>
> This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum!
>
> Please DO NOT OPEN AN ISSUE:
>  - If you have general questions or you need help on Tasmota usage, go to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/
>  - If your issue is about a new device, please use the Tasmota [Template](../wiki/Templates) feature.
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Tasmota/wiki/FAQ) and troubleshooting wiki articles
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### ISSUE DESCRIPTION - TROUBLESHOOTING
_A clear description of what the issue is and be as extensive as possible_

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in issues (https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of this command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

  ```

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-11-26:
>THIS WAS THE SOLUTION:
> Shutterelay1 1 Shutterrelay2 1 Shutterrelay2 0 PulseTime1 0 PulseTime2 0 Backlog Interlock 1,2 Interlock ON

Great that you found the solution.

ascillato2, 2019-11-26:
Anyway, as explained in the wiki at commands(https://github.com/arendst/Tasmota/wiki/Commands), `ShutterRelay` only accepts ODD NUMBER, so `shutterrelay2` won't be accepted as a command.

ascillato2, 2019-11-26:
Have you tried `reset 6` so as to start from scratch and use your first approach?

jlpetracca, 2019-11-26:
> Have you tried reset 6 so as to start from scratch and use your first approach?
where i the wiki?

You said, i should try this?

> Shutterelay1 1 
> Shutterrelay2 1 
> reset 6
> PulseTime1 0 
> PulseTime2 0 
> Backlog Interlock 1,2 
> Interlock ON

ascillato, 2019-11-26:
Reset 6 command will erase all your settings except for mqtt and wifi credentials, so you can start from scratch (with all settings to default) and you can try your first approach. 

Please, in the console type:
```
Reset 6
```
Wait for restart, then config your **module type** again and your **topic**. 

Then type in the console:
```
Interlock 1,2
Interlock on
Shutterrelay1 1
Restart 1
```

jlpetracca, 2019-11-27:
I did this:

> Interlock 1,2
> Interlock on
> setoption80 1
> Shutterrelay1 1
> Restart 1

OR

> setoption80 1
> Shutterrelay1 1
> Interlock 1,2
> Interlock on
> Restart 1

OR

> setoption80 1
> Shutterrelay1 1
>PulseTime1 0
>PulseTime2 0
> Interlock 1,2
> Interlock on
> Restart 1

WORKS PERFECTLY LIKE THIS
> Normal Operation (ShutterMode: 0) - First relay: OFF/DOWN, Second relay: OFF/UP 

But the problem was when i activate the shutterrelay2, now i deactivate 

Now that understand the problem, i might think that this explanation

> <value>
> 0 = disable this and all higher numbered shutters
> 1,3,5,7,... (must be an odd number) = define the Relay<value> component used to open the shutter. This relay's mate, the next higher numbered relay, closes the shutter. Depending on the shutter mode, the relays may need to be interlocked using the Interlock command.
> The ShutterRelay command must be executed first before any other shutter commands for Shutter<x> can be executed.

Might be a little bit difficult to understand properly.

> <value>
> 0 = disable this and all higher numbered shutters
>2,4,6,8,...(even numbers) = are not an available commands
> 1,3,5,7,... (must be an odd number) = define the Relay<value> component used to open the shutter. This relay's mate, the next even numbered relay, closes the shutter. (For ex. If you are using ShutterRelay1 Relay1 will open the shutter and Relay2 will be closing the shutter). Depending on the shutter mode, the relays may need to be interlocked using the Interlock command.
> The ShutterRelay command must be executed first before any other shutter commands for Shutter<x> can be executed.

closed 2019-11-27 by jlpetracca