arendst/Tasmota #7036: Blitzwolf BW-SHP7 supported?

iLLiac4, 2019-11-26:
Hi. Just a shot question. Is Blitzwolf BW-SHP7 supported in tasmota? I can not find anything related to W
BW-SHP7.

TNX

ascillato, 2019-11-26:
https://blakadder.github.io/templates/blitzwolf_BW-SHP7.html

ascillato2, 2019-11-26:
Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-26 by ascillato2

meingraham, 2019-11-26:
SHP7 v2 template awaiting merge. If the first doesn work, you may have a newer SHP7 so try this one.

{"NAME":"Blitzwolf SHP7v2","GPIO":[17,158,57,131,134,132,0,0,18,56,21,0,22],"FLAG":0,"BASE":45}