arendst/Tasmota #7038: problem with pzem004t v3.0 on 6.7.1

gumileeev, 2019-11-26:
I use tasmota firmware on the nodemcu together with the pzem004t v3.0 module. I have 2 of these modules, they are installed in different places (apartment and country house). Previously, I had firmware 6.6.0 and there were no problems, here is the instruction by which I installed: https://letscontrolit.com/forum/viewtopic.php?t=2595&start=100#p38310 . About 5 days ago, I installed update 6.7. And at first glance everything was fine until today I saw an increase on both counters of 12,000 kilowatts yesterday (not watts, just kilowatts) !!! I attach 2 screenshots. On the screen with an IP address of 4.8 - today from midnight until noon (12-13 hours) I continuously used 2 heaters of 1.5 kW and 0.5 kW, it turns out that today I had to run about 25 kW, the meter sees the load, but the consumption for today shows only 2.09 kW Please find a problem and release a fix, otherwise I don’t know what to do.
[image]
[image]

Jason2866, 2019-11-26:
Connect it via hardware serial as described in wiki. https://github.com/arendst/Tasmota/wiki/PZEM-0XX
Use template `{"NAME":"PZEM004TV3","GPIO":[0,62,0,98,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}`

gumileeev, 2019-11-26:
> Connect it via hardware serial as described in wiki. https://github.com/arendst/Tasmota/wiki/PZEM-0XX
> Use template `{"NAME":"PZEM004TV3","GPIO":[0,62,0,98,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}`

In my first post, I provided a link to my post where I wrote that I used this particular template, and earlier on 6.6.0 there were no problems.

Jason2866, 2019-11-26:
Our ability to provide assistance is greatly hampered since you did not take a few minutes to complete the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) with the requested information. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.
 
Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).
 
Thanks.
 
### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato, 2019-11-26:
The link https://letscontrolit.com/forum/ belongs to ESPEASY firmware, not Tasmota. Sorry. As you are using Tasmota, please address this issue in the Tasmota Support Chat. Thanks.

closed 2019-11-26 by ascillato2

gumileeev, 2019-11-26:
> The link https://letscontrolit.com/forum/ belongs to ESPEASY firmware, not Tasmota. Sorry. As you are using Tasmota, please address this issue in the Tasmota Support Chat. Thanks.

This link to my post is about how I configured Tasmota, because espeasy does not have such a plugin, I gave the link so that you can see how pzem004t v3.0 was registered in tasmota. I attached screenshots on which it is clear that I have problems with tasmota. The problem is not solved, and the topic is closed.
This link to my post is about how I configured Tasmota, because espeasy does not have such a plugin, I gave the link so that you can see how pzem004t v3.0 was registered in tasmota. I attached screenshots on which it is clear that I have problems with tasmota. The problem is not solved, and the topic is closed.

meingraham, 2019-11-26:
@gumileeev - This has been mentioned a few times...
> please address this issue in the Tasmota Support Chat
> See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

 Some troubleshooting is required to determine if configuration change may correct it, or if it can be determined definitively whether it's a bug and if so, capture more information to possibly help find it.