arendst/Tasmota #7046: MHCOZY Relay "clone"?

Off-Load, 2019-11-27:
Got two "MHCOZY" singel channel relay in the mail today, and was prepared to flash them with Tasmota, but when I got a closer look at them, the pinout was different, from what I've seen online, and I could not find any pinout diagram for the RX,TX, and 3,3 + GND on this board. 
Could anyone point out where RX,TX, and 3,3 + GND is?
[image]

Thanks :)

ascillato2, 2019-11-28:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-28 by ascillato2

sonnyg77, 2020-01-18:
Hello Off-load

were you able to get any assistance on the location of the pinouts for this board.
I have just received one of these boards and was surprised that it was completely different from the old board,  ive no clue how to start flashing it
Looking around on google i think i can see the location of the TX & RX  but dont know where to put the GND and 3.3v from the FTDI
Any help would be appreciated.
Thank you

Off-Load, 2020-01-18:
Yes, and no :) I did post in the discord chat, and did receive an hint about this thread [https://github.com/arendst/Tasmota/issues/453](url) look 30% down that tread an you find a picture of the same board.
The TX&RX is on the chip itself, but I gave up, and ordered a couple of D1 minis, Nodemcu, and some relays, and the MHCOZY ended up in the box for spare parts.

You could try to power it from the micro usb, to see if the 5v and ground from the usb is converted to 3.3v, and just connect the  TX & RX from the FTDI, to the chip. Like this guy did [https://www.hagensieker.com/wordpress/2019/02/21/hacking-unknown-2-channel-relay-with-tasmota/](url)

Good luck

sonnyg77, 2020-01-18:
Thank you for your prompt reply Off-Load ,  its been a while since i had to do anything like this and kind of lost touch with it all,  i got the unit expecting it to be easy like the last one i got !  wrong...lol    to save the headache ive just put it on ewee link.  It not used for anything important,  just to reboot a wifi router when the need arises,   for the odd occasion that needs to happen having it on ewee will do just fine,  doesnt really need to be incorporated into my smarthome setup.

Off-Load, 2020-01-18:
No problem. I was also thinking this unit was an easy thing to integrate  to my smart home, but it failed big time.
It's better to go for some boards like NodeMCU, or D1 mini in the future to avoid more frustration, divorce, or worse :) I bought them because some guy said they were easy to flash with Tasmota, but I received the same board you have now.
As said, I'm not going to waste more time on these boards, to much hassle, when there is alternatives, with an price that are cheaper.
If you want to have an watchdog for your router, there is an option in most "modern/old" routers to do that. No need for eWeLink. :)