arendst/Tasmota #7049: Rules can't be switched to 1 (on)

mijnmodelbaan, 2019-11-28:
> **GUIDE**
>
> This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
>
> Please DO NOT OPEN AN ISSUE:
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/
>  - If you have a issue when flashing was done via Tuya Convert
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Tasmota/wiki/FAQ) and troubleshooting wiki articles
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
After upgrading from version 6.6.0.18 to 7.0.0.4 I noticed my rules weren't working anymore. After some searching I noticed all rules were set to OFF. Trying to switch them on results in a restart of the machine with: INFO3 = {"RestartReason":"Hardware Watchdog"}

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Tuya Shutter Switch (Belkin WeMo emulation)
- [x] Tasmota binary firmware version number used: 7.0.0.6
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: VSC with PlatformIO
- [x] Flashing tools used: OTA
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
14:48:08 CMD: Backlog Template; Module; GPIO
14:48:08 RSL: RESULT = {"NAME":"Tuya Shutter","GPIO":[157,0,54,10,22,19,0,0,17,21,53,23,52],"FLAG":0,"BASE":18}
14:48:08 RSL: RESULT = {"Module":{"0":"Tuya Shutter"}}
14:48:08 RSL: RESULT = {"GPIO":"Not supported"}

  ```
- [x] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
14:49:19 CMD: Backlog Rule1; Rule2; Rule3
14:49:19 RSL: RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":36,"Rules":"on EVENT#DIEF do backlog var1 %mem1%; event HOOG; event WAIT=DNTI endon on EVENT#HALF do backlog var1 %mem3%; event HOOG; event WAIT=DNTI endon on EVENT#KIER do backlog var1 %mem2%; event LAAG; event WAIT=UPTI endon on EVENT#GAAT do backlog var1 %mem4%; event LAAG; event WAIT=UPTI endon on EVENT#HOOG do backlog Power2 1; delay 3; Power1 1 endon on EVENT#LAAG do backlog Power2 1; delay 3; Power3 1 endon on EVENT#STOP do backlog Pow ...
14:49:20 RSL: RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":75,"Rules":"on CLOCK#Timer=1 do event DIEF endon on CLOCK#Timer=2 do event KIER endon on CLOCK#Timer=3 do event LAAG endon on CLOCK#Timer=4 do event HOOG endon on CLOCK#Timer=5 do event DIEF endon on CLOCK#Timer=6 do event KIER endon on CLOCK#Timer=7 do event LAAG endon on CLOCK#Timer=8 do event HOOG endon on CLOCK#Timer=9 do event GAAT endon on System#Save do Publish %var9% Restart endon  on System#Boot do var9 TASMOTA/4B67B6/stat/RESULT end ...
14:49:20 RSL: RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":219,"Rules":"on EVENT#DNTI do backlog Power3 1; delay %var1%; Power3 0; Publish %var9% FINI endon on EVENT#UPTI do backlog Power1 1; delay %var1%; Power1 0; Publish %var9% FINI endon on EVENT#WAIT do backlog delay %mem5%; event %value% endon on EVENT#AUTO do Timers 1 endon on EVENT#TOFF do Timers 0 endon"}

  ```
- [x] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
14:50:24 CMD: Status 0
14:50:24 RSL: STATUS = {"Status":{"Module":0,"FriendlyName":["TASMOTA-4B67B6","TASMOTA-4B67B6","TASMOTA-4B67B6"],"Topic":"4B67B6","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":900,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:50:24 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"ROLLOS","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:26:17","StartupUTC":"2019-11-28T13:24:07","Sleep":50,"CfgHolder":5727,"BootCount":54,"SaveCount":162,"SaveAddress":"F5000"}}
14:50:24 RSL: STATUS2 = {"StatusFWR":{"Version":"7.0.0.6(basic)","BuildDateTime":"2019-11-28T13:37:46","Boot":7,"Core":"2_6_2","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX"}}
14:50:24 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["vanbergenfamily","vanbergenfamily"],"TelePeriod":150,"Resolution":"55A180C0","SetOption":["0400C002","2805C8000100060000005A64000000000000","000000C1","00000000"]}}
14:50:24 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":450,"Free":552,"Heap":29,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashMode":255,"Features":["00000809","8F002383","041083A1","20021702","010000C0","00000000"],"Drivers":"1,2,4,9,10,16,20,21","Sensors":""}}
14:50:24 RSL: STATUS5 = {"StatusNET":{"Hostname":"4B67B6-1974","IPAddress":"192.168.45.63","Gateway":"192.168.45.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"EC:FA:BC:4B:67:B6","Webserver":2,"WifiConfig":2}}
14:50:24 RSL: STATUS7 = {"StatusTIM":{"UTC":"Thu Nov 28 13:50:24 2019","Local":"Thu Nov 28 14:50:24 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"08:19","Sunset":"16:36"}}
14:50:24 RSL: STATUS10 = {"StatusSNS":{"Time":"2019-11-28T14:50:24","Switch2":"ON"}}
14:50:24 RSL: STATUS11 = {"StatusSTS":{"Time":"2019-11-28T14:50:24","Uptime":"0T00:26:17","UptimeSec":1577,"Vcc":3.439,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":32,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"vanbergenfamily","BSSId":"EC:55:F9:37:3A:89","Channel":11,"RSSI":24,"LinkCount":1,"Downtime":"0T00:00:07"}}}

  ```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
[... 19 lines cut ...]
00:00:06 RSL: POWER2 = OFF
00:00:06 RSL: RESULT = {"POWER3":"OFF"}
00:00:06 RSL: POWER3 = OFF
00:00:10 RSL: STATE = {"Time":"1970-01-01T00:00:10","Uptime":"0T00:00:12","UptimeSec":12,"Vcc":3.432,"Heap":31,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"vanbergenfamily","BSSId":"EC:55:F9:37:3A:89","Channel":11,"RSSI":26,"LinkCount":1,"Downtime":"0T00:00:07"}}
00:00:10 RSL: SENSOR = {"Time":"1970-01-01T00:00:10","Switch2":"ON"}

  ```

### TO REPRODUCE
_Steps to reproduce the behaviour:_

Every time I try to switch on a rule it restarts and the rule is still switched off.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Rules should be switched on, like before the upgrade.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-11-28:
Hi,

You are having issues with your firmware. See `00:00:06 RSL: INFO3 = {"RestartReason":"Hardware Watchdog"}`

So, as your device is experimenting problems at boot time, it starts disabling features as a **boot loop protection**.

Please, try:

1- full flash erase with esptool.py as explained in the wiki
2- flash the precompiled bins (http://thehackbox.org/tasmota/tasmota.bin)
3- config again and try your rules.

_(you have started to test core 2.6.2 that was released yesterday. We didn't include it in platformio.ini yet because it needs more testing)_

If you need further assistance, please, address to the Tasmota Support Chat. Thanks.

closed 2019-11-28 by ascillato2

mijnmodelbaan, 2019-11-30:
Thanks ascillato2 for your reply.

A full-flash-erase was easier written than done in this case, so I OTA'd the tasmota-minimal.bin first and then the tasmota.bin after the restart and that fixed the problem for now.
I'll do a full-flash-erase later, just to be sure.

Thanks again!

ascillato, 2019-11-30:
At this point, you can try reset 3 in the console for flash cleaning. Then, after the restart, do a power cycle.