arendst/Tasmota #7050: Failures in mqtt power update report

machadof1, 2019-11-28:
> **GUIDE**
>
> This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
>
> Please DO NOT OPEN AN ISSUE:
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/
>  - If you have a issue when flashing was done via Tuya Convert
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Tasmota/wiki/FAQ) and troubleshooting wiki articles
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
_A clear and concise description of what the bug is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Blitzwolf SP1 /Digoo NX-SP202 _____
- [ ] Tasmota binary firmware version number used: 6.7.1_____
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _NodeMCU PyFlasher____
- [X] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
16:17:28 MQT: test/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":1,"BASE":18}
16:17:28 MQT: test/stat/RESULT = {"Module":{"55":"Gosund SP1 v23"}}
16:17:29 MQT: test/stat/RESULT = {"GPIO":"Not supported"}

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
16:18:52 CMD: Status 0
16:18:52 MQT: test/stat/STATUS = {"Status":{"Module":55,"FriendlyName":["Sonoff"],"Topic":"test","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
16:18:52 MQT: test/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Software/System restart","Uptime":"0T00:11:44","StartupUTC":"2019-11-28T16:07:08","Sleep":50,"CfgHolder":4617,"BootCount":10,"SaveCount":45,"SaveAddress":"F9000"}}
16:18:52 MQT: test/stat/STATUS2 = {"StatusFWR":{"Version":"6.7.1(sonoff)","BuildDateTime":"2019-10-26T13:19:13","Boot":4,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
16:18:52 MQT: test/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0020A029","2805C8000100060000005AFF000000000000","00000008"]}}
16:18:52 MQT: test/stat/STATUS4 = {"StatusMEM":{"ProgramSize":564,"Free":440,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","0000F081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,30","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
16:18:52 MQT: test/stat/STATUS5 = {"StatusNET":{"Hostname":"Switch 2","IPAddress":"","Gateway":"","Subnetmask":"255.255.255.0","DNSServer":"","Mac":"","Webserver":2,"WifiConfig":4}}
16:18:52 MQT: test/stat/STATUS6 = {"StatusMQT":{"MqttHost":"xxx.xxx.xxx.xxx","MqttPort":1883,"MqttClientMask":"test","MqttClient":"xxx","MqttUser":"xxxxx","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
16:18:52 MQT: test/stat/STATUS7 = {"StatusTIM":{"UTC":"Thu Nov 28 16:18:52 2019","Local":"Thu Nov 28 16:18:52 2019","StartDST":"Sun Mar 31 01:00:00 2019","EndDST":"Sun Oct 27 02:00:00 2019","Timezone":99,"Sunrise":"07:18","Sunset":"15:58"}}
16:18:52 MQT: test/stat/STATUS9 = {"StatusPTH":{"PowerDelta":20,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
16:18:52 MQT: test/stat/STATUS10 = {"StatusSNS":{"Time":"2019-11-28T16:18:52","ENERGY":{"TotalStartTime":"2019-04-11T09:01:30","Total":0.106,"Yesterday":0.000,"Today":0.106,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":233,"Current":0.000}}}
16:18:52 MQT: test/stat/STATUS11 = {"StatusSTS":{"Time":"2019-11-28T16:18:52","Uptime":"0T00:11:44","UptimeSec":704,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"","BSSId":"","Channel":1,"RSSI":48,"LinkCount":1,"Downtime":"0T00:00:10"}}}

  ```

[... 40 lines cut ...]
16:43:05 MQT: test/tele/SENSOR = {"Time":"2019-11-28T16:43:05","ENERGY":{"TotalStartTime":"2019-04-11T09:01:30","Total":0.217,"Yesterday":0.000,"Today":0.217,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":233,"Current":0.000}}
16:43:32 MQT: test/stat/RESULT = {"POWER":"ON"}
16:43:32 MQT: test/stat/POWER = ON (retained)

**mqtt message only after 3 minutes and 57 seconds**

16:47:29 MQT: test/tele/STATE = {"Time":"2019-11-28T16:47:29","Uptime":"0T00:40:21","UptimeSec":2421,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"","BSSId":"","Channel":1,"RSSI":44,"LinkCount":1,"Downtime":"0T00:00:10"}}
16:47:29 MQT: test/tele/SENSOR = {"Time":"2019-11-28T16:47:29","ENERGY":{"TotalStartTime":"2019-04-11T09:01:30","Total":0.290,"Yesterday":0.000,"Today":0.290,"Period":84,"Power":1030,"ApparentPower":1030,"ReactivePower":0,"Factor":1.00,"Voltage":220,"Current":4.684}}
16:47:54 MQT: test/tele/SENSOR = {"Time":"2019-11-28T16:47:54","ENERGY":{"TotalStartTime":"2019-04-11T09:01:30","Total":0.298,"Yesterday":0.000,"Today":0.298,"Power":564,"ApparentPower":1051,"ReactivePower":887,"Factor":0.54,"Voltage":225,"Current":4.678}}
16:48:13 MQT: test/stat/RESULT = {"POWER":"OFF"}
16:48:13 MQT: test/stat/POWER = OFF (retained)
16:48:15 MQT: test/tele/SENSOR = {"Time":"2019-11-28T16:48:15","ENERGY":{"TotalStartTime":"2019-04-11T09:01:30","Total":0.301,"Yesterday":0.000,"Today":0.301,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":229,"Current":0.000}}
### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
turn on switch, with powerdelta 20 as soon as a 20% power difference is detected a mqtt message should be sent, even if is right after switch off

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-11-28:
Why do you have so many retained cases? It seems your mqtt setup does not work correctly.

ascillato2, 2019-11-28:
Hi,

in the template you said:

> Pre-compiled

but in `status 0` you have `"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"`

There aren't any precompiled bins with STAGE core. Please, can you re-test using the precompiled bins with latest Tasmota version (http://thehackbox.org/tasmota/tasmota.bin) ?

machadof1, 2019-11-28:
Hi ascillato2, as you requested, I flashed another device with the firmware you indicated, but there are still problems:
21:59:13 MQT: switch_2/stat/RESULT = {"POWER":"ON"}
21:59:13 MQT: switch_2/stat/POWER = ON
21:59:17 MQT: switch_2/tele/STATE = {"Time":"2019-11-28T21:59:17","Uptime":"0T00:00:12","UptimeSec":12,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"","BSSId":"","Channel":1,"RSSI":60,"LinkCount":1,"Downtime":"0T00:00:06"}}
21:59:17 MQT: switch_2/tele/SENSOR = {"Time":"2019-11-28T21:59:17","ENERGY":{"TotalStartTime":"2019-04-11T09:12:36","Total":0.032,"Yesterday":0.000,"Today":0.032,"Period":2,"Power":548,"ApparentPower":548,"ReactivePower":0,"Factor":1.00,"Voltage":226,"Current":2.428}}
21:59:27 MQT: switch_2/tele/SENSOR = {"Time":"2019-11-28T21:59:27","ENERGY":{"TotalStartTime":"2019-04-11T09:12:36","Total":0.034,"Yesterday":0.000,"Today":0.034,"Power":1063,"ApparentPower":1089,"ReactivePower":234,"Factor":0.98,"Voltage":227,"Current":4.806}}
21:59:30 MQT: switch_2/tele/SENSOR = {"Time":"2019-11-28T21:59:30","ENERGY":{"TotalStartTime":"2019-04-11T09:12:36","Total":0.035,"Yesterday":0.000,"Today":0.035,"Power":545,"ApparentPower":1065,"ReactivePower":915,"Factor":0.51,"Voltage":222,"Current":4.798}}
21:59:35 MQT: switch_2/stat/RESULT = {"POWER":"OFF"}
21:59:35 MQT: switch_2/stat/POWER = OFF

** the mqtt message is sent 4 minutes and 42 seconds after the switch off**

22:04:17 MQT: switch_2/tele/STATE = {"Time":"2019-11-28T22:04:17","Uptime":"0T00:05:12","UptimeSec":312,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"","BSSId":"","Channel":1,"RSSI":58,"LinkCount":1,"Downtime":"0T00:00:06"}}
22:04:17 MQT: switch_2/tele/SENSOR = {"Time":"2019-11-28T22:04:17","ENERGY":{"TotalStartTime":"2019-04-11T09:12:36","Total":0.035,"Yesterday":0.000,"Today":0.035,"Period":3,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}

22:11:20 MQT: switch_2/stat/RESULT = {"POWER":"ON"}
22:11:20 MQT: switch_2/stat/POWER = ON

** the mqtt message is sent 2 minutes and 57 seconds after the switch on**

22:14:17 MQT: switch_2/tele/STATE = {"Time":"2019-11-28T22:14:17","Uptime":"0T00:15:12","UptimeSec":912,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"","BSSId":"","Channel":1,"RSSI":54,"LinkCount":1,"Downtime":"0T00:00:06"}}
22:14:17 MQT: switch_2/tele/SENSOR = {"Time":"2019-11-28T22:14:17","ENERGY":{"TotalStartTime":"2019-04-11T09:12:36","Total":0.062,"Yesterday":0.000,"Today":0.062,"Period":27,"Power":547,"ApparentPower":547,"ReactivePower":0,"Factor":1.00,"Voltage":226,"Current":2.423}}
22:17:33 MQT: switch_2/stat/RESULT = {"POWER":"OFF"}
22:17:33 MQT: switch_2/stat/POWER = OFF

22:19:17 MQT: switch_2/tele/STATE = {"Time":"2019-11-28T22:19:17","Uptime":"0T00:20:12","UptimeSec":1212,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"","BSSId":"","Channel":1,"RSSI":60,"LinkCount":1,"Downtime":"0T00:00:06"}}
22:19:17 MQT: switch_2/tele/SENSOR = {"Time":"2019-11-28T22:19:17","ENERGY":{"TotalStartTime":"2019-04-11T09:12:36","Total":0.093,"Yesterday":0.000,"Today":0.093,"Period":30,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}

22:05:46 CMD: status 0
22:05:46 MQT: switch_2/stat/STATUS = {"Status":{"Module":55,"FriendlyName":["Sonoff"],"Topic":"switch_2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:05:46 MQT: switch_2/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Software/System restart","Uptime":"0T00:06:41","StartupUTC":"2019-11-28T20:59:05","Sleep":50,"CfgHolder":4617,"BootCount":9,"SaveCount":30,"SaveAddress":"F8000"}}
22:05:46 MQT: switch_2/stat/STATUS2 = {"StatusFWR":{"Version":"7.0.0.6(c47e111-tasmota)","BuildDateTime":"2019-11-28T22:00:23","Boot":4,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285"}}
22:05:46 MQT: switch_2/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005AFF000000000000","00000000","00000000"]}}
22:05:46 MQT: switch_2/stat/STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
22:05:46 MQT: switch_2/stat/STATUS5 = {"StatusNET":{"Hostname":"Switch 2","IPAddress":"","Gateway":"","Subnetmask":"255.255.255.0","DNSServer":"","Mac":"","Webserver":2,"WifiConfig":4}}
22:05:46 MQT: switch_2/stat/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"","MqttClient":"","MqttUser":"","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
22:05:46 MQT: switch_2/stat/STATUS7 = {"StatusTIM":{"UTC":"Thu Nov 28 21:05:46 2019","Local":"Thu Nov 28 22:05:46 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunris [... cut ...]
22:05:46 MQT: switch_2/stat/STATUS10 = {"StatusSNS":{"Time":"2019-11-28T22:05:46","ENERGY":{"TotalStartTime":"2019-04-11T09:12:36","Total":0.035,"Yesterday":0.000,"Today":0.035,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
22:05:46 MQT: switch_2/stat/STATUS11 = {"StatusSTS":{"Time":"2019-11-28T22:05:46","Uptime":"0T00:06:41","UptimeSec":401,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"","BSSId":"","Channel":1,"RSSI":54,"LinkCount":1,"Downtime":"0T00:00:06"}}}

machadof1, 2019-11-29:
> Why do you have so many retained cases? It seems your mqtt setup does not work correctly.

It was just a test configuration, this is the latest console log
 
12:37:09 MQT: switch_2/stat/RESULT = {"POWER":"ON"}
12:37:09 MQT: switch_2/stat/POWER = ON
12:39:20 MQT: switch_2/tele/STATE = {"Time":"2019-11-29T12:39:20","Uptime":"0T14:40:15","UptimeSec":52815,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"","BSSId":"","Channel":1,"RSSI":54,"LinkCount":1,"Downtime":"0T00:00:06"}}
12:39:20 MQT: switch_2/tele/SENSOR = {"Time":"2019-11-29T12:39:20","ENERGY":{"TotalStartTime":"2019-04-11T09:12:36","Total":0.127,"Yesterday":0.093,"Today":0.035,"Period":0,"Power":954,"ApparentPower":956,"ReactivePower":61,"Factor":1.00,"Voltage":211,"Current":4.524}}
12:39:34 MQT: switch_2/stat/RESULT = {"POWER":"OFF"}
12:39:34 MQT: switch_2/stat/POWER = OFF
12:44:20 MQT: switch_2/tele/STATE = {"Time":"2019-11-29T12:44:20","Uptime":"0T14:45:15","UptimeSec":53115,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"","BSSId":"","Channel":1,"RSSI":54,"LinkCount":1,"Downtime":"0T00:00:06"}}
12:44:20 MQT: switch_2/tele/SENSOR = {"Time":"2019-11-29T12:44:20","ENERGY":{"TotalStartTime":"2019-04-11T09:12:36","Total":0.132,"Yesterday":0.093,"Today":0.039,"Period":4,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}

flashy02, 2019-12-02:
It seems that then STD-Logging time of 300s is set, so you have to edit your config to send the state/sensor value every 30s.

machadof1, 2019-12-02:
> It seems that then STD-Logging time of 300s is set, so you have to edit your config to send the state/sensor value every 30s.

That is not the behavior that is expected, I want reports when there is something to report, not flood the system with mqtt messages all saying the same thing. The "PowerDelta" parameter is for this.

djashjones, 2019-12-16:
Until I downgraded to 6.6.0 as a workaround what I did was when the contacts changed state change the teleperiod to 10 and wait 20 secs then changed it back to 300.

I wish we could change the teleperiod to 1 for these cases or even a command to fire the tele report on sensors like we can for state

machadof1, 2019-12-16:
> Until I downgraded to 6.6.0 as a workaround what I did was when the contacts changed state change the teleperiod to 10 and wait 20 secs then changed it back to 300.
> 
> I wish we could change the teleperiod to 1 for these cases or even a command to fire the tele report on sensors like we can for state

You can use a rule with the publish command like this example [2880](https://github.com/arendst/Tasmota/issues/2880#issuecomment-416078047)

ascillato2, 2019-12-17:
If you power on and off your relay, powerdelta won't fire. That feature only fires when the power changes while the relay is on. 

You can use a rule with the publish command like this example https://github.com/arendst/Tasmota/issues/2880#issuecomment-416078047

closed 2019-12-17 by ascillato2

meingraham, 2019-12-17:
Or another way of saying it... when the circuit is open (relay not latched, power off), there is no current flowing through the circuit being monitored. P=IV. No Current = No Power. Thus no change in Power. No PowerDelta trigger.