arendst/Tasmota #7052: Feit WiFi "Smart" Dimmer Switch (TYWE2S)

backcountrymountains, 2019-11-28:
I bought a couple of Feit dimmer switches from Costco. They use the TYWE2S chip which is soldered to the main board. 

I think the 3.3v, GND, RX and TX are accessible through the holes in the main board, but according to other write-ups of the TYWE2S, I need to connect GPIO0 to ground to enable flashing mode. I don't have access to the back of the TYWE2S without desoldering it. [https://github.com/arendst/Tasmota/wiki/CE-Smart-Home---LA-WF3-Wifi-Plug-(TYWE2S)](url)

Tuya link:
[https://docs.tuya.com/docDetail?code=K8uhknau42nof](url)

[image]
[image]
[image]
[image]
[image]

Has anyone tried to use these? Any advice on flashing without desoldering?

How do I test it and get it working?

Thanks.

ascillato2, 2019-11-28:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-28 by ascillato2

mkormendy, 2020-01-05:
Where did we net out on this on the chat?

backcountrymountains, 2020-01-05:
Got a Pi and did ota tuya convert. Worked out. Went with esp home but I've got flickering. Might try tasmota to get rid of flickering. Probably should've closed this issue, sorry.

mkormendy, 2020-01-05:
I have this flashed with Tasmota, but the dimmer feature isn’t working correctly.

backcountrymountains, 2020-01-05:
My esphome config is here:
https://community.home-assistant.io/t/costco-feit-smart-dimmer-tuya-convert-tasmota/151390

Dimming works but flickers.

Somebody did it with tasmota there.

mkormendy, 2020-01-10:
@backcountrymountains Yeah I followed those instructions with no luck.

qwertangel, 2020-03-13:
this issue is closed. does it mean that it is working fine on new version of tasmota ?

danlat, 2020-04-10:
See the home assistant forums link, this is working with latest Tasmota except for some missed hardware button status updates

qwertangel, 2020-04-11:
> See the home assistant forums link, this is working with latest Tasmota except for some missed hardware button status updates

tasmota is highly unstable. it stops responding at randomly.

BlowOnFuel, 2020-10-04:
I got few of those which also could not be flashed OTA with TuyaConvert.

I took off the TYWE2S with a heat gun and was successful at flashing the TYWE2S and reinstalling them after reloading the contacts.  As there were no apparent traces/contacts under the TYWE2S, I tried poking a hole to gain access to GPIO0 but it still does not seem to enable flashing the ESP with the PCB is mounted on the mainboard.  I suspect the TuyaMCU on TX/RX also prevents access to the lines for flashing.

Hope that helps.

qwertangel, 2020-10-04:
i have given up this dimmer. replaced them with current dimmers.
btw @BlowOnFuel , if you want to flash while it is on board, MCU needs to be disabled to flash tasmota. generally that is done by grounding the reset pin of MCU.

BlowOnFuel, 2020-10-04:
With the gap between the PCBs, I doubt it is really an option to try to poking a hole to gain access to GPIO0 to flash without taking it off.  Without OTA, I feel it is still easier to take the PCB off and flash it wired.

Thanks for the info about disabling the MCU if needed.

Hope OTA will be back soon for these devices.

Any precisions on you alternate option dimmers?

qwertangel, 2020-10-04:
btw, don't hold your breath for OTA. seems like Tuya has really amped up it's defences.

other recent dimmers (treatlife, Martin Jerry) come with TYWE3S which has all pins exposed on sides and mostly well documented (regarding grounding MCU's reset pins) and all settings to make it work.  I have from both brands and they work fine. Avoid other brands (gosund , meross). They have either non esp chip or non standard MCU (non-tuya) which gives several headaches.

BlowOnFuel, 2020-10-05:
I would not have unsoldered the TYWE2S to flash it if I was holding my breath...

I'd probably be looking at dimmer for which I can get the long press or multi-buttons.

qwertangel, 2020-10-05:
then your choices are severely limited. AFAIK that is possible only on MJ dimmers (that I know of) which doesn't use MCU to control dimmer. Most of other dimmers are controlled by MCU and use ESP only as wifi radio and you can't configure long press or multi-press (since buttons are not connected to esp gpio). you may/not like the looks of MJ dimmers though.

magimat, 2020-10-13:
hey, I bough a pack of those FEIT dimmers today and even though it suck that OTA is not working anymore, I was able to flash is using wired method quite easily:

https://community.smartthings.com/t/costco-cheap-feit-smart-dimmer-wifi/208142

at under 40$ CAD for 3 of those, I feel like it's really worth a few soldering :)

NorthernMan54, 2020-11-16:
I just recently acquired several of these as well, and I was amazed at how good the button's feel compared to the 'CE Smart Wifi Dimmers'.  They also received a good WAF score when compared to the CE Smart ones.

The only negative feed I did receive was that the green LED's are too bright.  Did anyone find a method to dim them a little bit?

PS I managed to nab some new old stock, and they were easily converted with Tuya Convert.