arendst/Tasmota #7055: Hue Emulation on Logitech Harmony

nvschilleman, 2019-11-28:
Hi there!

I've been trying to pair my Logitech Harmony Hub to my Sonoff RF switch running on Tasmota with Hue emulation enabled.

The MyHarmony software discovers the emulated Hue Bridge, however, when trying to toggle the Sonoff switch from my Harmony Remote my remote gives the error "No connection". Is there any solution for this problem? The Tasmota web interface on my Sonoff is protected with a password, since MyHarmony tries to connect to the bridge over port 80, without asking for a password. Could this be the issue?

Jason2866, 2019-11-28:
HueEmulation is designed to be used with Alexa hardware.
It may! work with other hardware but this use case is NOT supported

nvschilleman, 2019-11-28:
Thats a bummer... I'd like to control my Sonoff with my Logitech Harmony remote, is there another possibility to achieve that?

Jason2866, 2019-11-28:
Install a Home Automation system which supports your remote (and many more).
Take a look in wiki

ascillato2, 2019-11-28:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-28 by ascillato2

s-hadinger, 2019-11-29:
Can you set `weblog 4` and paste the last logs of the discovery and the failed command?