arendst/Tasmota #7064: RF-Bridge-EFM8BB1-20191006.hex hangs the RF chip on Sonoff Bridge RF V2.0

tiagofreire-pt, 2019-11-29:


tiagofreire-pt, 2019-11-29:
```
18:00:29 CMD: rfraw aaff55
18:00:29 SRC: WebConsole from 192.168.10.1
18:00:29 CMD: Group 0, Index 1, Command "RFRAW", Data "aaff55"
18:00:29 MQT: stat/RF_Bridge_portisch/RESULT = {"RfRaw":"ON"}
18:00:30 MQT: tele/RF_Bridge_portisch/RESULT = {"Time":"2019-11-29T18:00:30","RfRaw":{"Data":"AA0355"}}
18:00:30 MQT: tele/RF_Bridge_portisch/RESULT = {"Time":"2019-11-29T18:00:30","RfRaw":{"Data":"AAA055"}}
```

ascillato, 2019-11-29:
So, RF-Bridge-EFM8BB1-20191006.hex hungs the sonoff bridge on RFRAW. And the previous versions work fine for you?

tiagofreire-pt, 2019-11-29:
I didn't try any previous version as I did go forward with the last one. May try one tonight.

ascillato, 2019-11-29:
Cool. Thanks

ascillato, 2019-11-30:
Hi, were you able to test previous Portisch Firmware?

ascillato2, 2019-12-01:
@tiagofreire-pt

Hi, were you able to test previous Portisch Firmware?

tiagofreire-pt, 2019-12-01:
It's stable since this afternoon with the previous version of Portisch. But it remains on Tasmota 6.7.

ascillato, 2019-12-01:
Ok. Cool. Great finding.

So, the issue was latest Portisch Firmware and not Tasmota.

I'm making a PR to remove latest Portisch from Tasmota repo and leaves the previous one. 

Thank you very much for testing.

PR #7082 (2019-12-01): Delete faulty Portisch Firmware
## Description:

As reported in #7064, the latest Portisch version that is in Tasmota repository, hungs when using rfraw. Using the previous version, it works fine.

**Related issue (if applicable):** fixes #7064

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.6.1
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github [... cut]
closed 2019-12-01 by arendst