arendst/Tasmota #7071: zigbee always uses software serial

clh27, 2019-11-30:
### PROBLEM DESCRIPTION
xdrv_23_zigbee_9_impl.ino calls 
     TasmotaSerial(pin[GPIO_ZIGBEE_RX], pin[GPIO_ZIGBEE_TX], 0, 0, 256);   // set a receive buffer of 256 bytes

in line 233:
    ZigbeeSerial = new TasmotaSerial(pin[GPIO_ZIGBEE_RX], pin[GPIO_ZIGBEE_TX], 0, 0, 256);   // set a receive buffer of 256 bytes

This is not enabling any hardware serial lines, please see TasmotaSerial.cpp lines 82+

_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): Wemos D1
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ X] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2019-11-30:
Good catch. Are you using GPIO 1/3 or 13/15?

clh27, 2019-11-30:
GPIO 13/15 in this case

Thx

s-hadinger, 2019-11-30:
I want to find a way to have either hardware or software serial for 13/15. Automatically switching to hardware serial would preempt the serial console which is useful when debugging.

ascillato, 2019-11-30:
> I want to find a way to have either hardware or software serial for 13/15. Automatically switching to hardware serial would preempt the serial console which is useful when debugging.

Mmm, what about being managed by the value of seriallog? If seriallog is 0 (no required info in the serial port) you can use 13/15 by hardware serial. If seriallog is >0 you have to use software serial for 13/15
.

s-hadinger, 2019-11-30:
Agreed. This is the best option.

PR #7081 (2019-12-01): Fix Zigbee uses Hardware Serial
## Description:

Zigbee will now use Hardware Serial instead of Software Serial if:

- GPIO 3 (Zigbee Rx) + 1 (Zigbee Tx)
- GPIO 13 (Zigbee Rx) + 15 (Zigbee Tx) **and** `SerialLog 0`

**Related issue (if applicable):** fixes #7071 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.6.1
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [ [... cut]
closed 2019-12-01 by arendst