arendst/Tasmota #7072: RGBW Scheme 4 has multiple issues

ratsputin, 2019-11-30:
### PROBLEM DESCRIPTION
With 7.1 Scheme 4 in an RGBW lamp (in this case an AiLight) seems to have two issues:

- Color cycling doesn't appear to honor the set Speed
- The W channel appears to stay set to maximum, causing colors to be washed-out

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): LOHAS Smart WiFi Light Bulb,MR16 LED, 50 W, 5000K
- [x] Tasmota binary firmware version number used: 7.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: OTA via Tasmota
- [x] Provide the output of command: ``Backlog Template; Module; GPIO``:
```
Configuration output here:
08:10:40 MQT: sonoff/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
08:10:40 MQT: sonoff/stat/RESULT = {"Module":{"27":"AiLight"}}
08:10:40 MQT: sonoff/stat/RESULT = {"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
```
Rules output here:

```
- [x] Provide the output of this command: ``Status 0``:
```
STATUS 0 output here:
08:11:22 CMD: status 0
08:11:22 MQT: sonoff/stat/STATUS = {"Status":{"Module":27,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:11:22 MQT: sonoff/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T22:05:06","StartupUTC":"2019-11-29T16:06:16","Sleep":50,"CfgHolder":4617,"BootCount":28,"SaveCount":319,"SaveAddress":"F6000"}}
08:11:22 MQT: sonoff/stat/STATUS2 = {"StatusFWR":{"Version":"7.1.0(tasmota)","BuildDateTime":"2019-11-29T15:33:26","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX"}}
08:11:22 MQT: sonoff/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":1,"LogHost":"192.168.1.11","LogPort":514,"SSId":["ORION_2GHz",""],"TelePeriod":300,"Resolution":"558180E0","SetOption":["00088009","2805C8000100060000005A00000000000000","00000200","00000000"]}}
08:11:22 MQT: sonoff/stat/STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
08:11:22 MQT: sonoff/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff-7911","IPAddress":"192.168.1.220","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"2C:F4:32:0A:9E:E7","Webserver":2,"WifiConfig":4}}
08:11:22 MQT: sonoff/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.11","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0A9EE7","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
08:11:22 MQT: sonoff/stat/STATUS7 = {"StatusTIM":{"UTC":"Sat Nov 30 14:11:22 2019","Local":"Sat Nov 30 08:11:22 2019","StartDST":"Sun Oct 27 02:00:00 2019","EndDST":"Sun Mar 31 01:00:00 2019","Timezone":99,"Sunrise":"01:20","Sunset":"09:56"}}
08:11:22 MQT: sonoff/stat/STATUS10 = {"StatusSNS":{"Time":1575123082}}
08:11:22 MQT: sonoff/stat/STATUS11 = {"StatusSTS":{"Time":1575123082,"Uptime":"0T22:05:06","UptimeSec":79506,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":100,"Color":"00FF0000","HSBColor":"120,100,100","Channel":[0,100,0,0],"Scheme":4,"Fade":"ON","Speed":10,"LedTable":"ON","Wifi":{"AP":1,"SSId":"ORION_2GHz","BSSId":"38:94:ED:B8:EA:CF","Channel":11,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
```
Console output here:
N/A

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
```
Fade 1
Speed 10
Scheme 4
```
### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
RBGW lamp should cycle random colors every ~5 seconds with fade

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
N/A

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Behavior changed since v6.7.1 (upgraded from this version directly to v7.1)

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-11-30:
The white channel is not controlled via the RGB schemes. You have to use `SetOption68 128`
to control it extra via Dimmer2 command.

arendst, 2019-11-30:
Speed is indeed not honored.

There were some changes recently in the speed and fade region. I'll have a look.

s-hadinger, 2019-11-30:
Theo. There was no direct change in Scheme 4 but it may be a side effect.

arendst, 2019-11-30:
Just solved it. Will release now consolidating the former random function in LightCycleColor with direction 0.

ascillato, 2019-11-30:
@ratsputin 

Hi, this has just been fixed. Could you please test it? Thanks 

http://thehackbox.org/tasmota/tasmota.bin

ratsputin, 2019-11-30:
@ascillato Just tested and it works like a champ.  Thank you very much for the quick turnaround, @arendst!

My Christmas uplights on my house are saved!  :)

closed 2019-11-30 by ratsputin