arendst/Tasmota #7074: Enable `force_update` for Home Assistant auto-discovery

floli, 2019-12-01:
**Have you looked for this feature in other issues and in the wiki?**  

Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

Feature request. Home Assistant has a `force_update` flag on sensors <https://www.home-assistant.io/integrations/sensor.mqtt/#force_update>. If **not** set, HA does not consider the sensor updated if the datum sent is identical to the previous one. By default it's not set.

Furthermore, HA does not allow to change settings of auto-discovered devices. Only alternative is to revert to full manual configuration.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

Tasmota sends configure messages like that one:
```
homeassistant/sensor/70DE99_ENERGY_Total/config:

{"tst":1575197711,"topic":"homeassistant/sensor/70DE99_ENERGY_Total/config","qos":0,"retain":1,"payloadlen":323,"payload":
{"name":"Janus1 ENERGY Total","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"70DE99_ENERGY_Total","device":{"identifiers":["70DE99"],"connections":[["mac","84:0D:8E:70:DE:99"]]},"~":"janus1/tele/","unit_of_meas":"kWh","val_tpl":"{{value_json['ENERGY'].Total}}","dev_cla":"power"}
}
```
simple add an option to add `force_update: true` to the payload. It will change nothing else on the device side and puts no additional load on the devices.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

Full manual configuration on the HA side.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2019-12-01:
@floli did you check if `last update` under sensor information in HA will correctly update when `force_update` is enabled? 
If you want graph representation is an important point and was the backside of force_update on early implementation.

floli, 2019-12-02:
Yes, I can confirm that. I modified the configuration message sent out from tasmota and added `force_update` to it (scroll to the right)
```
mosquitto_pub --retain -t homeassistant/sensor/70DE99_ENERGY_Power/config -m '{"name":"Janus1 ENERGY Power","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"70DE99_ENERGY_Power","device":{"identifiers":["70DE99"],"connections":[["mac","84:0D:8E:70:DE:99"]]},"~":"janus1/tele/","unit_of_meas":"W","val_tpl":"{{value_json['ENERGY'].Power}}","dev_cla":"power", "force_update":true}'
```
A HA template on two sensors:
```
{{ states.sensor.janus1_energy_power.last_updated }}
{{ states.sensor.janus1_energy_apparentpower.last_updated }}
```
gives me e.g.
```
2019-12-02 19:13:10.041374+00:00
2019-12-02 18:58:10.043246+00:00
```
As you can see, the apparent power is not updated, while the power, where I set `force_update: true` is updated. It is also reflected in the influxdb where the power is now updated every five minutes by HA. (iirc default of Tasmota).

That is exactly the effect I want.

> If you want graph representation is an important point and was the backside of force_update on early implementation.

Sorry, I don't understand what you mean.

Best Regards!

effelle, 2019-12-02:
On early implementation of force_update under home assistant last update was not updating correctly, meaning it was not updating the date/time of the last check, thus the internal graph was acting weird. 
Ok, I'm waiting a batch of PZEM, as soon as I have it I'll check. For now I've added it on my experimental build for temp and humidity sensors and all seems fine.

PR #7140 (2019-12-07): Home Assistant: force update added
## Description:
Enabled `force_update` for better graphs for common sensors under Home Assistant.
Updated `my_user_config.h` with the correct code size for Hass Discovery, now 4.1k.

**Related issue (if applicable):** fixes #7074

## Checklist:
  - [X] The pull request is done against the latest dev branch
  - [X] Only relevant files were touched
  - [X] Only one feature/fix was added per PR.
  - [X] The code change is tested and works on core 2.6.1
  - [X] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [X] I accept the  [... cut]
closed 2019-12-07 by arendst