arendst/Tasmota #7075: IRSEND with MITSUBISHI112 protocol appears not to work

DillonJ, 2019-12-01:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Using IRSend and with the MITSUBISHI112 protocol appears not to work. Decoding of received IR signals sent using this protocol from tasmota do work. Replacing the IR LED with an LED produces a visible flash for working protocols but there is no flash for the MITSUBISHI112 protocol. The sending device has been tested and works with other protocols.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: 7.0.0.5
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [ ] IDE / Compiler used: Atom/Platformio
- [x] Flashing tools used: OTA
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
11:47:24 MQT: stat/sonoff-2228/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
11:47:24 MQT: stat/sonoff-2228/RESULT = {"Module":{"0":"Generic"}}
11:47:25 MQT: stat/sonoff-2228/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"8":"IRsend"}}

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

  ```
- [x] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
11:48:09 MQT: stat/sonoff-2228/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff"],"Topic":"sonoff-2228","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:48:09 MQT: stat/sonoff-2228/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:16:13","StartupUTC":"2019-12-01T10:31:56","Sleep":50,"CfgHolder":4617,"BootCount":25,"SaveCount":239,"SaveAddress":"F6000"}}
11:48:09 MQT: stat/sonoff-2228/STATUS2 = {"StatusFWR":{"Version":"7.0.0.5(tasmota)","BuildDateTime":"2019-12-01T11:29:24","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(bb83b9b)","Hardware":"ESP8266EX"}}
11:48:09 MQT: stat/sonoff-2228/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Office-Wifi","Livebox-08A0"],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005AFF000000000000","00000000","00000000"]}}
11:48:09 MQT: stat/sonoff-2228/STATUS4 = {"StatusMEM":{"ProgramSize":563,"Free":440,"Heap":25,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
11:48:09 MQT: stat/sonoff-2228/STATUS5 = {"StatusNET":{"Hostname":"sonoff-2228-2228","IPAddress":"192.168.1.78","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"DC:4F:22:4C:88:B4","Webserver":2,"WifiConfig":4}}
11:48:09 MQT: stat/sonoff-2228/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.22","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4C88B4","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
11:48:09 MQT: stat/sonoff-2228/STATUS7 = {"StatusTIM":{"UTC":"Sun Dec 01 10:48:09 2019","Local":"Sun Dec 01 11:48:09 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:55","Sunset":"17:04"}}
11:48:09 MQT: stat/sonoff-2228/STATUS10 = {"StatusSNS":{"Time":"2019-12-01T11:48:09"}}
11:48:09 MQT: stat/sonoff-2228/STATUS11 = {"StatusSTS":{"Time":"2019-12-01T11:48:09","Uptime":"0T00:16:13","UptimeSec":973,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Office-Wifi","BSSId":"18:D6:C7:45:43:F0","Channel":7,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:06"}}}

  ```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
11:50:33 CMD: IRsend {"Protocol":"MITSUBISHI112","Bits":112,"Data":0x0x23CB260100A401073B0000000C08}
11:50:33 MQT: stat/sonoff-2228/RESULT = {"IRSend":"Done"}

  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Execute the command IRSEND {"Protocol":"MITSUBISHI112","Bits":112,"Data":0x0x23CB260100A401073B0000000C08}

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
A received and decoded IR message on the receiving device

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
Hi folks,

First of all, @kuchel77 thanks for implementing this protocol. I see that it has been done very recently and that it has involved a lot of work which is much appreciated. I am trying to control a Tectro air conditioner and using a SONOFF TH16 as an IR Receiver using an IRCustom build with all protocols enabled, I identified the protocol as MITSUBISHI112,

Using a NodeMCU device however, I cannot get IR signals sent via the same protocol to be received.

Here is the decoded message from the actual remote control: 

10:37:36 MQT: tele/sonoff-5308/RESULT = {"IrReceived":{"Protocol":"MITSUBISHI112","Bits":112,"Data":"0x0x23CB260100A401083B0000000C09","Repeat":0,"IRHVAC":{"Vendor":"MITSUBISHI112","Model":-1,"Power":"On","Mode":"Heat","Celsius":"On","Temp":23,"FanSpeed":"Low","SwingV":"Auto","SwingH":"Middle","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1},"RawData":[3432,1594,472,1182,474,1182,474,318,510,316,512,318,510,1180,474,318,510,318,510,1182,472,1180,476,320,508,1 ...

All good so far. So I try to send the same command with my NodeMCU using IRSEND:

IRsend {"Protocol":"MITSUBISHI112","Bits":112,"Data":0x0x23CB260100A401073B0000000C08}

No response.

To debug, I tried another command from my NodeMCU which works perfectly:

irsend { "Protocol": "NEC", "Bits": 32, "Data": 16593103}

And the response from the sonoff with IR receiver:

10:59:20 MQT: tele/sonoff-5308/RESULT = {"IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00FD30CF","DataLSB":"0x00BF0CF3","Repeat":0,"RawData":[8994,4466,624,506,598,532,600,532,624,506,624,504,598,530,598,530,600,530,600,1650,626,1626,622,1626,598,1650,624,1624,600,1650,602,528,600,1650,622,506,622,506,624,1626,600,1650,600,530,598,530,596,532,598,530,600,1650,602,1652,626,504,598,530,596,1652,596,1652,600,1650,628,1634,624],"RawDataInfo":[67,67,0]}}

So seems to be working fine with that protocol

To debug, I replaced the IR LED with a regular LED and using irsend with the NEC protocol, I see it flash - but with the MITSUBISHI112 protocol I see nothing at all.

Many thanks in advance

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-12-01:
Support for all but NEC, RC5 and RC6 has been removed from the default binaries. You'll need to use tasmota-ir.bin for all IR supported devices.

Also see RELEASENOTES.md

DillonJ, 2019-12-01:
> Support for all but NEC, RC5 and RC6 has been removed from the default binaries. You'll need to use tasmota-ir.bin for all IR supported devices.
> 
> Also see RELEASENOTES.md

Thanks for the response. I am using a self compiled version with the protocol included. As far as I understand, if I tried the IRSend command for a protocol that was not included in the compile, then I would get a protocol not supported error instead of `stat/sonoff-2228/RESULT = {"IRSend":"Done"}`

s-hadinger, 2019-12-01:
First, there seem to be a bug, because `"Data":"0x0x23CB260100A401083B0000000C09"` is invalid. There shouldn't be twice `0x`.

IrSend does not support more than 64 bits. You should use `IrHvac` instead.

Please try:

`IrHvac {"Vendor":"MITSUBISHI112","Model":-1,"Power":"On","Mode":"Heat","Celsius":"On","Temp":23,"FanSpeed":"Low","SwingV":"Auto","SwingH":"Middle","Quiet":"Off"}`

DillonJ, 2019-12-01:
Thanks for this. I tried the IrHvac command but get `{"Command":"Unknown"}`. I am running a build with just the MITSUBISHI112 protocol additionally included. Can I get access to IrHvac without the full IR build?

DillonJ, 2019-12-01:
Great! It's working now with the IR build! Thank you so much!!

arendst, 2019-12-01:
Closing as it has been solved

closed 2019-12-01 by arendst