arendst/Tasmota #7076: ubuntu/debian esptool does not work; suggest doc update

FransM, 2019-12-01:
### PROBLEM DESCRIPTION
on ubuntu 18.04 apt install esptool gave me esptool version 2.1
The command erase_flash did not work here.
I got the error: 
A fatal error occurred: ESP8266 ROM does not support function erase_flash.

The last comment in issue https://github.com/arendst/Tasmota/issues/5005 suggested that esptool 2.1 is too old and that 2.8 should be used.
When installing 2.8 as mentioned in ticket 5005 erase_flash worked.

Suggest updating the docs and not recommend to use apt to install esptool but only use the python way of installing or the method given in 5005

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Sonoff Basic
- [ x] Tasmota binary firmware version number used: 7.1.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used: esptool
- [ x] Provide the output of command: ``Backlog Template; Module; GPIO``: 
  ```
  Configuration output here:
 not applicable

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

  ```

### TO REPRODUCE
install esptool from apt (ubuntu 18.01), try to erase_flash

### EXPECTED BEHAVIOUR
flash erased

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2019-12-01:
Great findings. Thanks for reporting.

You can make a PR direct to the docs (https://tasmota.github.io/docs/#/). Click on improve this article to modify it.

Thanks

closed 2019-12-01 by ascillato2

ascillato2, 2019-12-01:
I have corrected the archived wiki with your findings. You can make those changes to the Tasmota docs. Thanks.

FransM, 2019-12-01:
Thanks, I was unaware that there was a separate github repo for docs.
Submitted a pull request

ascillato, 2019-12-01:
Merged. Thanks :+1: