arendst/Tasmota #7079: Tuya MCU Switchmode dosn't work

MacPolak, 2019-12-01:
### PROBLEM DESCRIPTION
Hi i have 1gang tuya pushbutton with Tasmota loaded via Tuya-convert.
The pushbutton is by default set as a switch but I want to use it as a doorbell (pushbutton). I tried Switchmode command and nothing. Pls help?

### REQUESTED INFORMATION
How can I perform Switchmode with Tuya MCU pushbutton instead of switch?

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
```
  STATUS 0 output here:
17:53:53 RSL: stat/sonoff/STATUS = {"Status":{"Module":54,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:53:53 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:27","StartupUTC":"2019-12-01T16:52:26","Sleep":50,"CfgHolder":4617,"BootCount":3,"SaveCount":11,"SaveAddress":"F9000"}}
17:53:53 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.7.1(sonoff)","BuildDateTime":"2019-10-26T13:19:13","Boot":7,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
17:53:53 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["PIEKIELKO",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000"]}}
17:53:53 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":564,"Free":440,"Heap":27,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","0000F081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,30","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
17:53:53 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-5966","IPAddress":"192.168.0.129","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"BC:DD:C2:BF:57:4E","Webserver":2,"WifiConfig":4}}
17:53:53 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_BF574E","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
17:53:53 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sun Dec 01 16:53:53 2019","Local":"Sun Dec 01 17:53:53 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:22","Sunset":"16:56"}}
17:53:53 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-12-01T17:53:53"}}
17:53:53 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-12-01T17:53:53","Uptime":"0T00:01:27","UptimeSec":87,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"PIEKIELKO","BSSId":"50:C7:BF:3E:17:63","Channel":11,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:05"}}}
17:53:54 TYA: Send "55aa00000000ff"
17:53:54 RSL: tele/sonoff/RESULT = {"TuyaMcuReceived":"55AA000000010101"}
17:53:54 TYA: Heartbeat
```

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-12-01:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

MacPolak, 2019-12-01:
Ok thanks

closed 2019-12-01 by MacPolak

ascillato2, 2019-12-02:
@MacPolak Were you able to solve your issue?

MacPolak, 2019-12-02:
No, I moved it to  Tasmota Support Chat on Discord but it's still unsolved

ascillato, 2019-12-02:
I was asking, because I don't know if this can be solved, because, if the button/switch is controlled by another MCU, Tasmota don't have access to it.

Which device is it?

MacPolak, 2019-12-02:
Yeah I am afraid its just like you wrote - the mru control the button, but maybe I could use rules to switch it on for 2 or 3 second and then to switch it off again, but I am not sure how it works when someone push the button couple of times. It's this one:[https://pl.aliexpress.com/item/4000241193526.html?spm=a2g0s.9042311.0.0.374d5c0fQB507b].

ascillato, 2019-12-02:
So, you have physically modified it from switch to button right? you have put a spring on it?

MacPolak, 2019-12-02:
no it was sold like that, there is a small tact switch (button inside like this [one:](https://www.elfadistrelec.pl/pl/przelacznik-klawiaturowy-pcb-1no-50-ma-6x6mm-omron-electronic-components-b3f-1020/p/11088841?channel=b2c&price_gs=.7995&source=googleps&ext_cid=shgooaqplpl-na&gclid=EAIaIQobChMIqf_oooKX5gIVwp13Ch0yFAqeEAYYASABEgLeC_D_BwE) )and the lid is with spring

ascillato, 2019-12-02:
Ok, your tuya device is a pushbutton.

So, you want to send a MQTT message to your doorbell?

MacPolak, 2019-12-02:
from the doorbell via MQTT to the broker than as a push notification to iPhones, but I also want to create some automation later for example: someone push the doorbell -> camera in the peephole take a photo and sends it to Home Assistant -> HA sends it to iPhone as notification

ascillato, 2019-12-02:
Ok, please try the following rule:

```
rule1 1
rule1 on switch1#state do publish stat/mytasmotatopic/doorbeel ON endon
```

if nothing happens, try this other instead:

```
rule1 1
rule1 on button1#state do publish stat/mytasmotatopic/doorbeel ON endon
```

MacPolak, 2019-12-02:
ok thanks but I need to add before endon because classic electric bell is connected to this button so I am afraid the command will be executed to fast to trigger the bell

ascillato, 2019-12-02:
Sorry, I don't understand your comment. Please explain.

MacPolak, 2019-12-02:
sorry I mean I need time delay before endon, I think about 2-3 seconds, that's just because without delay the rule be executed to fast for electrical bell to ring

ascillato, 2019-12-02:
I think I'm not following you. Why you would need a delay?

Exactly what do you want Tasmota to do?

MacPolak, 2019-12-02:
after switch is on, and before it goes back off (like a regular doorbell button pushed by someone for 2 seconds)

ascillato, 2019-12-02:
Mmmm. Sorry. Still not understanding what do you need.

Exactly what do you want Tasmota to do? Someone push the button and which steps Tasmota should do for you?

MacPolak, 2019-12-02:
As i understand your rule send message after when the button is pushed. But as I wrote it works not like a push button but like switch so after the push I will stay on. So I need a rule like: when switch is on then send MQTT message, wait 2 seconds and turn off the switch

ascillato, 2019-12-02:
How it will turn off the switch? The switch is physical. As per your link, your device is a pushbutton. So you push, it tells Tasmota that its states is 1 and when you release it it turns to 0.right?

ascillato, 2019-12-02:
Let's move to the Chat. This thread is getting longer just asking what you want to achieve and how is your hardware. It is much better to chat.

MacPolak, 2019-12-02:
Ok what chat?