arendst/Tasmota #7080: Sonoff T1 (3CH knockoff) not working has expected with 7.1.0 and 7.1.1

gajotnt, 2019-12-01:
### PROBLEM DESCRIPTION
With version 7.1.0 (and 7.1.1) my Sonoff T1 3Ch (knockOff version) touch sensor 2 and 3 stopped reacting to touch, by webpanel the relays will work as expected.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- X[ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [X ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): Sonoff T1 3Ch knockoff
- [X ] Tasmota binary firmware version number used: 7.1.0
  - [X ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [X ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
16:06:55 MQT: stat/CozQuintal/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
16:06:55 MQT: stat/CozQuintal/RESULT = {"Module":{"30":"Sonoff T1 3CH"}}
16:06:55 MQT: stat/CozQuintal/RESULT = {"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"}}

  ```
- [X ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
16:04:11 MQT: stat/CozQuintal/STATUS = {"Status":{"Module":30,"FriendlyName":["CozQuintal","CozQuintal1","CozQuintal2"],"Topic":"CozQuintal","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}

  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Flash a T1 3ch (knockoff) with version 7.1.0, even with 7.1.1 that i just tested issue remains.

With version **7.0.0.5** it is working has expected, touch sensor 2 and 3 detect the user input.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Well to work as a Sonoff T1 3ch (as it has alwasys been working so far :P )

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-12-01:
Investigating...

arendst, 2019-12-01:
The issue is caused by the hardware detection code. It fails to detect it is an ESP8285 therefor it disables GPIO9 and GPIO10 where your buttons are connected.

I will revert the hardware detection code.

arendst, 2019-12-01:
Pls give the latest dev version a try. It works for me now.

gajotnt, 2019-12-01:
Also confirm its working now. Thanks :)

closed 2019-12-01 by gajotnt