arendst/Tasmota #7084: Sonoff B1 Module not working - gui slider missing 7.1.1.1

timtimsson, 2019-12-02:
Hello team,
Sonoff B1 Module not working - gui slider missing 7.1.1.1
Thanks

closed 2019-12-02 by close-issue-app[bot]

arendst, 2019-12-02:
????

[image]

timtimsson, 2019-12-02:
hi theo,
i compile with arduino 1.8.10 - 2.6.2 by my one with the latest github from my old 6.6.0.
//#define USE_CONFIG_OVERRIDE
#define FIRMWARE_BASIC
???
[image]

[image]

arendst, 2019-12-02:
There you have it. 

tasmota-basic does NOT support the B1 driver as can be found in the Release notes.

timtimsson, 2019-12-02:
OMG! i waste your time - i'm so sorry!!!