arendst/Tasmota #7085: Alexa problem with WS2812 & Relais

Vujagig, 2019-12-02:
### PROBLEM DESCRIPTION
If a port is defined with WS2812, everything works fine.
Once a second port is defined as a relay, Alexa 2 relay ports will be reported and the WS2812 will no longer work on Alexa. The problem persists, even if the relay port is hidden by $ in front of Alexa.
Furthermore, I noticed that the color picker and brightness is not updated when a different color or brightness is set via Alexa. Only after manually updating the web page will the new values be displayed.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Wemos D1 min Pro
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: Atom/PlatformIO
- [X] Flashing tools used: OTA
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-12-02:
Regarding sliders it is as designed.

For Alexa you provided not enough information as requested when you openend the issue.

Vujagig, 2019-12-02:
If, in addition to the port defined as WS2812, another port is defined with a relay, think Alexa that both ports are relays. Of course, after clear and re-recognize Alexa.
1 port only WS2812 > OK
1 port WS2812 & 1 port relay > fail

ascillato, 2019-12-02:
Hi, please provide the output of `status 0` and also set weblog to 4 for more debug information and post the output of the console when you are trying to control ws2812 and the relay by Alexa. Thanks.

Vujagig, 2019-12-02:
OK, now I'm a little confused, Alexa recognizes only one device, but classifies this as a Realis. I also tried completely different names. these are now no longer recognized, instead only a device with the name "LED". Does Alexa have a bad day today?
Friendly Name 1: LED 1 >  D7 GPIO13 Relay1
Friendly Name 2: LED 2 > D8 GPIO15 WS2812

***********Alexa search start*************
`16:30:30 WIF: Checking connection...
16:30:30 WIF: Connected
16:30:32 UDP: Packet (94)
16:30:32 UDP: Packet (101)
16:30:32 UDP: Packet (94)
16:30:32 UDP: Packet (101)
16:30:32 UDP: Packet (94)
16:30:32 UDP: Packet (101)
16:30:32 UDP: Packet (94)
16:30:32 UDP: Packet (101)
16:30:32 UDP: Packet (94)
16:30:32 UDP: Packet (101)
16:30:34 UPP: Hue 3 response packets sent to 192.168.10.31:50000
16:30:35 HTP: Hue setup
16:30:35 HTP: Hue API (/1f6f85/lights)
16:30:35 HTP: Hue User-Agent: Dalvik/2.1.0 (Linux; U; Android 7.1.2; AEODN Build/NS6431), gen=2
16:30:35 HTP: Hue User-Agent: Dalvik/2.1.0 (Linux; U; Android 7.1.2; AEODN Build/NS6431), gen=2
16:30:35 HTP: Hue Result ({"32962641":{"state":{"on":false,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"LED 1","modelid":"LCT007","uniqueid":"2c:3a:e8:1f:6f:85:00:11-1","swversion":"5.50.1.19085"},"32962642":{"state":{"on":true,"bri":254,"colormode":"hs","xy":[0.46766,0.15076],"hue":56955,"sat":254,"ct":153,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"LED 2","modelid":"LCT007","uniqueid":"2c:3a:e8:1f:6f:85:00:11-2","swversion":"5.50.1.19085
16:30:35 HTP: Hue API (/1f6f85/lights/32962642)
16:30:35 /lights path=/lights/32962642
16:30:35 HTP: Hue User-Agent: Dalvik/2.1.0 (Linux; U; Android 7.1.2; AEODN Build/NS6431), gen=2
16:30:35 HTP: Hue Result ({"state":{"on":true,"bri":254,"colormode":"hs","xy":[0.46766,0.15076],"hue":56955,"sat":254,"ct":153,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"LED 2","modelid":"LCT007","uniqueid":"2c:3a:e8:1f:6f:85:00:11-2","swversion":"5.50.1.19085"})
16:30:35 HTP: Hue API (/1f6f85/lights/32962641)
16:30:35 /lights path=/lights/32962641
16:30:35 HTP: Hue User-Agent: Dalvik/2.1.0 (Linux; U; Android 7.1.2; AEODN Build/NS6431), gen=2
16:30:35 HTP: Hue Result ({"state":{"on":false,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"LED 1","modelid":"LCT007","uniqueid":"2c:3a:e8:1f:6f:85:00:11-1","swversion":"5.50.1.19085"})
16:30:35 MQT: tele/tasmota/STATE = {"Time":"2019-12-02T16:30:35","Uptime":"0T00:05:13","UptimeSec":313,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER1":"OFF","POWER2":"ON","Dimmer":100,"Color":"FF00CC00","HSBColor":"312,100,100","Channel":[100,0,80,0],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"SAW-AP","BSSId":"44:4E:6D:39:96:E6","Channel":1,"RSSI":44,"LinkCount":1,"Downtime":"0T00:00:07"}}
16:30:35 MQT: tele/tasmota/SENSOR = {"Time":"2019-12-02T16:30:35","DS18B20":{"Id":"031672BEDFFF","Temperature":9.1},"TempUnit":"C"}
16:30:50 WIF: Checking connection...
16:30:50 WIF: Connected
16:31:10 WIF: Checking connection...
16:31:10 WIF: Connected
16:31:12 HTP: Hue API (/1f6f85/lights/32962641)
16:31:12 /lights path=/lights/32962641
16:31:12 HTP: Hue User-Agent: Dalvik/2.1.0 (Linux; U; Android 7.1.2; AEODN Build/NS6437), gen=2
16:31:12 HTP: Hue Result ({"state":{"on":false,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"LED 1","modelid":"LCT007","uniqueid":"2c:3a:e8:1f:6f:85:00:11-1","swversion":"5.50.1.19085"})
16:31:30 WIF: Checking connection...
16:31:30 WIF: Connected
16:31:31 UDP: Packet (199)
16:31:31 UDP: Packet (199)
16:31:31 UDP: Packet (199)

*****************Alexa LED on/off**************
16:38:57 SRC: WebConsole from 192.168.10.101
16:39:07 HTP: Hue API (/1f6f85/lights/32962641/state)
16:39:07 HTP: Hue POST args ({"on":true})
16:39:07 SRC: Hue
16:39:07 MQT: stat/tasmota/RESULT = {"POWER1":"ON"}
16:39:07 MQT: stat/tasmota/POWER1 = ON
16:39:07 HTP: Hue Result ([{"success":{"/lights/32962641/state/on":true}}])
16:39:07 HTP: Hue API (/1f6f85/lights/32962641)
16:39:07 /lights path=/lights/32962641
16:39:07 HTP: Hue User-Agent: Dalvik/2.1.0 (Linux; U; Android 7.1.2; AEODN Build/NS6437), gen=2
16:39:07 HTP: Hue Result ({"state":{"on":true,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"LED 1","modelid":"LCT007","uniqueid":"2c:3a:e8:1f:6f:85:00:11-1","swversion":"5.50.1.19085"})
16:39:08 CFG: Saved to flash at F7, Count 77, Bytes 4096
16:39:10 WIF: Checking connection...
16:39:10 WIF: Connected
16:39:16 HTP: Hue API (/1f6f85/lights/32962641/state)
16:39:16 HTP: Hue POST args ({"on":false})
16:39:16 SRC: Hue
16:39:16 MQT: stat/tasmota/RESULT = {"POWER1":"OFF"}
16:39:16 MQT: stat/tasmota/POWER1 = OFF
16:39:16 HTP: Hue Result ([{"success":{"/lights/32962641/state/on":false}}])
16:39:16 HTP: Hue API (/1f6f85/lights/32962641)
16:39:16 /lights path=/lights/32962641
16:39:16 HTP: Hue User-Agent: Dalvik/2.1.0 (Linux; U; Android 7.1.2; AEODN Build/NS6437), gen=2
16:39:16 HTP: Hue Result ({"state":{"on":false,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"LED 1","modelid":"LCT007","uniqueid":"2c:3a:e8:1f:6f:85:00:11-1","swversion":"5.50.1.19085"})
16:39:17 CFG: Saved to flash at F6, Count 78, Bytes 4096
16:39:28 UDP: Packet (199)
16:39:28 UDP: Packet (199)
`

ascillato2, 2019-12-02:
Please, post the output of `status 0` command

ascillato2, 2019-12-04:
Please, post the output of status 0 command

s-hadinger, 2019-12-07:
@Vujagig I looked at the Alexa logs you posted earlier, and I don't see anything wrong.

Tasmota reports two devices to Alexa:

- 32962641: simple power on/off, named 'LED 1'
- 32962642: full color bulb, named 'LED 2'

When you trigger On/Off on 'LED 1', everything looks normal in the logs. What is wrong?

ascillato2, 2019-12-09:
Closing this issue as there is no feedback from the user.

closed 2019-12-09 by ascillato2