arendst/Tasmota #7090: WS2812 Power OFF not working when scheme is selected

helmar74, 2019-12-02:
Since Tasmota 7.1.0 (and also in 7.1.1) I have the following issue:
A WS2812 LED Stripe is connected and working fine. If I set webcommand "scheme 12" the scheme is working.
But I can't turn off the LED Stripe anymore.
If I set scheme 0 (one color scheme) I can power off the LEDs

I also tried "power1 off", without any success.
Here's a webconsole LOG:
17:21:01 MQT: stat/sonoffWemosMini_5/SCHEME = {"Scheme":12} 17:21:01 MQT: stat/sonoffWemosMini_5/DIMMER = {"POWER1":"ON","Dimmer":100,"Color":"255,255,255,0","HSBColor":"0,0,100","Channel":[100,100,100,0]} 17:21:01 MQT: stat/sonoffWemosMini_5/SCHEME = {"Scheme":12} 17:21:01 MQT: stat/sonoffWemosMini_5/PIXELS = {"Pixels":198} 17:21:02 MQT: stat/sonoffWemosMini_5/SPEED = {"Speed":1} 17:21:02 MQT: stat/sonoffWemosMini_5/FADE = {"Fade":"ON"} 17:21:03 CMD: power off -> LEDS are still on 17:21:03 MQT: stat/sonoffWemosMini_5/POWER1 = {"POWER1":"OFF"} 17:21:03 MQT: stat/sonoffWemosMini_5/POWER1 = OFF 17:21:07 CMD: power1 off -> LEDS are still on 17:21:07 MQT: stat/sonoffWemosMini_5/POWER1 = {"POWER1":"OFF"} 17:21:07 MQT: stat/sonoffWemosMini_5/POWER1 = OFF -> LEDS are still on 17:22:14 CMD: scheme 0 17:22:17 CMD: power off -> LEDs are off

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Wemos Mini
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ X] Self-compiled
    - [ X] IDE / Compiler used: Arduino
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
7:26:58 CMD: Backlog Template; Module; GPIO
17:26:58 MQT: stat/sonoffWemosMini_5/TEMPLATE = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":1,"BASE":18}
17:26:58 MQT: stat/sonoffWemosMini_5/MODULE = {"Module":{"18":"Generic"}}
17:26:58 MQT: stat/sonoffWemosMini_5/GPIO = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"7":"WS2812"},"GPIO3":{"0":"None"},"GPIO4":{"6":"I2C SDA"},"GPIO5":{"5":"I2C SCL"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
17:24:05 MQT: stat/sonoffWemosMini_5/STATUS = {"Status":{"Module":18,"FriendlyName":["sonoffWemosMini-5-GRBW","Sonoff2"],"Topic":"sonoffWemosMini_5","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"sonoffWemosMini_5","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}} 17:24:05 MQT: stat/sonoffWemosMini_5/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffWemosMiniWS2812GRBWGruppe","OtaUrl":"http://10.1.1.2:5555/data/firmwares/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T01:03:09","StartupUTC":"2019-12-02T15:20:56","Sleep":50,"CfgHolder":4617,"BootCount":69,"SaveCount":898,"SaveAddress":"F5000"}} 17:24:05 MQT: stat/sonoffWemosMini_5/STATUS2 = {"StatusFWR":{"Version":"7.1.1(tasmota)","BuildDateTime":"2019.12.01 21:07:30","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(bb83b9b)","Hardware":"ESP8266EX"}} 17:24:05 MQT: stat/sonoffWemosMini_5/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SSID_24ghz","sonoffap"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00028019","2805C8000100060000005A00000000000000","00000001","00000000"]}} 17:24:05 MQT: stat/sonoffWemosMini_5/STATUS4 = {"StatusMEM":{"ProgramSize":593,"Free":408,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000407","8FAAE19F","00278FA0","22BE1FCF","012033C0","00007081"],"Drivers":"1,2,3,4,5,6,8,9,10,13,16,17,18,19,22,24,26","Sensors":"1,3,4,5,6,7,8,9,10,14,15,16,17,18,20,22,26,35,41"}} 17:24:05 MQT: stat/sonoffWemosMini_5/STATUS5 = {"StatusNET":{"Hostname":"sonoffWemosMini-5-GRBW","IPAddress":"10.1.1.178","Gateway":"10.1.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.1.1.2","Mac":"DC:4F:22:1D:88:E5","Webserver":2,"WifiConfig":2}} 17:24:05 MQT: stat/sonoffWemosMini_5/STATUS6 = {"StatusMQT":{"MqttHost":"10.1.1.50","MqttPort":1883,"MqttClientMask":"sonoffWemosMini-5-GRBW","MqttClient":"sonoffWemosMini-5-GRBW","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}} 17:24:05 MQT: stat/sonoffWemosMini_5/STATUS7 = {"StatusTIM":{"UTC":"Mon Dec 02 16:24:05 2019","Local":"Mon Dec 02 17:24:05 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"07:41","Sunset":"16:15"}} 17:24:05 MQT: stat/sonoffWemosMini_5/STATUS10 = {"StatusSNS":{"Time":"2019-12-02T17:24:05","BH1750":{"Illuminance":11}}} 17:24:05 MQT: stat/sonoffWemosMini_5/STATUS11 = {"StatusSTS":{"Time":"2019-12-02T17:24:05","Uptime":"0T01:03:09","UptimeSec":3789,"Vcc":3.062,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","Dimmer":100,"Color":"255,255,255,0","HSBColor":"0,0,100","Channel":[100,100,100,0],"Scheme":12,"Width":1,"Fade":"ON","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"SSID_24ghz","BSSId":"E8:DE:27:FE:A9:E5","Channel":11,"RSSI":94,"LinkCount":1,"Downtime":"0T00:00:06"}}}

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-12-02:
Pls give it a try and report back.

helmar74, 2019-12-02:
Hi, 
I compiled and installed the Dev Version 7.1.1.1.
Yes, it's fixed now! I can Power OFF and ON in every scheme now.
BUT....
The Schemes 2 and 3 seems not work anymore.
My LEDs only shows white, when scheme 2 or 3 are selected...
This is shown in the Webconsole:
20:56:35 MQT: stat/sonoffWemosMini_5/SCHEME = {"Scheme":2}
20:56:35 MQT: stat/sonoffWemosMini_5/DIMMER = {"POWER1":"ON","Dimmer":100,"Color":"255,255,255,0","HSBColor":"101,0,100","Channel":[100,100,100,0]}
20:56:35 MQT: stat/sonoffWemosMini_5/SCHEME = {"Scheme":2}
20:56:35 MQT: stat/sonoffWemosMini_5/PIXELS = {"Pixels":198}
20:56:35 MQT: stat/sonoffWemosMini_5/SPEED = {"Speed":1}
20:56:35 MQT: stat/sonoffWemosMini_5/FADE = {"Fade":"ON"}

arendst, 2019-12-02:
That's a new feature. By changing saturation you'll get the colors hack. Use the slider or command hsbcolor2 100

helmar74, 2019-12-02:
Hi Theo.
Yeah that's fixed it. 
Thanks a lot for your great work!!!
- Closing the issue -

closed 2019-12-02 by helmar74