arendst/Tasmota #7091: new TuyaMCU device for RGB strips (named Spider T)

volkerrichert, 2019-12-02:
**Have you looked for this feature in other issues and in the wiki?**  yes

**Is your feature request related to a problem? Please describe.**  
I received a new tuya MCU device called spider T. It's used as a RGB controller. But a TuyaMCU doesn't have a RGB control and the color seems to be 12 char log

**Describe the solution you'd like**  
I'd like to have the color mode been supported 

**Additional context**  
I already found DpId 20 as on / off switch. But changing the color "red" results in
`18:03:07 {"TuyaReceived":{"Data":"55AA03070005150400010129","Cmnd":7,"CmndData":"1504000101","DpId":21,"DpIdType":4,"DpIdData":"01"}}
18:03:07 TYA: fnId=0 is set for dpId=21
18:03:07 {"TuyaReceived":{"Data":"55AA03070008160200040000006491","Cmnd":7,"CmndData":"1602000400000064","DpId":22,"DpIdType":2,"DpIdData":"00000064"}}
18:03:07 TYA: fnId=0 is set for dpId=22
18:03:07 {"TuyaReceived":{"Data":"55AA030700101803000C303030303033653830303634CA","Cmnd":7,"CmndData":"1803000C303030303033653830303634","DpId":24,"DpIdType":3,"DpIdData":"303030303033653830303634","Type3Data":"000003e80064"}}
18:03:07 TYA: fnId=0 is set for dpId=24`

green is
`18:03:52 {"TuyaReceived":{"Data":"55AA03070005150400010129","Cmnd":7,"CmndData":"1504000101","DpId":21,"DpIdType":4,"DpIdData":"01"}}
18:03:52 TYA: fnId=0 is set for dpId=21
18:03:52 {"TuyaReceived":{"Data":"55AA03070008160200040000006491","Cmnd":7,"CmndData":"1602000400000064","DpId":22,"DpIdType":2,"DpIdData":"00000064"}}
18:03:52 TYA: fnId=0 is set for dpId=22
18:03:52 {"TuyaReceived":{"Data":"55AA030700101803000C303037383033653830303634D9","Cmnd":7,"CmndData":"1803000C303037383033653830303634","DpId":24,"DpIdType":3,"DpIdData":"303037383033653830303634","Type3Data":"007803e80064"}}
18:03:52 TYA: fnId=0 is set for dpId=24`

and blue:

`18:04:21 {"TuyaReceived":{"Data":"55AA03070005150400010129","Cmnd":7,"CmndData":"1504000101","DpId":21,"DpIdType":4,"DpIdData":"01"}}
18:04:21 TYA: fnId=0 is set for dpId=21
18:04:21 {"TuyaReceived":{"Data":"55AA03070008160200040000006491","Cmnd":7,"CmndData":"1602000400000064","DpId":22,"DpIdType":2,"DpIdData":"00000064"}}
18:04:21 TYA: fnId=0 is set for dpId=22
18:04:21 {"TuyaReceived":{"Data":"55AA030700101803000C30306630303365383030363400","Cmnd":7,"CmndData":"1803000C303066303033653830303634","DpId":24,"DpIdType":3,"DpIdData":"303066303033653830303634","Type3Data":"00f003e80064"}}
18:04:21 TYA: fnId=0 is set for dpId=24` 

It also has some "color modes" like FLASH, FADE, etc. and support 3 kinds of music reaction

volkerrichert, 2019-12-13:
next finding,

dpId controlls color & brightness:

```
#FF0000 -> 000003e803e8	              <- last 4 char are for brightness in 1/1000 steps
	   	   000003e802bc
		   ..
		   000003e80064
```

so, `000003e8` means red, green is `007803e8` and blue `00f003e8`. White is coded by "00000000" mixed colors are also possible: purple is `012c03e8` (all without brightness)

Any idea who this could be encoded? 

thw, dpID 21 it for switching modes (type 4) like color, flash, strobe, fade, smooth and music

volkerrichert, 2019-12-13:
ok, next finding. The color seems to be HS(B)

HHHHSSSSBBBB

HHHH in degree form 0-360
SSSSS in 1/10 % -> form 0 to 1000
BBBB in 1/10 % -> form 0 to 1000

is there a way to use the tasmota color widgets to set those encoding?

closed 2020-01-12 by stale[bot]

JulianMined, 2020-01-20:
Hello,
i got the same values, did you have more informations?

volkerrichert, 2020-02-23:
@JulianMined : There are a few modes (music, etc) on other DpIds. It's easy to see them. I'll stay with mqtt atm and use ioBroker JS to handle my alarms, blinking, flashing etc. If there important additional founding, i'll doc them here

cangurellobello, 2020-07-30:
Check out #8880 if you still have trouble setting colors, i have a tuyaMCU lamp with a dpId that yields the same data structure as yours. I managed to fully map the data stream.

ilanoud, 2021-01-24:
Hi,
Recently got the same LED strip (spider T), successfully flashed it with Tasmota but now I can’t figure how to configurate it...

Can someone explain please?
I’d like to select colors, brightness and use on/off, all by using Tasmota directly at first...

Thx

ascillato, 2021-01-24:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md)

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)