arendst/Tasmota #7092: Return scaled peak to peak range of analog input as an option

DillonJ, 2019-12-02:
**Have you looked for this feature in other issues and in the wiki?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
I have configured a NodeMCU as a simple AC current meter using a 20A/1V current transformer and a simple circuit to add a dc bias to the ac voltage (similar to here: https://learn.openenergymonitor.org/electricity-monitoring/ct-sensors/interface-with-arduino

The current code returns the average of 32 samples with a 1ms delay. I would like instead to receive the peak to peak voltage over this range. I have modified the code to do this myself as it is quite simple, but I am sure my code is inefficient compared to what someone else might write. This is how I have rewritten xsns_02_analog.ino, just so you know what I'm on about.

This works quite nicely and when scaled properly gives a pretty accurate current measurement and an apparent power measurement using the voltage from a PZEM based monitor I have elsewhere in the house.

Additionally it would be nice to be able to scale the value.

I can imagine this working by having an additional option in the analog dropdown like AC range or something like that, or even, scaled AC range. It would be nice to be able to send the scaling factor via MQTT.

```
uint16_t AdcRead(uint8_t factor)
{
  // factor 1 = 2 samples
  // factor 2 = 4 samples
  // factor 3 = 8 samples
  // factor 4 = 16 samples
  // factor 5 = 32 samples
  uint8_t samples = 1 << factor;
  uint16_t analog = 0;
  uint16_t analog_min = 1023;
  uint16_t analog_max = 0;
  for (uint32_t i = 0; i < samples; i++) {
    //analog += analogRead(A0);
    analog = analogRead(A0);
    if (analog < analog_min) {
      analog_min=analog;
    }
    if (analog > analog_max) {
      analog_max=analog;
    }

    delay(1);
  }
//  analog >>= factor;
  analog=analog_max-analog_min;
  return analog;
}
```

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
Doing the scaling etc via node red - but it would be nice to have the scaled value at source and see it displayed on the webserver

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2019-12-02:
This could be added as a new **AdcParam** (https://github.com/arendst/Tasmota/wiki/Commands#sensor).

If you can, you can try make a PR for this feature.

DillonJ, 2019-12-02:
> This could be added as a new **AdcParam** (https://github.com/arendst/Tasmota/wiki/Commands#sensor).
> 
> If you want, you can try make a PR for this feature.

Nice idea... I will take a look and give it a shot

DillonJ, 2019-12-03:
OK, so I am nearly ready with a pull request for this. 

The only thing I'm struggling with is how to run tests? The pull request form mentions all tests should work - but where are they and how are they run?

ascillato, 2019-12-03:
> The pull request form mentions all tests should work - but where are they and how are they run?

After you test your code locally with your device, you can make the Pull Request and it will be automatically tested the compilation in all firmware types by travis. No need for you to do something extra. Just check, after making the PR, that the yellow dot that it is shown right after the PR name turns to a green tick. If it turns to a red cross, you can click it and see what is the compilation issue. Don't worry, we will help you on that.

DillonJ, 2019-12-04:
Ok, so I have made a pull request and all the tests have passed...
https://github.com/arendst/Tasmota/pull/7100

ascillato2, 2019-12-04:
Closing this issue as the PR #7100 has been submitted for review. Thanks

closed 2019-12-04 by ascillato2